networkupstools/nut #1893: Unable to start the addon

MiAutomations, 2023-04-01:
Hello to all 

I'm using the last versions available for Home Assistant and for this addon 

I'm currently facing this issue: 

s6-rc: warning: unable to start service legacy-cont-init: command exited 1
/run/s6/basedir/scripts/rc.init: warning: s6-rc failed to properly bring all the services up! Check your logs (in /run/uncaught-logs/current if you have in-container logging) for more information.
/run/s6/basedir/scripts/rc.init: fatal: stopping the container.
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

I Already try several ways in order to try solve this: uninstall the addon restart the  Home Assistant remove the Integration 

But the issue persists ...

Can anyone please help me in order to try solve this issue ? 

Thank you in advance for your help and support

jimklimov, 2023-04-01:
You may be better off asking in Home Assistant community. As far as NUT is concerned, you have probably misconfigured something and it does not find a device, etc. Without knowing even the version of NUT installed there, and whether you can get into that (container?) environment to modify CLI options like debug, it is hard to suggest more.

It might help to roll your own plugin from current NUT sources (somebody's instructions were copied to Wiki here), to have current codebase friendliest for troubleshooting. And to generally revise other issues tagged as about HA, maybe somebody stepped on the same rake before.

MiAutomations, 2023-04-04:
Hello 

No the device is online on the IMOU App and I can access to the device using the IMOU App without any issue 

But over this integration I'm facing this issues 

[image]