networkupstools/nut #1902: 5E 850i USB support

Proche72, 2023-04-11:
Hey, I got EATON 5E 850i USB for a very good price and I was hoping to use it with NUT in HA.  Smaller and bigger ones are compatible (650i and 1100i) but the 850i is not.

I am not able to connect it to the NUT addon in HA. Tried several cables and also USB hub. None helped.
Is there a way to connect it?

Proche72, 2023-04-12:
NUT addon in HA is version 0.11.2.
driver is usbhid-ups
How can I try simple build of NUT?
log from addon:
`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/nut.sh
[18:34:40] INFO: Setting mode to netserver...
[18:34:40] INFO: Generating /etc/nut/upsd.users...
[18:34:40] INFO: Configuring user: Proche72
[18:34:41] INFO: Password is NOT in the Have I Been Pwned database! Nice!
[18:34:41] INFO: Configuring Device named myups...
[18:34:41] INFO: Starting the UPS drivers...
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
No matching HID UPS found
Network UPS Tools - UPS driver controller 2.7.4
Driver failed to start (exit status=1)
cont-init: info: /etc/cont-init.d/nut.sh exited 1
cont-init: info: running /etc/cont-init.d/nutclient.sh
cont-init: info: /etc/cont-init.d/nutclient.sh exited 0
cont-init: warning: some scripts exited nonzero
s6-rc: warning: unable to start service legacy-cont-init: command exited 1
/run/s6/basedir/scripts/rc.init: warning: s6-rc failed to properly bring all the services up! Check your logs (in /run/uncaught-logs/current if you have in-container logging) for more information.
/run/s6/basedir/scripts/rc.init: fatal: stopping the container.
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped`

UPS is connected to miniPC running proxmox (with USB passtrough) but the problem was there even when I was using RPi4 with HAOS directly on it.

jimklimov, 2023-04-12:
To try a simple build of NUT, spin up a computer/vm/container with any supported OS, install the tools and dependencies as listed in the docs/config-prereqs.txt, and build it - often as easy as `./ci_build.sh` for a quick shot. You can then run the driver right from the build workspace in dump mode with debug, to see if new codebase handles your device better (or just how it complains). One overview is at https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests

As for HA, ask them about troubleshooting and nag to issue a current plugin (NUT 2.7.4 is 7 years old). That said, a contributed post may help you build your own: https://github.com/networkupstools/nut/wiki/Building-NUT-integration-for-Home-Assistant

Proche72, 2023-04-13:
Thank you, I will try it.

closed 2023-05-02 by Proche72