networkupstools/nut #1913: Configuring NUT server on APC SMT750IC more sensors

platini76, 2023-04-19:
hello I configured nut on my APC SMT750IC
with this config… don’t know if this is correct:

`users:
  - username: XXXXX
    password: XXXXX
    instcmds:
      - all
    actions: []
devices:
  - name: SmartUPS
    driver: usbhid-ups
    port: auto
    config:
      - vendorid = 051d
mode: netserver
shutdown_host: "true"
`

and I have only these sensors:
[image]
no input voltage, no output voltage, no load, no controls…

can I have more on this UPS?

or someone can suggest me a better ups?

i see that other apc ups like 900 and 1000 has load and voltage infos... why 750 not?

jimklimov, 2023-04-19:
I would try to guess this is HomeAssistant, and so NUT v2.7.4 (pretty old now), but generally these things should not be guessed by readers ;)

Regarding info and controls, it may be that this is among APC models which all but dropped USB HID support and went for modbus, a driver for which is not currently implemented in NUT => #139

jimklimov, 2023-04-19:
Do you have a chance to build NUT to test if a modern version supports your device better (if the problem is not Modbus), and iterate from there? See https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests

platini76, 2023-04-19:
sorry:

[21:18:39] INFO: Setting mode to netserver...
[21:18:39] INFO: Generating /etc/nut/upsd.users...
[21:18:39] INFO: Configuring user: platini76
[21:18:40] INFO: Password is NOT in the Have I Been Pwned database! Nice!
[21:18:41] INFO: Configuring Device named SmartUPS...
[21:18:41] INFO: Starting the UPS drivers...
interrupt pipe disabled (add 'pollonly' flag to 'ups.conf' to get rid of this message)
Using subdriver: APC HID 0.96
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
Network UPS Tools - UPS driver controller 2.7.4
[21:18:42] WARNING: UPS Shutdown will shutdown the host
cont-init: info: /etc/cont-init.d/nut.sh exited 0
cont-init: info: running /etc/cont-init.d/nutclient.sh
cont-init: info: /etc/cont-init.d/nutclient.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun upsd (no readiness notification)
services-up: info: copying legacy longrun upsmon (no readiness notification)
s6-rc: info: service legacy-services successfully started
[21:18:42] INFO: Starting the UPS information server...
   0.000000	fopen /run/nut/upsd.pid: No such file or directory
   0.001717	listening on 0.0.0.0 port 3493
   0.003905	Connected to UPS [SmartUPS]: usbhid-ups-SmartUPS
[21:18:42] INFO: Starting the UPS monitor and shutdown controller...
   0.000000	fopen /run/nut/upsmon.pid: No such file or directory
   0.001102	Using power down flag file /etc/killpower
   0.001690	UPS: SmartUPS@localhost (master) (power value 1)
   0.002130	debug level is '1'
   0.002380	Warning: running as one big root process by request (upsmon -p)
   0.003220	Init SSL without certificate database
   0.008634	Trying to connect to UPS [SmartUPS@localhost]
   0.032386	User upsmonmaster@127.0.0.1 logged into UPS [SmartUPS]
   0.011487	Logged into UPS SmartUPS@localhost

platini76, 2023-04-19:
sorry I cannot build.. I am a noob on this ...

jimklimov, 2023-04-19:
We have all started somewhere :) It is fun ;)

platini76, 2023-04-19:
it seems the problem is 2.7.2 or 2.7.4
how to update to latest version in homeassistant?

jimklimov, 2023-04-19:
As far as I know - build one. Someone contributed this article: https://github.com/networkupstools/nut/wiki/Building-NUT-integration-for-Home-Assistant

Might help to raise the question on HA support forums, so maybe someone would issue a newer plugin

platini76, 2023-04-19:
maybe... i did not find the nut folder in my installation ahahhaha