networkupstools/nut #1919: [HCL] CyberPower GX1500U supported by `usbhid-ups$CP1500AVRLCD`

A9G-Data-Droid, 2023-04-21:
I am writing to confirm testing of my new UPS. It's the [CyberPower GX1500U](https://www.cyberpowersystems.com/product/ups/battery-backup/gx1500u/) "pure sinewave" model.

I connected it via USB to a system running TrueNAS-13.0-U4 Core and selected the driver option `usbhid-ups$CP1500AVRLCD` 

Then I unplugged the UPS from the wall and I saw a message:
`Apr 20 23:49:35 .local 1 2023-04-20T23:49:35.802130-07:00 .local upsmon 20359 - - UPS GX1500U@localhost:3493 on battery`

This tells me it is working!

## upsc
```                                                                                                                                                                                        
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 4750
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 27.5
battery.voltage.nominal: 24
device.mfr: CPS
device.model: GX1500U
device.serial: ?????????
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/ugen2.3
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: CyberPower HID 0.4
driver.version.internal: 0.41
input.voltage: 120.0
input.voltage.nominal: 120
output.voltage: 136.0
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 9
ups.mfr: CPS
ups.model: GX1500U
ups.productid: 0601
ups.realpower.nominal: 900
ups.serial: QA9MU7000577
ups.status: OL
ups.test.result: No test initiated
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
```

## upsrw
```
[battery.charge.low]
Description unavailable
Type: STRING
Maximum length: 10
Value: 10

[battery.runtime.low]
Description unavailable
Type: STRING
Maximum length: 10
Value: 300

[ups.delay.shutdown]
Description unavailable
Type: STRING
Maximum length: 10
Value: 20

[ups.delay.start]
Description unavailable
Type: STRING
Maximum length: 10
Value: 30
```

## upscmd -l
```
Instant commands supported on UPS [GX1500U]:

beeper.disable - Description unavailable
beeper.enable - Description unavailable
beeper.mute - Description unavailable
beeper.off - Description unavailable
beeper.on - Description unavailable
load.off - Description unavailable
load.off.delay - Description unavailable
load.on - Description unavailable
load.on.delay - Description unavailable
shutdown.return - Description unavailable
shutdown.stayoff - Description unavailable
shutdown.stop - Description unavailable
test.battery.start.deep - Description unavailable
test.battery.start.quick - Description unavailable
test.battery.stop - Description unavailable
```

jimklimov, 2023-04-21:
Thank you! This is a FreeBSD-based TrueNAS, correct?

A9G-Data-Droid, 2023-04-21:
Correct!

`freebsd-version 13.1-RELEASE-p7`