networkupstools/nut #1920: Extend driver-server socket protocol with verdicts

jimklimov, 2023-04-21:
Follow-up from research in issue #1914 per implementation experience in PR #1918; possibly a solution vector for bug #1901.

Currently `dstate::sock_arg()` in many cases quietly does:
````
/* The command was handled, status is a separate consideration */
return 1;
````

Which does not tell the direct client (`upsd`) or eventually the user's client (`upscmd`, `upsrw`...) how that handling ended up.

At least some (new) cases might instantly report whether they succeeded or not, similar to how `DATADUMP` handling ends with `DATAOK` (as opposed to `DATASTALE`), or `PING` ends with `PONG`, but currently do not due to lack of protocol keywords for that at the moment.

The return values from main/driver-specific handlers (per enums in `drivers/upshandler.h`) might be reported in a similar fashion at least on the socket protocol. Just to know the command was acknowledged and perhaps how it ended up.

In the bigger picture, may the socket protocol be treated as a private experience between current versions of `upsd` and NUT drivers - e.g. if someone built/scripted interactions around that, and those third-party tools fail due to unexpected key words, we commiserate but it is not a NUT problem as such?

To be clearer, this is not immediately a question about on-line NUT protocol (data server to clients), although I would not be surprised if it sprouts into that too (or practical handling of that, to more relevantly report OK/ERR for client requests).

CC @aquette @clepple @NUT-RogerPrice

gdt, 2023-04-21:
I think it's good to allow a reasonable path to maintaining a driver not in the main nut source tree, although that could be with the obligation to adapt to changes.   I thus mean with deprecation/etc. more than flag days.   Of course, complexity is bad too.

It sounds like the main issue is to add some sort of async "command is really done" after having an "in progress" return.   Once could either
  - document this now and then add it in 6 months, which would result in feeling righteous with no real gain
  - add it to nut git, and release it after a while, expecting private driver authors to adapt in the meantime
  - add a "enable async" command that drivers that understand the new way would issue at startup, as a transititon mechanism, perhaps with that being default and a no-op a year after the formal release of a nut version with it.
  -

jimklimov, 2023-04-21:
Thanks for help with the brainstorming session :)

Interesting point about out-of-tree drivers. I'd expect those to be built against NUT `dirvers/main.c` and `drivers/dstate.c` (of the files relevant to socket protocol handling - with `main.c` being a recent addition to such list in these recent PRs, adding "shared" commands/settings like changing debug verbosity on the fly).

As I found, every driver starting registers the instant command names (e.g. `load.off`) and queryable/settable values (e.g. `driver.debug`) when it connects to the server, so "private" drivers should not be impacted by the change whether they are built against NUT codebase (getting new binary methods linked) or are completely maintained out of tree and/or against older NUT and then only talk protocol to `upsd`.

It seems that at worst they would see unrecognized words and ignore them. For "real" clients of NUT online protocol, like `upsrw` and `upscmd`, the `upsd` facade would quickly not accept unrecognized targets which the driver did not announce. So that coupling was designed future-proof a long time ago, kudos to predecessors :)

It is more problematic however with requests to known targets that did not in fact succeed (like in the newly added `driver.killpower` command which relies on a safety toggle for the action to be actually permitted): the command tells its verdict quickly as an invalid situation, but that knowledge dissipates in `dstate::sock_arg()`. This is one of the framework bits I wondered about updating, to pass such immediately known verdicts on to caller. Perhaps as
````
ACK INSTCMD INVALID
````
- just a quick example OTOH, assuming an `ACK <OPERATION> <STATUS>` keyword gets added to generally complete the operations handled by driver over socket protocol, with STATUS including at least the existing enum values like HANDLED (successfully), UNKNOWN, INVALID, FAILED.

Note: Actually the "async" part is usable with `TRACKING` keyword, although I guess my new methods (and/or default handling of unsuccessful `INSTCMD`/`SET` operations?) should play better with it - some of my new experiments with them tended to exceed the timeout from `upscmd -w ...` and `upsrw -w ...` requests.

jimklimov, 2023-04-21:
The question actually appeared due to my testing of new features, where I connect to driver's socket and type commands like `NOBROADCAST` (to avoid noise irrelevant to testing) and have no idea if it was handled, until I see the stream of updateinfo's interrupted. Usually some update did come after I pressed ENTER, since sending a command to pipe is one thing, and queuing/handling it in driver is another (with remaining work in driver's processing loop still in flight), so it was just humanly confusing.

jimklimov, 2023-04-21:
And for yet more context: ultimately, some extension to the protocol would be needed for my new feature of `driver.reload` where the driver can detect that the config change requires actually a restart - and would tell so to the protocol caller ("I could reload just fine" vs. "Sorry, restart me"). Since on the face value this is an `INSTCMD`, I wanted to have (or add) a generic way for such responses.

Although *maybe* this is just where `TRACKING` could step in, to couple requests and replies, and I was just confused in the first place by the interactiveness of the `sockdebug` or `socat` I used. What I wrote above, with request posting and replies posting being two independent activities, seeing a response in the terminal is a "coincidence" of sorts, but not part of (what felt like) a dialogue.

Gotta sleep on it some more...

jimklimov, 2023-04-21:
Heh, so maybe proper `TRACKING` support was it (had an issue in my new implem):
````
:; ./clients/upscmd -u admin -p mypass -w UPS2@localhost:$NUT_PORT  driver.killpower
ERR INVALID-ARGUMENT

### On driver side:
 205.995575     [D3] sock_arg: TRACKING = F4F78A1F-D59B-4F21-82C6-9729871FA9DC
 205.995601     [D2] entering main_instcmd(driver.killpower, TRACKING) for [UPS2] on socket 4
 205.995605     Got socket protocol request for UPS [UPS2] to power off, but driver.flag.allow_killpower does notpermit this - request was currently ignored!
 205.995609     [D2] send_to_one: sending TRACKING F4F78A1F-D59B-4F21-82C6-9729871FA9DC 2
````

jimklimov, 2023-04-23:
UPDATE: Per later discussion and research, it seems that the `TRACKING` mechanism in NUT 2.8.0+ addresses most of this issue for `INSTCMD` and `SET` operations, so the clients can not just queue their request, but also wait and learn how it ended up. 

This does not address possible extension of protocol as seen to its client (`upsd`), e.g. adding keywords with acknowledgements for newly added operations like `(NO)BROADCAST` requests, but to keep the protocol simpler - those might be shoehorned into the same approach (adding an optional `TRACKING id`) set of parameters, and waiting for those.

closed 2023-04-24 by jimklimov