networkupstools/nut #1932: EAST EA-850U

CrewingTop, 2023-04-25:
Please add support for East UPS family. I have EAST EA-850U UPS, tried all drivers with NUT Home Assistant addon but unsuccessful. I have the latest version of HA and addon on Debian 11. UPS has applications for Windows and Linux. East UPS family is big and even supports bank systems in some of South America countries.

jimklimov, 2023-04-25:
According to #1670 there was some luck with "Megatec Qx" protocol family, so blazer* or newer nutdrv_qx; you may have to specify vendorid and productid (in case the older driver builds do not recognize EAST identifiers and so state it is not supported) and pick one of many subdrivers/protocols available.

See also NUT Github Wiki for contributed suggestions about building a HA plugin, maybe you can get a more modern NUT running that way.

pawelek-org, 2024-01-26:
In case if someone is struggling with EAST products like I was... here's my working configuration for EAST UPS1200-T-LI/LCD connected with my RPi5 using USB cable.

```
[ups]
driver=nutdrv_qx
vendorid=0001
productid=0000
desc="EAST UPS1200-T-LI/LCD"
langid_fix=0x409
protocol=megatec
subdriver=hunnox
novendor
noscanlangid
runtimecal=180,100,600,50
bus = 003
port = auto
```

- NUT v. 2.8.0-7 (installed via apt)
- Raspberry Pi OS Lite (Debian Bookworm arm64)

CrewingTop, 2024-01-26:
> In case if someone is struggling with EAST products like I was... here's my working configuration for EAST UPS1200-T-LI/LCD connected with my RPi5 using USB cable.

It's really useful.
I'm using NUT Addon.
With your assistance my addon configuration is following:
`- name: myups
  driver: nutdrv_qx
  port: auto
  config:
    - vendorid=0001
    - productid=0000
    - desc="EAST EA-850U"
    - langid_fix=0x409
    - protocol=megatec
    - subdriver=hunnox
    - novendor
    - noscanlangid`

Thank you!

closed 2024-01-26 by CrewingTop