networkupstools/nut #1934: Battery switch?

ohuu, 2023-05-04:
Sorry if this is the wrong place to ask this question, but is there a command to manually switch a UPS between battery and mains mode? If so which kinds of UPS would have this feature?
Thanks

jimklimov, 2023-05-06:
Typically that would be called "bypass", as used for maintenance on enterprisish UPSes - to unpower the internals so you can change batteries, etc.

Also depending on what exactly you are asking for, a self-testing or calibration mode can have this effect for most UPSes, although temporarily and/or until battery depletion.

I wonder also if your question implied "online" vs. "line-interactive", where some UPSes actually do switch inputs, and others always feed the load from inverter/battery a d there is no gap for a switch-over if wall-power is lost.

Metus88, 2023-06-13:
> I wonder also if your question implied "line-interactive" vs. "online", where some UPSes actually do switch inputs, and others always feed the load from inverter/battery and there is no gap for a switch-over if wall-power is lost.

Do you know if there is a way to switch between these modes? I have a Riello Sentinel pro 3000 that it has this possibility from the frontal panel but using driver "riello_usb" I can only see this commands:
```
/ # upscmd -l NAMEOFMYUPS
Instant commands supported on UPS [NAMEOFMYUPS]:

load.off - Turn off the load immediately
load.off.delay - Turn off the load with a delay (seconds)
load.on - Turn on the load immediately
load.on.delay - Turn on the load with a delay (seconds)
shutdown.return - Turn off the load and return when power is back
shutdown.stop - Stop a shutdown in progress
test.battery.start - Start a battery test
test.panel.start - Start testing the UPS panel
```
But nothing for switching between modes:
1 ECO (bypass)
2 SMART (automatic switch between ECO and ONLINE depending on quality of grid parameters)
3 ONLINE (always internal inverter active, no bypass)

I suppose that it is not possible (for now) but maybe you know a way. Thanks

jimklimov, 2023-06-13:
Interesting question. If you put it this way, some drivers do have the notion of ECO mode to report it, and maybe can toggle it. So systemically the idea is there - but might not be implemented in this particular driver and/or protocol.

If you're handy with sources and google-fu for docs, check if you can find a way to make it work...