networkupstools/nut #1935: Setting up NUT and starter options

SheepDomination, 2023-05-05:
I was reviewing the documentation although I don't know how to get this setup and running and what options are available over the manufacturers abysmal documentation for most UPS.

jimklimov, 2023-05-06:
Sadly, this is largely trial and error to see what your device supports (and NUT drivers in your version). Some vendors effectively ship very different things under same name and model, with different FW/HW revisions.

Several typical setup scenarios were documented pretty well two decades ago already (in docs, FAQ, config comments, NUT Config Examples book), like 1 PC + 1 UPS, several computers on same UPS, multi-PSU servers on several UPSes...

jimklimov, 2023-05-08:
Sorry for slow replies, i'm on a trip now with very occasional signal.

Not really sure about modern docs and how-to's. When I was figuring this out some 20+ years ago, almost before tge Internet, the docs in source tarball were all there was, and they sufficed. I don't think they got any less informative or detailed over the years... e.g. `nut-scanner` appeared since then to help draft first setups for a device.

They might remain techy, hacky and cryptic in spirit (assuming of reader readiness to experiment and investigate) though. PRs are welcome for a beginner-friendly how-to :)

SheepDomination, 2023-05-10:
What do you mean PRs are welcome for a beginner-friendly how-to ?

Would you by means of Team Viewer to help me to get NUT setup ?

jimklimov, 2023-05-10:
PRs are "pull requests".

NUT sources do include the setup documentation which sufficed for many readers over quarter of a century; nowadays there are also a Wiki, as well as many third-party blogs detailing their authors' experiences (and woes).

It is becoming evident that the original documentation made assumptions about the readers (who would be sysadmins or similar "geeks", ready to hack, investigate and experiment until success), and the current generation of new users wants something ready-made single-click; more so with NUT embedded as a plugin to some other systems and so not easily accessible by itself.

Nothing wrong with that desire per se, but somebody with that mindset should take the pile of knowledge already available and rearrange it in a manner understandable by "laymen", something like a Quick start guide (ideally compact for TL;DR readers who doze off after two minutes of reading, and thus referencing existing docs for nitty-gritty details for readers still awake). And post a pull request to the project to add a yet another text to the sources.

SheepDomination, 2023-05-11:
Does NUT have to be installed using Linux or can it be installed in Windows.
I've read the documentation although that is the first question as to how it should be installed ?

SheepDomination, 2023-05-12:
NUT is still being developed for Windows; are you running NUT and if so, how did you install. Start with that so I can get something up and running and what is the first command you suggest I should run ?

jimklimov, 2023-05-12:
Who were these questions directed to?

If to myself, I already wrote most of what I can say about this, from a developer perspective. I do not currently use NUT on Windows, just develop it there among other places.

If this question is a shout into the crowd, I guess it would take time for someone who does to notice and reply. Given the recent and experimental nature of that port, I guess there are not many people with hands-on experience so far.

Note that as said above, currently there is no installer, so...

Probably the first thing to run is a browser to find the tarball of binaries made from master branch (see above). Then the archiver to unpack it. Then notepad to edit configs. Then something to get libusb to handle the UPS, if it is an USB one.