networkupstools/nut #1942: immediate LOW BATTERY after power failure

cyayon, 2023-05-12:
Hello,

I use a Ellipse PRO 1600 connected viq USB on Linux (archlinux) and nut-2.8.0.

I just got a power failure and see that upsc return ups.status: LB immediately.
Which is strang because the UPS is loaded for about 30% and give about 1000s of runtime.

After the power has been restored, i have to restart nut-server to remove FSD flags.

Then i waited the battery.charge reach 100% and test again a power loss. 
The flag ups.status return to LB immediately (of course i bypassed the shutdown script). I wait a few minutes, the UPS seems to work as expected and i restored the power. 

Any idea please ?

thanks.

Here is the return of upsc command :
battery.charge: 88
battery.charge.low: 20
battery.runtime: 969
battery.type: PbAc
device.mfr: EATON
device.model: Ellipse PRO 1600
device.serial: xxxx
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.version: 2.8.0
driver.version.data: MGE HID 1.46
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.frequency: 50.0
input.transfer.high: 285
input.transfer.low: 165
input.voltage: 229.0
input.voltage.extended: no
outlet.1.desc: PowerShare Outlet 1
outlet.1.id: 2
outlet.1.status: on
outlet.1.switchable: no
outlet.2.desc: PowerShare Outlet 2
outlet.2.id: 3
outlet.2.status: on
outlet.2.switchable: no
outlet.desc: Main Outlet
outlet.id: 1
outlet.switchable: no
output.frequency: 50.0
output.frequency.nominal: 50
output.voltage: 230.0
output.voltage.nominal: 230
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.firmware: 01.16.0020
ups.load: 23
ups.mfr: EATON
ups.model: Ellipse PRO 1600
ups.power: 298
ups.power.nominal: 1600
ups.productid: ffff
ups.realpower: 232
ups.serial: P363G40H25
ups.status: OL
ups.timer.shutdown: -1
ups.timer.start: -1
ups.vendorid: 0463

jimklimov, 2023-05-12:
One idea that comes to mind is: how old is the battery, and what tech is it, and is it intensively used -- e.g. "online" UPS might always feed the load from battery, trading its lifetime for reliable (lack of) switch-over in an outage? Typical PbAc wear out after 2-3 years.

Another idea is if there is a test/calibration ability on that model? Maybe it underestimates the load?

Either way, I'm almost sure the LB status is reported by the device. Almost - because in some drivers and mappings NUT might interpret clues for simpler devices that do not say directly; I don't think this is the case here.

jimklimov, 2023-05-12:
CC @aquette @dzomaya

cyayon, 2023-05-14:
Hi,
I just replaced the batteries and everything is fine now.
thanks !

closed 2023-05-14 by cyayon

jimklimov, 2026-01-17:
As a data point: some years passed, and my own Eaton Ellipse ECO 650 got into a similar situation: during an outage, the computer quickly went down. Too quickly, I expected it to last 5-10 minutes based on (much) earlier experiments.

Debug-log investigation showed that the driver reported that the UPS was on battery, and a second later that it was low-battery, so `upsmon` started the shutdown which went well enough, up to `nutshutdown` systemd hook powering off the UPS. The reported `battery.charge` was above 90, but `ups.load` jumped from 19-23 to 37-40.

I re-ran the situation with `upsmon` disabled, and after a couple of minutes the UPS just turned off, last reporting 80-ish battery charge values. It seems that in fact the flag raised on USB HID layer is `SHUTDOWNIMM` (which should have also raised an alarm `Shutdown imminent` but for some reason did not, maybe a bug - seems due to be only handled at "full update" not "quick update"), and *that* is interpreted along with LOWBATT and TIMELIMITEXP as `status_set("LB")` (unless calibrating etc. when decisions are more complicated).

So somehow internally the UPS knows the battery health status (rather worn out, but holds up a minute or two), and despite the formal numbers being high, tells the host to shut down. I did not (at least knowingly) run any calibrations on the device, at least not in the past 2-3 years. The battery is about 3-4 years old (not reported), so not being in its prime condition is expected TBH.

`upslog` from the later experiment shows how I pulled the plug at approx 15:41:24, `ups.status` LB was raised a second later, and an `ALARM` appeared at 15:41:42; power was abruptly lost at 15:42:04 with a claimed 89% `battery.charge` (down from claimed 94% during those 40 seconds, not 100% as it was not fully charged from the initial blackout yet):
```
eco650@localhost:3493   20260117 154122 94 NA 24 [OL CHRG] NA NA
eco650@localhost:3493   20260117 154123 94 NA 24 [OL CHRG] NA NA
eco650@localhost:3493   20260117 154124 94 NA 24 [OB DISCHRG] NA NA
eco650@localhost:3493   20260117 154125 94 NA 24 [OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154126 94 NA 24 [OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154127 94 NA 24 [OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154128 94 NA 24 [OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154129 94 NA 24 [OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154130 94 NA 24 [OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154131 93 NA 24 [OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154132 93 NA 24 [OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154133 93 NA 24 [OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154134 93 NA 24 [OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154135 93 NA 24 [OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154136 93 NA 24 [OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154137 92 NA 24 [OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154138 92 NA 24 [OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154139 92 NA 24 [OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154140 92 NA 24 [OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154141 92 NA 24 [OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154142 92 NA 37 [ALARM OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154143 92 NA 37 [ALARM OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154144 92 NA 37 [ALARM OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154145 91 NA 37 [ALARM OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154146 91 NA 37 [ALARM OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154147 91 NA 37 [ALARM OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154148 91 NA 37 [ALARM OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154149 91 NA 37 [ALARM OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154150 91 NA 37 [ALARM OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154151 90 NA 37 [ALARM OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154152 90 NA 37 [ALARM OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154153 90 NA 37 [ALARM OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154154 90 NA 37 [ALARM OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154155 90 NA 37 [ALARM OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154156 90 NA 37 [ALARM OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154157 90 NA 37 [ALARM OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154158 90 NA 37 [ALARM OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154159 89 NA 37 [ALARM OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154200 89 NA 37 [ALARM OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154201 89 NA 37 [ALARM OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154202 89 NA 37 [ALARM OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154203 89 NA 37 [ALARM OB DISCHRG LB] NA NA
eco650@localhost:3493   20260117 154204 89 NA 37 [ALARM OB DISCHRG LB] NA NA
```

I did not have an `upsc` reading with the alarm report as I was fumbling under the table, but have a few before it:
* before test
```
:; date ; upsc eco650
Sat Jan 17 15:40:56 CET 2026

battery.charge: 94
battery.charge.low: 20
battery.runtime: 1057
battery.type: PbAc
device.mfr: EATON
device.model: Ellipse ECO 650
device.serial: 000000000
device.type: ups
driver.debug: 6
[... 1600 lines cut ...]
[... cut ...]
Jan 17 15:41:55 pve nut-driver@eco650[483779]: 2189.981550        [D2:483779] Path: UPS.PowerSummary.DelayBeforeStartup, Type: Feature, ReportID: 0x0a, Offset: 0, Size: 32, Value: -1
Jan 17 15:41:55 pve nut-driver@eco650[483779]: 2189.981556        [D5:483779] ups_infoval_set: report descriptor mapping for 'UPS.PowerSummary.DelayBeforeStartup' (Feature) was already set as handled
Jan 17 15:41:55 pve nut-driver@eco650[483779]: 2189.981566        [D5:483779] refresh_report_buffer: investigating path: 00840004.00840024.