networkupstools/nut #1947: Bus 001 Device 004: ID 0001:0000 Fry's Electronics

ilker-aktuna, 2023-05-24:
Hi,

I have a noname UPS which shows up like below on lsusb command:
"Bus 001 Device 004: ID 0001:0000 Fry's Electronics"

From what I read on different discussions, it might work with "nutdrv_qx" driver.
But I am not sure how I should configure this.

What I have now in config:

```
[servers]
driver = nutdrv_qx
port = auto
bus = "001"
desc = "Test UPS"
```

and the driver command shows : 

```
root@raspbx:~# /lib/nut/nutdrv_qx -DD -a servers
Network UPS Tools - Generic Q* USB/Serial driver 0.28 (2.7.4)
USB communication driver 0.33
   0.000000     debug level is '2'
   0.011074     upsdrv_initups...
   0.012757     Checking device (0001/0000) (001/004)
   0.014358     - VendorID: 0001
   0.014466     - ProductID: 0000
   0.014519     - Manufacturer: unknown
   0.014578     - Product: unknown
   0.014632     - Serial Number: unknown
   0.014686     - Bus: 001
   0.014747     - Device release number: 0100
   0.014801     Trying to match device
   0.014959     Device matches
   0.015290     qx_process_answer: short reply (input.voltage)
   0.015364     qx_process_answer: short reply (input.voltage)
   0.015434     qx_process_answer: short reply (input.voltage)
   0.015520     voltronic_qs_protocol: invalid protocol [M]
   0.015591     voltronic_qs_protocol: invalid protocol [M]
   0.015662     voltronic_qs_protocol: invalid protocol [M]
   0.015747     voltronic_qs_hex_protocol: invalid protocol [M]
   0.015818     voltronic_qs_hex_protocol: invalid protocol [M]
   0.015886     voltronic_qs_hex_protocol: invalid protocol [M]
   0.015965     qx_process_answer: short reply (input.voltage)
   0.016026     qx_process_answer: short reply (input.voltage)
   0.016081     qx_process_answer: short reply (input.voltage)
   0.016156     qx_process_answer: short reply (input.voltage)
   0.016222     qx_process_answer: short reply (input.voltage)
   0.016283     qx_process_answer: short reply (input.voltage)
   0.018161     qx_process_answer: short reply (input.voltage)
   0.019252     Checking device (0001/0000) (001/004)
   0.020768     - VendorID: 0001
   0.020836     - ProductID: 0000
   0.020893     - Manufacturer: unknown
   0.020948     - Product: unknown
   0.021007     - Serial Number: unknown
   0.021063     - Bus: 001
   0.021117     - Device release number: 0100
   0.021171     Trying to match device
   0.021251     Device matches
   0.022895     qx_process_answer: short reply (input.voltage)
   0.023880     Checking device (0001/0000) (001/004)
   0.025556     - VendorID: 0001
   0.025633     - ProductID: 0000
   0.025688     - Manufacturer: unknown
   0.025747     - Product: unknown
   0.025802     - Serial Number: unknown
   0.025856     - Bus: 001
[... 159 lines cut ...]
   0.013764     - VendorID: 0001
   0.013847     - ProductID: 0000
   0.013902     - Manufacturer: unknown
   0.013958     - Product: unknown
   0.014026     - Serial Number: unknown
   0.014081     - Bus: 001
   0.014136     - Device release number: 0100
   0.014192     Trying to match device
   0.014345     Device matches
   0.014564     Skipping protocol Voltronic 0.06
   0.014630     Skipping protocol Voltronic-QS 0.07
   0.014689     Skipping protocol Voltronic-QS-Hex 0.10
   0.014750     Skipping protocol Mustek 0.07
   0.014808     Skipping protocol Megatec/old 0.07
   0.014864     Skipping protocol BestUPS 0.06
   0.014934     Skipping protocol Mecer 0.07
   0.014993     Skipping protocol Megatec 0.06
   0.015052     Skipping protocol Zinto 0.06
   0.015108     Skipping protocol Q1 0.07
   0.015165     Device not supported!
   0.015229     Device not supported!
```

I am new to NUTS so I don't know how to configure and/or troubleshoot.
I'd appreciate any help.

ilker-aktuna, 2023-05-24:
Additional info:
The device comes with a software named UPSmart V1.5

jimklimov, 2023-05-24:
In the configuration, you would benefit from setting additionally
````
vendorid = 0001
productid = 0000
````
...which are checked first generally.

Matching by `bus` and `device` numbers happens later in the stack (and `device` matching in particular was broken until after NUT 2.8.0 was released, and you are on old 2.7.4 - check if your distro offers the newer version BTW).

Looking how it returns "unknown" for the basic fields, it seems that the NUT driver's run-time account (and libusb as part of the driver) can not access the device. Possibly kernel has grabbed it and won't give back, try finding and updating the `udev.rules` file on your system to set up the hand-off to NUT.

While you experiment, you can also try running `nutdrv_qx -u root ...` (starting as `root`) to not drop privileges. This would not fully help if some other process (or kernel) holds the device exclusively, but may work around some basic filesystem privileges for USB device nodes (if the run-time user is mismatched, but nobody else is trying to hold the device).

Also make sure another copy of a NUT driver (e.g. via init-scripts or systemd units) is not already holding the device. There can be only one... exclusive owner at a time :)

ilker-aktuna, 2023-05-24:
thanks.
added vendorid and productid now but that did not change the result.
I also tried with "-u root" parameter , also no good result from there :(

I am using raspbian (buster) and apt only provides this version.
How can I get 2.8.0 ?

and how can I update udev.rules ?

ilker-aktuna, 2023-05-24:
root@raspbx:~# ps -ef | grep nut
root      6137  1356  0 19:39 pts/0    00:00:00 grep nut
root@raspbx:~#

ilker-aktuna, 2023-05-24:
I just tested with the included Windows software (UPSmart) 
When I select connection type Mega(USB) , it conencts and displays correct load and voltage values.
So the USB port provides data.

How can I get 2.8.0 ?
ANd how can I check udev.rules ?

ilker-aktuna, 2023-05-24:
> Matching by `bus` and `device` numbers happens later in the stack (and `device` matching in particular was broken until after NUT 2.8.0 was released, and you are on old 2.7.4 - check if your distro offers the newer version BTW).

just compiled from source now. (didn't install, running from the source folder)
result is  similar:

`root@raspbx:~/nut-master# ./drivers/nutdrv_qx  -DD -a servers -u root
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0.1)
USB communication driver (libusb 1.0) 0.45
   0.000001     [D1] Built-in default or configured user for drivers 'nobody' was ignored due to 'root' specified on command line
   0.000134     [D1] Network UPS Tools version 2.8.0.1 (release/snapshot of 2.8.0.1) built with gcc (Raspbian 8.3.0-6+rpi1) 8.3.0 and configured all by default guesswork
   0.000197     [D1] debug level is '2'
   0.010741     [D1] Succeeded to become_user(root): now UID=0 GID=0
   0.011201     [D1] upsdrv_initups...
   0.029130     [D2] Checking device 1 of 4 (0001/0000)
   0.031015     [D1] nut_libusb_open get iProduct failed, retrying...
   0.032668     [D1] nut_libusb_open get iProduct failed, retrying...
   0.034355     [D1] nut_libusb_open get iProduct failed, retrying...
   0.034455     [D2] - VendorID: 0001
   0.034507     [D2] - ProductID: 0000
   0.034558     [D2] - Manufacturer: unknown
   0.034608     [D2] - Product: unknown
   0.034658     [D2] - Serial Number: unknown
   0.034708     [D2] - Bus: 001
   0.034757     [D2] - Device: 011
   0.034807     [D2] - Device release number: 0100
   0.034857     [D2] Trying to match device
   0.035146     [D2] Device matches
   0.035201     [D2] Reading first configuration descriptor
   0.035427     [D2] Claimed interface 0 successfully
   0.035678     [D2] qx_process_answer: short reply (input.voltage)
   0.035758     [D2] qx_process_answer: short reply (input.voltage)
   0.035828     [D2] qx_process_answer: short reply (input.voltage)
   0.035933     [D2] voltronic_qs_protocol: invalid protocol [M]
   0.036003     [D2] voltronic_qs_protocol: invalid protocol [M]
   0.036069     [D2] voltronic_qs_protocol: invalid protocol [M]
   0.036155     [D2] voltronic_qs_hex_protocol: invalid protocol [M]
   0.036223     [D2] voltronic_qs_hex_protocol: invalid protocol [M]
   0.036290     [D2] voltronic_qs_hex_protocol: invalid protocol [M]
   0.036376     [D2] qx_process_answer: short reply (input.voltage)
   0.036443     [D2] qx_process_answer: short reply (input.voltage)
   0.036507     [D2] qx_process_answer: short reply (input.voltage)
   0.036589     [D2] qx_process_answer: short reply (input.voltage)
   0.036657     [D2] qx_process_answer: short reply (input.voltage)
   0.036722     [D2] qx_process_answer: short reply (input.voltage)
   0.055908     [D2] Checking device 1 of 4 (0001/0000)
   0.057805     [D1] nut_libusb_open get iProduct failed, retrying...
   0.059531     [D1] nut_libusb_open get iProduct failed, retrying...
   0.061150     [D1] nut_libusb_open get iProduct failed, retrying...
   0.061344     [D2] - VendorID: 0001
   0.061518     [D2] - ProductID: 0000
   0.061691     [D2] - Manufacturer: unknown
   0.061862     [D2] - Product: unknown
   0.062034     [D2] - Serial Number: unknown
   0.062205     [D2] - Bus: 001
   0.062376     [D2] - Device: 011
   0.062546     [D2] - Device release number: 0100
   0.062720     [D2] Trying to match device
   0.062937     [D2] Device matches
   0.063110     [D2] Reading first configuration descriptor
   0.063409     [D2] Claimed interface 0 successfully
   0.082129     [D2] Checking device 1 of 4 (0001/0000)
   0.084035     [D1] nut_libusb_open get iProduct failed, retrying...
   0.085720     [D1] nut_libusb_open get iProduct failed, retrying...
   0.087368     [D1] nut_libusb_open get iProduct failed, retrying...
   0.087571     [D2] - VendorID: 0001
   0.087747     [D2] - ProductID: 0000
   0.087919     [D2] - Manufacturer: unknown
   0.088090     [D2] - Product: unknown
   0.088263     [D2] - Serial Number: unknown
   0.088434     [D2] - Bus: 001
[... 286 lines cut ...]
   0.600461     [D1] nut_libusb_open get iProduct failed, retrying...
   0.602501     [D1] nut_libusb_open get iProduct failed, retrying...
   0.602600     [D2] - VendorID: 0001
   0.602660     [D2] - ProductID: 0000
   0.602720     [D2] - Manufacturer: unknown
   0.602778     [D2] - Product: unknown
   0.602837     [D2] - Serial Number: unknown
   0.602897     [D2] - Bus: 001
   0.602957     [D2] - Device: 011
   0.603016     [D2] - Device release number: 0100
   0.603075     [D2] Trying to match device
   0.603185     [D2] Device matches
   0.603248     [D2] Reading first configuration descriptor
   0.603440     [D2] Claimed interface 0 successfully
   0.605659     Device not supported!
   0.605757     Device not supported!`

I also copied nut-usbups.rules to /etc/udev/rules.d/62-nut-usbups.rules as described in:
https://fossies.org/linux/nut/scripts/udev/README

srefreshed the bus : 
udevadm trigger --subsystem-match=usb_device

but none of these changed the result.
what else should I try ?

ilker-aktuna, 2023-05-24:
installed 2.8.0 and run "nut_scanner"

says:
```
root@raspbx:~/nut-master#  /usr/local/ups/bin/nut-scanner 
Scanning USB bus.
No start IP, skipping NUT bus (old connect method)
[nutdev1]
        driver = "nutdrv_qx"
        port = "auto"
        vendorid = "0001"
        productid = "0000"
        bus = "001"
        device = "004"
        ###NOTMATCHED-YET###bcdDevice = "0100"
```

output is:

```
root@raspbx:~/nut-master#  /usr/local/ups/bin/nutdrv_qx -DD -a nutdev1 -u root
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0.1)
USB communication driver (libusb 1.0) 0.45
   0.000001     [D1] Built-in default or configured user for drivers 'nobody' was ignored due to 'root' specified on command line
   0.000165     [D1] Network UPS Tools version 2.8.0.1 (release/snapshot of 2.8.0.1) built with gcc (Raspbian 8.3.0-6+rpi1) 8.3.0 and configured all by default guesswork
   0.000225     [D1] debug level is '2'
   0.012433     [D1] Succeeded to become_user(root): now UID=0 GID=0
   0.012671     [D1] upsdrv_initups...
   0.029671     [D2] Checking device 1 of 4 (0001/0000)
   0.031603     [D1] nut_libusb_open get iProduct failed, retrying...
   0.033260     [D1] nut_libusb_open get iProduct failed, retrying...
   0.034988     [D1] nut_libusb_open get iProduct failed, retrying...
   0.035167     [D2] - VendorID: 0001
   0.035316     [D2] - ProductID: 0000
   0.035769     [D2] - Manufacturer: unknown
   0.035941     [D2] - Product: unknown
   0.036174     [D2] - Serial Number: unknown
   0.036433     [D2] - Bus: 001
   0.036581     [D2] - Device: 004
   0.036845     [D2] - Device release number: 0100
   0.037189     [D2] Trying to match device
   0.037738     [D2] Device matches
   0.037927     [D2] Reading first configuration descriptor
   0.038269     [D2] Claimed interface 0 successfully
   0.038842     [D2] qx_process_answer: short reply (input.voltage)
   0.039050     [D2] qx_process_answer: short reply (input.voltage)
   0.039332     [D2] qx_process_answer: short reply (input.voltage)
   0.039662     [D2] voltronic_qs_protocol: invalid protocol [M]
   0.040081     [D2] voltronic_qs_protocol: invalid protocol [M]
   0.040245     [D2] voltronic_qs_protocol: invalid protocol [M]
   0.040452     [D2] voltronic_qs_hex_protocol: invalid protocol [M]
   0.040700     [D2] voltronic_qs_hex_protocol: invalid protocol [M]
   0.040865     [D2] voltronic_qs_hex_protocol: invalid protocol [M]
   0.041243     [D2] qx_process_answer: short reply (input.voltage)
   0.041415     [D2] qx_process_answer: short reply (input.voltage)
   0.041810     [D2] qx_process_answer: short reply (input.voltage)
   0.042009     [D2] qx_process_answer: short reply (input.voltage)
   0.042170     [D2] qx_process_answer: short reply (input.voltage)
   0.042643     [D2] qx_process_answer: short reply (input.voltage)
   0.060405     [D2] Checking device 1 of 4 (0001/0000)
   0.062280     [D1] nut_libusb_open get iProduct failed, retrying...
   0.063977     [D1] nut_libusb_open get iProduct failed, retrying...
   0.065575     [D1] nut_libusb_open get iProduct failed, retrying...
   0.065752     [D2] - VendorID: 0001
   0.065923     [D2] - ProductID: 0000
   0.066188     [D2] - Manufacturer: unknown
   0.066582     [D2] - Product: unknown
   0.066781     [D2] - Serial Number: unknown
   0.066948     [D2] - Bus: 001
   0.067180     [D2] - Device: 004
   0.067285     [D2] - Device release number: 0100
   0.067643     [D2] Trying to match device
   0.067854     [D2] Device matches
   0.068174     [D2] Reading first configuration descriptor
   0.068442     [D2] Claimed interface 0 successfully
```

still no luck.
I've compiled from source, so I if there is any mod I could try on source, I can.
Any suggestions ?

ilker-aktuna, 2023-05-24:
it worked with this config:

```
[PowerCool]
        driver="nutdrv_qx"
        subdriver="hunnox"
        desc="PowerCool 1200"
        port="auto"
        vendorid="0001"
        productid="0000"
        bus="001"
        protocol="hunnox"
        langid_fix=0x0409
        novendor
        norating
        noscanlangid

```

now the driver output is like :

```
root@raspbx:~#  /usr/local/ups/bin/nutdrv_qx -DD -a PowerCool -u root
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0.1)
USB communication driver (libusb 1.0) 0.45
   0.000001     [D1] Built-in default or configured user for drivers 'nobody' was ignored due to 'root' specified on command line
   0.000146     [D1] Network UPS Tools version 2.8.0.1 (release/snapshot of 2.8.0.1) built with gcc (Raspbian 8.3.0-6+rpi1) 8.3.0 and configured all by default guesswork
   0.000220     [D1] debug level is '2'
   0.025720     [D1] Succeeded to become_user(root): now UID=0 GID=0
   0.025963     [D1] upsdrv_initups...
   0.026089     [D2] Language ID workaround enabled (using '0x409')
   0.047926     [D2] Checking device 1 of 4 (0001/0000)
   0.049917     [D1] nut_libusb_open get iProduct failed, retrying...
   0.051750     [D1] nut_libusb_open get iProduct failed, retrying...
   0.053619     [D1] nut_libusb_open get iProduct failed, retrying...
   0.053871     [D2] - VendorID: 0001
   0.054049     [D2] - ProductID: 0000
   0.054224     [D2] - Manufacturer: unknown
   0.054397     [D2] - Product: unknown
   0.054571     [D2] - Serial Number: unknown
   0.054744     [D2] - Bus: 001
   0.054918     [D2] - Device: 004
   0.055095     [D2] - Device release number: 0100
   0.055273     [D2] Trying to match device
   0.055840     [D2] Device matches
   0.056076     [D2] Reading first configuration descriptor
   0.056343     [D2] successfully set kernel driver auto-detach flag
   0.058267     [D2] Claimed interface 0 successfully
   0.060650     [D2] Skipping protocol Voltronic 0.06
   0.060991     [D2] Skipping protocol Voltronic-QS 0.08
   0.061256     [D2] Skipping protocol Voltronic-QS-Hex 0.10
   0.061458     [D2] Skipping protocol Mustek 0.07
   0.062801     [D2] Skipping protocol Megatec/old 0.07
   0.065188     [D2] Skipping protocol BestUPS 0.06
   0.065421     [D2] Skipping protocol Mecer 0.07
   0.065629     [D2] Skipping protocol Megatec 0.06
   0.065808     [D2] Skipping protocol Zinto 0.06
   0.065989     [D2] Skipping protocol Masterguard 0.02
   0.104245     Using protocol: Hunnox 0.01
   0.104383     [D2] blazer_initups: skipping input.voltage.nominal
   0.104463     [D2] blazer_initups: skipping input.current.nominal
   0.104530     [D2] blazer_initups: skipping battery.voltage.nominal
   0.104592     [D2] blazer_initups: skipping input.frequency.nominal
   0.104669     [D2] blazer_initups: skipping device.mfr
   0.104731     [D2] blazer_initups: skipping device.model
   0.104791     [D2] blazer_initups: skipping ups.firmware
   0.104919     [D1] upsdrv_initinfo...
   0.110175     Can't autodetect number of battery packs [-1/26.90]
   0.110297     Battery runtime will not be calculated (runtimecal not set)
   0.110419     [D1] upsdrv_updateinfo...
   0.110487     [D1] Quick update...
   0.113439     [D2] dstate_init: sock /var/state/ups/nutdrv_qx-PowerCool open on fd 11
   0.113971     [D1] Group and/or user account for this driver was customized ('root:nogroup') compared to built-in defaults. Fixing socket '/var/state/ups/nutdrv_qx-PowerCool' ownership/access.
   0.114210     [D1] Group access for this driver successfully fixed
   0.114311     Running as foreground process, not saving a PID file
   0.114419     [D1] Driver initialization completed, beginning regular infinite loop
   0.114485     [D1] upsdrv_updateinfo...
   0.114541     [D1] Quick update...
   2.115419     [D1] upsdrv_updateinfo...
   2.115551     [D1] Quick update...
   4.115989     [D1] upsdrv_updateinfo...
   4.116131     [D1] Quick update...
   6.118097     [D1] upsdrv_updateinfo...
   6.118250     [D1] Quick update...
^C   7.941913   [D1] set_exit_flag: raising exit flag due to signal 2
   7.942071     Signal 2: exiting
   7.942217     [D1] upsdrv_cleanup...
root@raspbx:~# 
```

AND , upsd can be run, upsc can read from it:

```
root@raspbx:~# nut-master/server/upsd 
Network UPS Tools upsd 2.8.0.1
fopen /var/state/ups/upsd.pid: No such file or directory
Could not find PID file '/var/state/ups/upsd.pid' to see if previous upsd instance is already running!
/usr/local/ups/etc/upsd.conf is world readable
listening on 127.0.0.1 port 3493
listening on ::1 port 3493
Connected to UPS [PowerCool]: nutdrv_qx-PowerCool
Found 1 UPS defined in ups.conf
/usr/local/ups/etc/upsd.users is world readable
root@raspbx:~# 
root@raspbx:~# /usr/local/ups/bin/upsc PowerCool@localhost
battery.voltage: 26.90
device.type: ups
driver.debug: 2
driver.flag.allow_killpower: 0
driver.flag.norating: enabled
driver.flag.noscanlangid: enabled
driver.flag.novendor: enabled
driver.name: nutdrv_qx
driver.parameter.bus: 001
driver.parameter.langid_fix: 0x0409
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0000
driver.parameter.protocol: hunnox
driver.parameter.subdriver: hunnox
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0001
driver.state: quiet
driver.version: 2.8.0.1
driver.version.data: Hunnox 0.01
driver.version.internal: 0.32
driver.version.usb: libusb-1.0.22 (API: 0x1000106)
input.frequency: 50.0
input.voltage: 234.0
input.voltage.fault: 0.0
output.voltage: 234.0
ups.beeper.status: disabled
ups.delay.shutdown: 60
ups.delay.start: 0
ups.load: 0
ups.productid: 0000
ups.status: OL
ups.temperature: 29.0
ups.type: offline / line interactive
ups.vendorid: 0001
root@raspbx:~# 
```

But, since I compiled from source, I don't know how to setup.

What components should be run for the server to start at system boot ?
How can I make sure it will run driver & server when I restart ?

jimklimov, 2023-05-24:
I'm a bit surprised how it can't find the product name and other USB metadata, but sees reasonable info from it. Every day for us, something new...

Regarding installation,  https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests suggests a reasonable approach to getting systemd units enabled. You may want to re-configure and rebuild first, at least to use better user accounts than root or nobody by default.

ilker-aktuna, 2023-05-25:
systemd is installed and works fine now.
I won't recompile. I've already spent a lot of time on this. now that it works, I am happy.

thanks for all the help.

ilker-aktuna, 2023-05-25:
btw, the UPS brand is Makelsan. It is a local company in Turkey.
If anyone from Turkey wants to make it work with NUT, this thread might be helpful.

jimklimov, 2023-05-25:
Ok, so I misunderstood your earlier post? It is not a Powercool, just compatible? Or same OEM model, different sales brand?

jimklimov, 2023-05-25:
FWIW, to mark it better in HCL/DDL - can you please post what is the sold product fully called? :)

ilker-aktuna, 2023-05-25:
It is not PowerCool. I copied powercool config from another post.
The brand is Makelsan and they have a UPS series named Lion
So it is Makelsan Lion 1200VA
and I've also ordered a Makelsan Lion 650VA
Hopefully they have the same USB processor.

ilker-aktuna, 2023-05-27:
with this driver it generally works but there is no battery.runtime in the output.
Can I do something about it , or is it that the UPS is not providing that value ?

jimklimov, 2023-05-27:
Many devices indeed do not offer that information (popularly also load or charge level are often missing), so that may well be just absent completely. If vendor software does provide that, it might be interesting to somehow sniff and debug and find out how it does query (or estimate) that.

Some NUT drivers calculate that information artificially, if some other data is available, and `nutdrv_qx` is among them. Check the documentation for (IIRC) `runtimecal` settings. Note also the currently open issues #1279 and #1652 in this area.

ilker-aktuna, 2023-05-28:
I have an interesting issue.
I compiled NUT from source on another raspberry PI and connected another UPS of same model/brand.
BUt on the second server, nutdrv_qx complains about configuration parameter:
```

root@octopi:~/nut-master# /usr/local/ups/bin/nutdrv_qx  -DD -a Makelsan
Network UPS Tools - Generic Q* Serial driver 0.32 (2.8.0.1)

Fatal error: 'subdriver' is not a valid variable name for this driver.

Look in the man page or call this driver with -h for a list of
valid variable names and flags.
root@octopi:~/nut-master# 

```

The difference is the "Generic Q* Serial driver 0.32"
this one reports 0.32 , where the first server reports 0.45

what is this , and how can I make it 0.45 ?

jimklimov, 2023-05-28:
Actually the `nutdrv_qx` driver did report `0.32` in your post above, and it is the current version for about a year. There were a few wholesale changes of all drivers (#1767, #1923) so I guess a bump is warranted, but it is irrelevant for your `subdriver` question :\

The `0.45` is the `libusb1.c` connector variant version to handle USB devices.

Looking at https://github.com/networkupstools/nut/blob/master/drivers/nutdrv_qx.c#LL2681C1-L2682C72 I guess your other system did not include a `libusb*dev` package or otherwise (explicitly?) did not elect to use `libusb`, so only built this driver for serial connections:
````
#ifdef QX_USB
	addvar(VAR_VALUE, "subdriver", "Serial-over-USB subdriver selection");
````

Check with `docs/config-prereqs.txt` for deppendencies that may apply to your (Debian/Ubuntu-like?) distro.

ilker-aktuna, 2023-05-28:
I'm away from home now so I will check the prereqs later.

I want to make sure I understand correct. You mean, 0.32 is the correct version but my first PI missed something and compiled/used another variant of the driver. As a result , it reports 0.45 version which worked for me.
But the other PI has all prereqs and compiles the correct driver. And this one does not have the "subdriver" config parameter.

now I am confused.
How shall I make my second PI use the "wrong" driver (0.45) so that it works with my UPS ?

jimklimov, 2023-05-29:
No, not quite so. It has the `nutdrv_qx` driver 0.32 on both systems. This driver can use serial and/or USB media with devices that talk variants of Megatec Qx protocols (x being a number - so generalization of names like Q1 or Q5).

When libusb (library and development files like headers and pkgconf manifest) is present on the build system, that layer is compiled into the main driver and opens up both USB ability and thr device-specific subdriver (how is serial protocol smuggled in USB packets here).

ilker-aktuna, 2023-05-29:
weird.

Both systems have same version of libusb.

root@octopi:~/nut-master# apt list --installed | grep libusb

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libusb-0.1-4/oldstable,now 2:0.1.12-32 armhf [installed,automatic]
libusb-1.0-0/oldstable,now 2:1.0.22-2 armhf [installed]
libusbmuxd4/oldstable,now 1.1.0~git20181007.07a493a-1 armhf [installed,automatic]
root@octopi:~/nut-master#

root@raspbx:~# apt list --installed | grep libusb

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libusb-0.1-4/oldstable,now 2:0.1.12-32 armhf [installed]
libusb-1.0-0-dev/oldstable,now 2:1.0.22-2 armhf [installed]
libusb-1.0-0/oldstable,now 2:1.0.22-2 armhf [installed]
libusb-1.0-doc/oldstable,now 2:1.0.22-2 all [installed,automatic]
libusbmuxd4/oldstable,now 1.1.0~git20181007.07a493a-1 armhf [installed,automatic]
root@raspbx:~# 

The difference is libusd-dev package.
I would expect autogen or configure commands to report what is missing...

ilker-aktuna, 2023-05-29:
OK. installed libusb-dev and now it works. detects the UPS and gets info.

However, I have another issue when installing the service.

```
root@octopi:~/nut-master# /usr/local/ups/sbin/upsdrvctl start Makelsan
Network UPS Tools - UPS driver controller 2.8.0.1
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0.1)
USB communication driver (libusb 1.0) 0.45
libusb1: Could not open any HID devices: insufficient permissions on everything
No supported devices found. Please check your device availability with 'lsusb'
and make sure you have an up-to-date version of NUT. If this does not help,
try running the driver with at least 'subdriver', 'vendorid' and 'productid'
options specified. Please refer to the man page for details about these options
(man 8 nutdrv_qx).

Driver failed to start (exit status=1)
```

also "/usr/local/ups/bin/nutdrv_qx -DD -a Makelsan" says "Access denied (insufficient permissions)"
I can only run it with "-u root" parameter successfully.
What is the solution to this ?

jimklimov, 2023-05-29:
> I would expect autogen or configure commands to report what is missing...

I would expect `configure` to have reported what it found and set up, detailed in the wall of yext and summarized in the end. Seems you configured by default (without hard requirements like `--with-all` or detailed by driver family), so it finds what it can and skips the rest. With a requirement, it would crash upon trying to find the build dependency (headers and all). There is a gazillion ways to provide that, depending on OS and build circumstances, so it is up to human expertize to find why it failed and pick the best resolution to the issue.

autogen is offtopic here, it typically makes a `configure` script tailored by your distro/version of autotools. In case of NUT does a bit more generation in similar areas.

As for perms - did you `configure --with-user=...`? Does it exist? Was udev restarted after installation of hwdb file? All the nits packaging for particular distro does...

ilker-aktuna, 2023-05-30:
:(
I had skipped the "configure --with-user=..." part.
Now I configured and built wtih that parameter and it works.
Thanks a lot.

jackmaninov, 2023-06-22:
I can also confirm a Makelsan Lion 850VA working with this configuration. The manufacturer's listing of the product line is here, for future inclusion in the compatibility list: https://www.makelsan.com.tr/en/makelsan-ups-kgk-urundetay.php?urunID=500

ilker-aktuna, 2023-06-22:
I also got a Makelsan 650VA. It did not work with this configuration. I had to change the config a little. 
I am not able to reach the config now , I'll post when I'm home.

RecNes, 2023-12-09:
Got a Lion 2200VA recently. I've to setup an SNMP to manage shut down the NAS properly in case of power outage. I hope your adventure will help me. Thanks for the detailed comments.

RecNes, 2023-12-19:
Hey there!

I managed to configure my Makelsan Lion 2200VA USB UPS with this configuration I found [here](https://github.com/networkupstools/nut/issues/1836#issue-1552982597)

```
[makelsanlion2200]
        driver="nutdrv_qx"
        port=auto
        vendorid="0001"
        productid="0000"
        langid_fix=0x409
        protocol="megatec"
        subdriver="snr"
        novendor
        noscanlangid
        runtimecal=180,100,540,50
```

I changed subdriver from hunnox to snr to try to find anything will be changed, but the change is not affected to anything.

Readings looks like normal:
[image] 

But, `/lib/nut/nutdrv_qx -a makelsanlion2200 -DDD` command output still doesn't feels quite right:

```
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000	[D1] debug level is '3'
   0.009481	[D1] upsdrv_initups...
   0.010006	[D2] Language ID workaround enabled (using '0x409')
   0.195482	[D2] Checking device 1 of 7 (09DA/054F)
   0.196051	[D1] Failed to open device (09DA/054F), skipping: Access denied (insufficient permissions)
   0.196150	[D2] Checking device 2 of 7 (1D6B/0001)
   0.196257	[D1] Failed to open device (1D6B/0001), skipping: Access denied (insufficient permissions)
   0.196300	[D2] Checking device 3 of 7 (1D6B/0002)
   0.196374	[D1] Failed to open device (1D6B/0002), skipping: Access denied (insufficient permissions)
   0.196415	[D2] Checking device 4 of 7 (0001/0000)
   0.199184	[D1] nut_libusb_open get iProduct failed, retrying...
   0.202141	[D1] nut_libusb_open get iProduct failed, retrying...
   0.205418	[D1] nut_libusb_open get iProduct failed, retrying...
   0.205510	[D2] - VendorID: 0001
   0.205588	[D2] - ProductID: 0000
   0.205632	[D2] - Manufacturer: unknown
   0.205673	[D2] - Product: unknown
   0.205713	[D2] - Serial Number: unknown
   0.205746	[D2] - Bus: 004
   0.205781	[D2] - Device: unknown
   0.205823	[D2] - Device release number: 0100
   0.205863	[D2] Trying to match device
   0.205897	[D3] match_function_regex: matching a device...
   0.206428	[D2] Device matches
   0.206539	[D2] Reading first configuration descriptor
   0.206616	[D3] libusb_kernel_driver_active() returned 0
   0.206659	[D2] failed to claim USB device: Resource busy
   0.206731	[D2] Kernel driver already detached
   0.206775	[D2] failed to claim USB device: Resource busy
   0.206825	[D2] Kernel driver already detached
   0.206866	[D2] failed to claim USB device: Resource busy
   0.206906	[D2] Kernel driver already detached
   0.206949	[D2] failed to claim USB device: Resource busy
   0.206986	[D2] Kernel driver already detached
   0.207030	Can't claim USB device [0001:0000]@0/0: Entity not found
```

So, with this case, do I need to change anything to work things better?

jimklimov, 2023-12-20:
@RecNes : congratulations, I guess. How did you run the NUT driver for the first screenshots (with working graphs) and later (with access errors)? I feel the former was as a service (so as the user account who may access the USB device node, per udev/upower/... config), and the latter you tried as "yourself" and could not see the USB port. And/or some other program/driver-instance/kernel/hypervisor/... held the port at that time. Does this guess fit?

msurg, 2023-12-20:
@RecNes : 

`runtimecal=180,100,540,50`

How did you specify those values?

RecNes, 2023-12-20:
> congratulations, I guess. How did you run the NUT driver for the first screenshots (with working graphs) and later (with access errors)? I feel the former was as a service (so as the user account [...]

After identifying the correct driver and defining users, I just make services to be restarted, Then I completed the necessary configurations by gathered from various sites, installing nginx, ensuring CGI security, etc.

Here is full configuration and action steps:

- Added line below to /etc/nut/hosts.conf:

`MONITOR makelsanlion2200@localhost "Local Makelsan Lion 2200VA UPS"`

- Changed mode of the nut by changing the line below in /etc/nut/nut.conf file:

`MODE=netserver`

- Added UPS configuration to /etc/nut/ups.conf file:

```
[makelsanlion2200]
	driver="nutdrv_qx"
	port=auto
	vendorid="0001"
	productid="0000"
	langid_fix=0x409
	protocol="megatec"
	subdriver="snr"
	novendor
	noscanlangid
	runtimecal=180,100,540,50
	desc="Makelsan Lion 2200VA"
```
- Make reachable the upsd service from the network by modifying the IP configuration in /etc/nut/upsd.conf at the line below
(this might be unnecessary):

`LISTEN 0.0.0.0 3493` 

- Added a user to /etc/nut/upsd.users file:

```
[upsmon]
        password = xpasswdx
        upsmon primary
```

- Added this line to /etc/nut/upsmon.conf:

`MONITOR makelsanlion2200@localhost 1 upsmon xpasswdx primary`

- Uncomment the `I_HAVE_SECURED_MY_CGI_DIRECTORY` in /etc/nut/upsset.conf

- Installed nginx, nut-cgi and fcgiwrap

`apt install nginx nut-cgi fcgiwrap`

- Configured nginx defult configuration file to make nginx serve the nut html by adding those lines into /etc/nginx/sites-available/default:

```
        location /nut {
        alias /usr/share/nut/www/;
        try_files $uri $uri/ /index.html;
        }

        location /cgi-bin/ {
                gzip off;
                root /usr/lib;

                include fastcgi_params;
                fastcgi_pass unix:/var/run/fcgiwrap.socket;
                fastcgi_param SCRIPT_FILENAME  $document_root$fastcgi_script_name;
        }
```

When it's all ready I just restarted the services:

```
systemctl restart nut-server
systemctl restart nut-monitor
systemctl restart fcgiwrap
systemctl restart nginx
```

- Then just went to the addres via browser and login with the credentials defined in /etc/nut/upsd.users file:

`http://x.x.x.x/nut/`

That's how I get the screenshot I posted in previous message. While I'm seeing the UPS data on browser the `/lib/nut/nutdrv_qx -a makelsanlion2200 -DDD` command gives me the headache :)

RecNes, 2023-12-20:
> @RecNes :
> 
> `runtimecal=180,100,540,50`
> 
> How did you specify those values?

I didn't. I found the whole driver configuration from another topic in here and just used. Link is in my previous message.

jimklimov, 2023-12-21:
Check the definition in man page about the two couples of numbers and what they mean ;)

RecNes, 2023-12-21:
> Check the definition in man page about the two couples of numbers and what they mean ;)

If you are pointing my reply, I'm trying to say, that line might have no function for "my configuration". Not the other way. I think, if I remove that line, the system will still work though.

Also you're right, I have to check the mam page.

msurg, 2023-12-23:
> Check the definition in man page about the two couples of numbers and what they mean ;)

I will. Thank you

binarydepth, 2024-03-20:
My model is R-Smart 1510 from CDP. I have run sucesfully with Megatec and Lagfix but lately it fails to run on boot and session login. Manajro KDE. And below is my config. I have tried all sub drivers on man 8 blazer_usb: phoenix, ippon, cypress, and krauler.

```
[R-Smart_1510]
        driver="blazer_usb"
#        subdriver="krauler"
        desc="R-Smart_1510"
        port="auto"
        vendorid="0001"
        productid="0000"
        bus="005"
        device="004"
        busport= 002"
#               protocol="megatec"
        langid_fix=0x0409
#               novendor
#        norating
#               noscanlangid
```

Errors:

```
sudo upsdrvctl start
Network UPS Tools - UPS driver controller 2.8.1
Network UPS Tools - Megatec/Q1 protocol USB driver 0.17 (2.8.1)
libusb1: Could not open any HID devices: insufficient permissions on everything
No supported devices found. Please check your device availability with 'lsusb'
and make sure you have an up-to-date version of NUT. If this does not help,
try running the driver with at least 'subdriver', 'vendorid' and 'productid'
options specified. Please refer to the man page for details about these options
(man 8 blazer_usb).

upsnotify: notify about state 4 with libsystemd: was requested, but not running as a service unit now, will not spam more about it
upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
Driver failed to start (exit status=1)
Network UPS Tools - Megatec/Q1 protocol USB driver 0.17 (2.8.1)
libusb1: Could not open any HID devices: insufficient permissions on everything
No supported devices found. Please check your device availability with 'lsusb'
and make sure you have an up-to-date version of NUT. If this does not help,
try running the driver with at least 'subdriver', 'vendorid' and 'productid'
options specified. Please refer to the man page for details about these options
(man 8 blazer_usb).

upsnotify: notify about state 4 with libsystemd: was requested, but not running as a service unit now, will not spam more about it
upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
Driver failed to start (exit status=1)
Network UPS Tools - Megatec/Q1 protocol USB driver 0.17 (2.8.1)
libusb1: Could not open any HID devices: insufficient permissions on everything
No supported devices found. Please check your device availability with 'lsusb'
and make sure you have an up-to-date version of NUT. If this does not help,
try running the driver with at least 'subdriver', 'vendorid' and 'productid'
options specified. Please refer to the man page for details about these options
(man 8 blazer_usb).

upsnotify: notify about state 4 with libsystemd: was requested, but not running as a service unit now, will not spam more about it
upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
Driver failed to start (exit status=1)
```

nut-scanner:

```
sudo nut-scanner                                                                                                                                                                                                  
Scanning USB bus.
No start IP, skipping SNMP
Scanning XML/HTTP bus.
No start IP, skipping NUT bus (old connect method)
[nutdev1]
        driver = "nutdrv_qx"
        port = "auto"
        vendorid = "0001"
        productid = "0000"
        bus = "005"
        device = "006"
        busport = "002"
        ###NOTMATCHED-YET###bcdDevice = "0100"

```

With that conf:

```
sudo upsdrvctl start                                                                                                                                                                                                
Network UPS Tools - UPS driver controller 2.8.1
Network UPS Tools - Generic Q* USB/Serial driver 0.36 (2.8.1)
USB communication driver (libusb 1.0) 0.46
Device not supported!
Device not supported!
upsnotify: notify about state 4 with libsystemd: was requested, but not running as a service unit now, will not spam more about it
upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
Driver failed to start (exit status=1)
Network UPS Tools - Generic Q* USB/Serial driver 0.36 (2.8.1)
USB communication driver (libusb 1.0) 0.46
Device not supported!
Device not supported!
upsnotify: notify about state 4 with libsystemd: was requested, but not running as a service unit now, will not spam more about it
upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
Driver failed to start (exit status=1)
Network UPS Tools - Generic Q* USB/Serial driver 0.36 (2.8.1)
USB communication driver (libusb 1.0) 0.46
Device not supported!
Device not supported!
upsnotify: notify about state 4 with libsystemd: was requested, but not running as a service unit now, will not spam more about it
upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
Driver failed to start (exit status=1)
```

jimklimov, 2024-03-20:
@binarydepth : bus/device/port are enumeration-based so on some systems tend to change over time (device replugs, bus resets...) and may cause more problems than they solve :\ They are not suggested by default with `nut-scanner` in master branch anymore (eventually NUT v2.8.2). Commenting them away completely can help, it seems, specifically in your case.

I am a bit more concerned about "libusb1: Could not open any HID devices: insufficient permissions on everything" -- this means that either:
* the OS did not delegate devfs nodes for these USB devices (problem with udev/upower/similar config)
* some other program, possibly a systemd-wrapped copy of same driver, holds the device (see NDE in wiki)
* sort of mis-reported problem with all *other* devices (mouse etc.) not permitted validly, and the UPS skipped because not found on the expected bus/port/device...

binarydepth, 2024-03-23:
@jimklimov 

I tried this solution from [Artcle](https://technotim.live/posts/NUT-server-guide/) and no go still. I took out the bus entries and still works with command after login. Can read read on [NDE](https://github.com/networkupstools/nut/wiki/nut%E2%80%90driver%E2%80%90enumerator-(NDE)) but in that instance is there a know solution you can provide?

```
SUBSYSTEM!="usb", GOTO="nut-usbups_rules_end"

ACTION=="add|change", SUBSYSTEM=="usb|usb_device", SUBSYSTEMS=="usb|usb_device", ATTR{idVendor}=="0001", ATTR{idProduct}=="0000", MODE="664", GROUP="nut", RUN+="/sbin/upsdrvctl stop; /sbin/upsdrvctl start"

LABEL="nut-usbups_rules_end"
```

Jipok, 2024-04-17:
I have CROWN CMU-SP800EURO LCD USB
`lsusb ID 0001:0000 Fry's Electronics`

I think I was able to get it to work, but I'm confused by the voltage information. Here are my configs and logs:
<details><summary>ups.conf</summary>

```
[ups]
        driver = "nutdrv_qx"
        port = "auto"
        vendorid = "0001"
        productid = "0000"
        langid_fix=0x0409
```

</details>

<details><summary>/etc/udev/rules.d/50-ups.rules</summary>

```
SUBSYSTEM=="usb", ATTR{idVendor}=="0001", ATTR{idProduct}=="0000", GROUP="nut", RUN+="/usr/bin/upsdrvctl stop; /usr/bin/upsdrvctl start"
```

</details>

<details><summary>upsdrvctl start</summary>

```
Network UPS Tools - UPS driver controller 2.8.0
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43
Using protocol: Megatec 0.06
No values for battery high/low voltages
Using 'guesstimation' (low: 10.400000, high: 13.000000)!
Battery runtime will not be calculated (runtimecal not set)
```

</details>

<details><summary>upsc</summary>

Power on:
```
battery.charge: 100
battery.voltage: 13.60
battery.voltage.high: 13.00
battery.voltage.low: 10.40
battery.voltage.nominal: 12.0
device.type: ups
driver.name: nutdrv_qx
driver.parameter.langid_fix: 0x0409
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0000
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0001
driver.version: 2.8.0
driver.version.data: Megatec 0.06
driver.version.internal: 0.32
driver.version.usb: libusb-1.0.27 (API: 0x1000109)
input.current.nominal: 0.0
input.frequency: 49.9
input.frequency.nominal: 0
input.voltage: 209.0
input.voltage.fault: 0.0
input.voltage.nominal: 0
output.voltage: 211.0
ups.beeper.status: disabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware: V3.8
ups.load: 42
ups.productid: 0000
ups.status: OL
ups.temperature: 29.0
ups.type: offline / line interactive
ups.vendorid: 0001
```
Power off:
```
battery.charge: 81
battery.voltage: 12.50
battery.voltage.high: 13.00
battery.voltage.low: 10.40
battery.voltage.nominal: 12.0
device.type: ups
driver.name: nutdrv_qx
driver.parameter.langid_fix: 0x0409
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0000
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0001
driver.version: 2.8.0
driver.version.data: Megatec 0.06
driver.version.internal: 0.32
driver.version.usb: libusb-1.0.27 (API: 0x1000109)
input.current.nominal: 0.0
input.frequency: 0.0
input.frequency.nominal: 0
input.voltage: 0.0
input.voltage.fault: 0.0
input.voltage.nominal: 0
output.voltage: 220.0
ups.beeper.status: disabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware: V3.8
ups.load: 25
ups.productid: 0000
ups.status: OB
ups.temperature: 29.0
ups.type: offline / line interactive
ups.vendorid: 0001
```

</details>

Is it possible to fix this?
How to add information to the list on the site?

jimklimov, 2024-04-17:
Thanks for the report!

Can you please elaborate about the voltage - what exactly looks wrong to you?

For PbAc (typical) batteries with 12V nominal, the reported numbers look OK. Input/output voltage is a bit low for areas with 220V nominal - it seems your utility network gives too little (house overloaded the nearest transformer, etc.) but it is deemed sufficiently high by the device and its settings (if it has any for boosting) that it is not trying to use the battery or inverter to boost the low voltage into closer to nominal - UPS thinks it is already in acceptable range.

As for adding to the site, it is a bit of a sore point regarding a long backlog of low-priority issues to update HCL (in main NUT repo, `data/driver.list.in` file) and DDL (in `nut-ddl` repo). PRs are welcome to both, otherwise just have to wait until I get around to this, I guess :)

Jipok, 2024-04-17:
> what exactly looks wrong to you?

In log: `No values for battery high/low voltages`
Also, the battery voltage dropped sharply when the power was turned off. But after switching on for some time it showed 14 volts.
I thought maybe this had something to do with the incorrect interpretation of the values from ups (well, or ups itself produces such data).

So what do you mean that all functionality works correctly? But how can I turn off the beep?

I also don’t quite understand how I can configure the server to shut down when the charge level drops below 50%. And how to turn off the UPS.
```
upsdrvctl shutdown
Network UPS Tools - UPS driver controller 2.8.0
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43
Can't claim USB device [0001:0000]@0/0: Entity not found
Driver failed to start (exit status=1)
```

> just have to wait until I get around to this, I guess :)

Well, I'll leave it to you. If you need any other logs or tests, I can provide them.

jimklimov, 2024-04-17:
PbAc nominal 12V is because it indeed can show 13.5-14V when fully charged, then quickly loses it down to ~12V, and stays close to 12V (slowly decreasing) as it discharges; then a near-dead battery quickly drops to 10-something V.

I did have some Powercom UPS 20 years ago which displayed completely different data when on-line or on-battery. So it may also be that different chips or different calibrations are responsible for different power cases, too.

kikiwora, 2024-11-20:
> I also got a Makelsan 650VA. It did not work with this configuration. I had to change the config a little. I am not able to reach the config now , I'll post when I'm home.

Hey, are you home yet?
Would you kindly share the 650VA config with us?

I have the same device and am interested in running it properly on macOS

Though funny enough, by using a config you provided earlier, I am getting this error in Terminal: 
`
Can't claim USB device [0001:0000]@0/0/0: Access denied (insufficient permissions)
`

jimklimov, 2024-11-20:
Generally there may be several fundamental reasons for "Access denied", to rule some of them out, can you please add `user=root` to `ups.conf` in the device configuration section (assuming the launcher begins life as `root` and then the daemon drops privileges)?

Namely, one reason is permissions for the device filesystem node (or equivalent) used by libusb for us, that may be alleviated by running as root. Normally chown/chmod that node, e.g. using a facility like udev, devd, hotplugd and such (whichever is available for the platform). 

Another common reason is that some other program (different copy of NUT driver, default OS USB handler, etc.) grabbed the device. Maybe `lsof`, `fuser` and such can reveal this.

Also there well may be some MacOS idiosyncrasies I am only aware of by hearsay, and recent releases seem more constrained than those a bit older (perhaps @clepple knows more).

trablin45, 2025-03-16:
Hello everyone my first time using nut and after several attempts the information in this post was the first useful configs I found to make my ups work with nut at least partially so thank you all. Im too working with a fry's electronics generic usb ups, 850va, 8A single battery. 

>ups.conf
 [myups]
    driver="nutdrv_qx"
    subdriver="snr" (hunnox also worked similary)
    protocol ="megatec"
    port="auto"
    vendorid="0001"
    productid="0000"
    bus="001"
    langid_fix="0x0409"
    novendor
    norating
    noscanlangid

> 
upsc outputs this:

> Init SSL without certificate database
battery.voltage: 12.70
device.type: ups
driver.flag.norating: enabled
driver.flag.noscanlangid: enabled
driver.flag.novendor: enabled
driver.name: nutdrv_qx
driver.parameter.bus: 001
driver.parameter.langid_fix: 0x0409
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0000
driver.parameter.protocol: megatec
driver.parameter.subdriver: snr
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0001
driver.version: 2.8.0
driver.version.data: Megatec 0.06
driver.version.internal: 0.32
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.frequency: 50.2
input.voltage: 236.0
input.voltage.fault: 0.0
output.voltage: 236.0
ups.beeper.status: disabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.load: 13
ups.productid: 0000
ups.status: OL
ups.temperature: 29.0
ups.type: offline / line interactive
ups.vendorid: 0001

> 
I don't have battery.charge report and other outputs are missing. Im trying to set a safe shutdown on the host when batteries are low (25% for safety) what else am I missing in my config ?  I think without the battery.charge I can't program the safe shutdown, I manually tested and ups just switch off when it was low bat (on the little ups screen it was 2 bars/5 when this happened)

Steve-Tech, 2025-03-22:
Hi, I just wanted to document this for the internet.

I managed to get a Jaycar MP5205 UPS (MP5207 will probably also work) working with NUT.

lsusb: `Bus 001 Device 010: ID 0001:0000 Fry's Electronics`

`/etc/nut/ups.conf`
```
[mp5205]
driver = "nutdrv_qx"
port = "auto"
vendorid = "0001"
productid = "0000"
protocol="megatec"
langid_fix=0x0409
```

`upsc mp5205`
```
Init SSL without certificate database
battery.charge: 100
battery.voltage: 13.7
battery.voltage.high: 13.00
battery.voltage.low: 10.40
battery.voltage.nominal: 12.0
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: nutdrv_qx
driver.parameter.langid_fix: 0x0409
driver.parameter.pollfreq: 2
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0000
driver.parameter.protocol: megatec
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0001
driver.state: quiet
driver.version: 2.8.1
driver.version.data: Megatec 0.07
driver.version.internal: 0.36
driver.version.usb: libusb-1.0.27 (API: 0x100010a)
input.current.nominal: 0.0
input.frequency: 50.2
input.frequency.nominal: 0
input.voltage: 240.0
input.voltage.fault: 0.0
input.voltage.nominal: 0
output.voltage: 241.0
ups.beeper.status: disabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware: V3.9
ups.load: 0
ups.productid: 0000
ups.status: OL
ups.temperature: 29.0
ups.type: offline / line interactive
ups.vendorid: 0001
```