networkupstools/nut #1949: TOTHINK: In git snapshot rebuilds, suffix NUT version with number of commits since release

jimklimov, 2023-05-26:
Autoconf tricks to allow embedding the git version are explored in autoconf's own build scripts:
* http://git.savannah.gnu.org/gitweb/?p=autoconf.git;a=blob_plain;f=build-aux/git-version-gen;hb=refs/heads/master
* https://stackoverflow.com/questions/22871340/using-a-variable-in-ac-init

The idea here might be to determine a release tag (annotated? by pattern?) as opposed to major milestones, release candidates etc. tagged during development), and count the number of commits since then.

A release might be `v2.8.0` (disregarding items like `v2.8.0-Windows`, `Windows-v2.8.0-alpha4` or `v2.8.1-rc1` which might have happened later, but maybe including patterns like `v2.8.0-signed` (which should refer to same commit as original `v2.8.0` but with a signed tag object) and then the custom build would be configured as `AC_INIT([nut], [2.8.0.123]...)` for 123 commits after that release. 

If this tweak to metadata works, it would be more idiomatic (and welcoming to rolling-release packaging form master branch by CI, for example) than calling all such custom builds formally `2.8.0.1` and detailing the git maturity level in plain comments or help messages which are not easily machine-parsable.

For bonus points, we may want to track the codebase maturity compared to both master and last release, perhaps `2.8.0.123.45` meaning that "master" branch had 123 commits since last release (2.8.0) and then the currently branched-off effort for a PR added 45 commits on top of that newest commit in the PR branch that is also in known master branch history (the branching or last-resync point), or alternately - just say 345 commits tracked in the PR branch overall when also counting since that release, which may be easier to tackle. 

Then a build from newer master branch with 130 commits (or of a PR with foundations on that baseline) would win in terms of automated packaging version resolution, etc. despite the longer (and variable) amount of commits in development branches.

CC @clepple @aquette : WDYT?

dak180, 2023-05-29:
[Autorevision](https://github.com/Autorevision/autorevision) is made for doing exactly this and there are even examples using [make](https://autorevision.github.io/blog/2015/03/15/Examples).

jimklimov, 2023-05-29:
Looks neat, thanks!

jimklimov, 2023-06-01:
Looking around, I saw we have precedent of a similar approach in configuration of nut-website at https://github.com/networkupstools/nut-website/blob/64361632b90ed427b5bc2e364e37ccdc8118ffe6/configure.ac#L5

Granted, changes or mishaps there are less intrusive (no end-users or tarballs to get wrong) in case of mistakes.

jimklimov, 2023-06-02:
Related trickery: https://github.com/MidnightCommander/mc/commit/eab84392efb974a4b53c5411e3a3d7bcb7cf881f

jimklimov, 2024-07-24:
Variant of the above updates, with debugging and restored tail of the `git describe` (commit count since tag and current hash):
````
getver() (
    TRUNK=master;
    DESC="`git describe --match 'v[0-9]*.[0-9]*.[0-9]' --exclude '*-signed' --exclude '*rc*' --exclude '*alpha*' --exclude '*beta*' --exclude '*Windows*' --exclude '*IPM*' --always`";
    TAG="`echo "$DESC" | sed 's/-[0-9][0-9]*-g[0-9a-fA-F][0-9a-fA-F]*$//'`";
    SUFFIX="`echo "$DESC" | sed 's/^.*\(-[0-9][0-9]*-g[0-9a-fA-F][0-9a-fA-F]*\)$/\1/'`" && [ x"$SUFFIX" != x"$TAG" ] || SUFFIX="";
    BASE="`git merge-base HEAD $TRUNK`" ;
    VER5="${TAG/v}.`git log --oneline $TAG..$BASE | wc -l`.`git log --oneline $TRUNK..HEAD | wc -l`";
    VER50="`echo "$VER5" | sed -e 's/\.0$//' -e 's/\.0$//'`";
    BIGDESC="${VER50}${SUFFIX}";
    echo "DESC='${DESC}' => TAG='${TAG}' + SUFFIX='${SUFFIX}'; BASE='${BASE}' => VER5='${VER5}' => VER50='${VER50}' => BIGDESC='${BIGDESC}'" >&2
    echo "$BIGDESC"
 )

:; git checkout FTY && getver >/dev/null
DESC='v2.8.2-3854-g8fa89a270' => TAG='v2.8.2' + SUFFIX='-3854-g8fa89a270'; BASE='7dcc938bdc9b26a2e9b861940f70d0e17d3c9921' => VER5='2.8.2.680.3174' => VER50='2.8.2.680.3174' => BIGDESC='2.8.2.680.3174-3854-g8fa89a270'

:; git checkout master~20 && getver >/dev/null
DESC='v2.8.2-655-g91ea6270b' => TAG='v2.8.2' + SUFFIX='-655-g91ea6270b'; BASE='91ea6270b287220a22cc1fd1a6de420b026d0c6c' => VER5='2.8.2.655.0' => VER50='2.8.2.655' => BIGDESC='2.8.2.655-655-g91ea6270b'

:; git checkout v2.8.2 && getver >/dev/null
DESC='v2.8.2' => TAG='v2.8.2' + SUFFIX=''; BASE='440ca2348e665abf3787c30bdcd9373b479f4efc' => VER5='2.8.2.0.0' => VER50='2.8.2' => BIGDESC='2.8.2'

:; git checkout issue-1316-str_concat && getver >/dev/null
DESC='v2.8.2-339-ga1d3b8fe9' => TAG='v2.8.2' + SUFFIX='-339-ga1d3b8fe9'; BASE='dc099e8e57729a221e8343fdf1449ee0466e06a3' => VER5='2.8.2.338.1' => VER50='2.8.2.338.1' => BIGDESC='2.8.2.338.1-339-ga1d3b8fe9'
````

So this hack seems to best marry the two worlds for NUT purposes. Note that the local git repo index must know of the fresh trunk history which is not always a given (see #2551) to properly generate such information.

jimklimov, 2024-07-25:
For autotools integration, got interesting inspiration from discussion at https://lists.gnu.org/archive/html/bug-autoconf/2024-01/msg00022.html to have the information generated into an m4 file (whether by `autogen.sh` during build from git, or passed via tarball), and include that into the `configure.ac` as m4 code:

* https://gitlab.com/secsh/pkixssh/-/blob/master/configure.ac
* https://gitlab.com/secsh/pkixssh/-/blob/master/version.m4
* * https://gitlab.com/secsh/pkixssh/-/blob/master/version.sh

jimklimov, 2024-07-26:
So far the PR #2554 looks quite promising for the main NUT codebase. 

I've considered briefly if other repos need it (nut-website, nut-ddl, etc.) but their use of branching is very rare and limited, so standard `git describe` should suffice.

closed 2024-07-28 by jimklimov