networkupstools/nut #1957: Missing anchor in docs for `_status_data`

jimklimov, 2023-06-05:
`docs/nut-names.txt` refers from `ups.status` entry to `_status_data` anchor, I think, but I did not find such tag elsewhere in the sources. I believe it should be in `new-drivers.txt` description of working with statuses.

Need to cross-check with rendered docs on site, and maybe add the anchor to that chapter.

jimklimov, 2023-06-05:
That's odd: the anchor `_status_data` should get generated implicitly with presence of a header named `Status Data` in the document, to the extent that adding one explicitly does break the ASCIIDOC XML validation step.

In fact, the proper URL is present in single-HTML build of developer guide - but absent/dysfunctional in the single-PDF and chunked-HTML renditions.

I can define and explicitly add a `[[status-data]]` anchor and refer to it from `docs/nut-names.txt` instead of the original underscored anchor:
````
| ups.status            | UPS status                   | linkdoc:developer-guide[OL,status-data]  
````
...but the results behave the same (single-HTML link changes and works; two others remain invalid).

jimklimov, 2023-06-05:
However changing the link markup from the custom `linkdoc` macro to angle brackets "just works" for all three rendered documents. @aquette : can you weigh in with any insight on this, please?

````
-| ups.status            | UPS status                   | linkdoc:developer-guide[OL,_status_data]
+| ups.status            | UPS status                   | <<_status_data,OL>>
````

arnaudquette-eaton, 2023-06-05:
EREALLYNOTIME :/
the ref. link is https://github.com/networkupstools/nut/blob/master/docs/new-drivers.txt#L210

arnaudquette-eaton, 2023-06-05:
and you're hitting the limit I've been hitting. No basic solution on my side, apart from a variable linkdoc macro that get the right page+anchor in chunked format

PR #1959 (2023-06-06): Fix documentation references to `_status_data`
Closes: #1957
closed 2023-06-06 by jimklimov