networkupstools/nut #1960: No data after ~1 day

hajkenek, 2023-06-06:
Hi, i have strange problem. After about 1 day of running nut-upsd in docker there is no new data from ups.
No logs - only:
```
2023-06-04T14:39:03.582115162Z Network UPS Tools - UPS driver controller 2.8.0
2023-06-04T14:39:03.583145442Z Network UPS Tools - Generic HID driver 0.47 (2.8.0)
2023-06-04T14:39:03.609229914Z Using subdriver: Phoenixtec/Liebert HID 0.41
2023-06-04T14:39:03.609278764Z USB communication driver (libusb 1.0) 0.43
2023-06-04T14:39:03.617037196Z Network UPS Tools upsd 2.8.0
2023-06-04T14:39:03.617104686Z Ignoring invalid pid number 0
2023-06-04T14:39:03.617393778Z listening on 0.0.0.0 port 3493
2023-06-04T14:39:03.617778073Z Connected to UPS [tatooine-ups]: usbhid-ups-tatooine-ups
2023-06-04T14:39:03.623763083Z Network UPS Tools upsmon 2.8.0
2023-06-04T14:39:03.623808699Z    0.000000	Ignoring invalid pid number 0
2023-06-04T14:39:03.624036619Z    0.000197	UPS: tatooine-ups@localhost (primary) (power value 1)
2023-06-04T14:39:03.624061594Z    0.000247	[D1] debug level is '1'
2023-06-04T14:39:03.624076525Z    0.000259	[D1] debug level is '1'
2023-06-04T14:39:03.624632712Z    0.000810	[D1] Saving PID 18 into /var/run/upsmon.pid
2023-06-04T14:39:03.625282309Z Init SSL without certificate database
2023-06-04T14:39:03.627546847Z    0.003691	[D1] Trying to connect to UPS [tatooine-ups@localhost]
2023-06-04T14:39:03.628238592Z    0.004423	[D1] Logged into UPS tatooine-ups@localhost
```

And everything works fine till ~1 day (the same after restarting container)
[image]

I'm using pollonly=true in config

jimklimov, 2023-06-06:
In an earlier post version I saw a graph from your switch going from some 5Mbit/s to zero and intermittent connectivity, I think - did it reflect the loss of networking in the container (or its host)? E.g. trying to point fingers that the problem is not in NUT itself? ;)

jimklimov, 2023-06-06:
Generally, what do the NUT logs say around the time the data disappears?

I assume you have some single script running the driver, upsd and upsmon in the same container - so loss of connectivity between them over localhost link would be peculiar. And some graphing client separately, which loses the data stream (but this may be due to various causes)?

hajkenek, 2023-06-06:
It was wrong screenshot accidently copied from my clipboard. Totally unrelated to main topic. I tried to delete this, but probably it's somewhere in assets. 
All logs that I have are pasted above. So looks like nothing change. After restarting nut container I see new data again for about 1 day.

hajkenek, 2023-06-06:
Nothing interesting for now. Disconnecting and connecting USB does nothing. There is no new logs. Container usage looks normal. Now I'm trying to retrive some data from nut server somehere else to be 100% sure if this is related to nut server not nut client on my homeassistant instance.

jimklimov, 2023-06-06:
Ok, so mentioning Home Assistant is a useful clue :)

I do not have direct experience with it, but some references from NUT Wiki and earlier discussions might help...

* https://github.com/networkupstools/nut/wiki/NUT-with-Proxmox-and-Home-Assistant
* https://github.com/networkupstools/nut/wiki/Building-NUT-integration-for-Home-Assistant
* https://github.com/networkupstools/nut/labels/Home%20Assistant%20%28HA%29

As for logging, with NUT 2.8.0 and newer (if you try custom builds) the `debug_min NUM` setting should be supported in daemons (drivers, upsd, upsmon) config files to set their verbosity levels; practical values are up to 6. With earlier NUT versions, you are limited to providing many `-D` arguments to launching command line.

hajkenek, 2023-08-07:
No more logs, I set up all three config files with this line:
[image]

Maybe I make something wrong?

jimklimov, 2023-08-07:
Well, by config it seems there should be more logs. Maybe in syslog sink or just anonther log/journal file?