networkupstools/nut #1961: Update downloads.txt for newer VMWare client builds

jimklimov, 2023-06-06:
http://rene.margar.fr/2012/05/client-nut-pour-esxi-5-0/ is referenced in the doc, but only lists 2.7.4 builds. In fact there is a 2.8.0 artifact in that blog as well.

rgc2000, 2023-06-07:
Here is the clone of my git server in GitHub : https://github.com/rgc2000/NutClient-ESXi
I have also uploaded all the binary releases since the project is managed in git

jimklimov, 2023-06-08:
Merci beaucoup! :)
I will update the site and Wiki links.

jimklimov, 2023-06-08:
Thank you for the historic details in https://rene.margar.fr/2012/05/client-nut-pour-esxi-5-0/comment-page-22/#comment-13325

Strangely, `initgroups` never popped up on other platforms. Does ESXi libc just lack it? Does it make sense to add a NUT `configure.ac` check (and `ifdef` in `common.c`) as a more future-proof option?..

jimklimov, 2023-06-08:
Updated (forgot to cross-reference): https://github.com/networkupstools/nut/commit/8a1d532db04cd2d35a547b3135082a90c3b005ae

closed 2023-06-08 by jimklimov

rgc2000, 2023-06-08:
The initgroup function does exist on ESXi libc but I think that the problem comes from the group database that is very minimalistic. On earlier versions of ESXi (let's say 5.0) the /etc/passwd and /etc/group were as we use to have in a linux system. But on later versions they have been reduced and user "nobody" does not even exist anymore, and so the group. So upsmon has to run as root otherwise there is no existing unprivileged user for that. There are also some other security reasons that makes upsmon to run with root privileges mandatory otherwise it cannot perform some operations. ESXi was customized linux kernel at the beginning, it's not so true anymore now.