networkupstools/nut #1966: cant log in to nut server from nut client (home assistant)

tomasenskede, 2023-06-15:
[image]

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting
-----------------------------------------------------------
 Add-on: Network UPS Tools
 Manage battery backup (UPS) devices
-----------------------------------------------------------
 Add-on version: 0.12.0
 You are running the latest version of this add-on.
 System: Home Assistant OS 10.2  (amd64 / qemux86-64)
 Home Assistant Core: 2023.6.2
 Home Assistant Supervisor: 2023.06.2
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
[13:48:47] INFO: Configuring timezone (Europe/Stockholm)...
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/nut.sh
[13:48:48] INFO: Setting mode to netserver...
[13:48:48] INFO: Generating /etc/nut/upsd.users...
[13:48:48] INFO: Configuring user: nutusername
[13:48:49] INFO: Password is NOT in the Have I Been Pwned database! Nice!
[13:48:49] INFO: Configuring Device named PP100...
[13:48:49] INFO: Starting the UPS drivers...
Network UPS Tools - UPS driver controller 2.8.0
cont-init: info: /etc/cont-init.d/nut.sh exited 0
cont-init: info: running /etc/cont-init.d/nutclient.sh
cont-init: info: /etc/cont-init.d/nutclient.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun upsd (no readiness notification)
services-up: info: copying legacy longrun upsmon (no readiness notification)
s6-rc: info: service legacy-services successfully started
[13:48:55] INFO: Starting the UPS information server...
   0.000000	fopen /run/nut/upsd.pid: No such file or directory
   0.000780	Could not find PID file '/run/nut/upsd.pid' to see if previous upsd instance is already running!
   0.001264	[D1] debug level is '1'
   0.002871	listening on 0.0.0.0 port 3493
   0.003383	Connected to UPS [PP100]: richcomm_usb-PP100
   0.004110	Running as foreground process, not saving a PID file
[13:48:55] INFO: Starting the UPS monitor and shutdown controller...
   0.000000	fopen /run/nut/upsmon.pid: No such file or directory
   0.000495	Could not find PID file to see if previous upsmon instance is already running!
   0.000946	Using power down flag file /etc/killpower
   0.001285	UPS: PP100@localhost (primary) (power value 1)
   0.001537	[D1] debug level is '1'
   0.001779	[D1] debug level is '1'
   0.002027	Warning: running as one big root process by request (upsmon -p)
   0.002338	[D1] Saving PID 459 into /run/nut/upsmon.pid
Init SSL without certificate database
   0.005671	[D1] Trying to connect to UPS [PP100@localhost]
   0.029350	User upsmonmaster@127.0.0.1 logged into UPS [PP100]
   0.008154	[D1] Logged into UPS PP100@localhost

```
[image]

[image]

[image]

Whats wrong?

jimklimov, 2023-06-15:
>    0.001285	UPS: ***@***.*** (primary) (power value 1)
>    0.029350	User ***@***.*** logged into UPS [PP100]

For one, in the later screenshot there is a typo in the name...

tomasenskede, 2023-06-15:
> ot there is a typo in the na

ntusername? it was corrected, I had to retype the username for the screenshot, its corrected now. but regardless if I leave it empty or copy paste it from config I get the same error "failed to connect"... not "incorrect credentials"

jimklimov, 2023-06-15:
On NUT side I can suggest to somehow boost debug verbosity for logging (see wiki for some options) - but not sure how to properly do that in the container.

Also, did you ask the HA communities (general and/or NUT plugin) about that?

jimklimov, 2023-06-15:
Where does that Web(?)UI run? is it same container, or another system? If the latter, does HA networking let it through (names, routing, firewalls ...)

tomasenskede, 2023-06-15:
> Where does that Web(?)UI run? is it same container, or another system? If the latter, does HA networking let it through (names, routing, firewalls ...)

I am running it all in one HA with HA O/S. Installed NUT as Add-in thats it according to this: https://www.home-assistant.io/integrations/nut

jimklimov, 2023-06-15:
Well, other than checking the systemic basics - bumping the logs, kicking the tires - not much comes to mind right now.

Can other clients connect to that server, at all (upsc) and with auth (upscmd beeper or something like that)?

tomasenskede, 2023-06-15:
I will try to install the server on another server and see if its the server or the client that is the issue.

tomasenskede, 2023-06-16:
Clean install of the nut server on Debian vm the integration in home assistant could connect to the nut server and it all works just fine!

jimklimov, 2023-06-27:
So I guess the problem is somewhere outside NUT. Can this issue be closed?

tomasenskede, 2023-06-27:
> So I guess the problem is somewhere outside NUT. Can this issue be closed?

yes, the resulution was to install the NUT services on an seperate VM and then it worked fine.

closed 2023-06-27 by jimklimov