networkupstools/nut #1967: Improved Phoenixtec 0x06DA:0xFFFF support

blazzakrajsek, 2023-06-16:
I have noticed that upgrade of Debian 11 (bullseye) to Debian 12 (bookworm) upgraded the `nut` package from 2.7.4 to 2.8.0.
The support for PowerWalker VI 1200 SHL is now worse, see my previous report #1270. I have noticed that the driver has changed from `MGE HID 1.40` to `Phoenixtec/Liebert HID 0.41`.

Current status under NUT 2.8.0 for PowerWalker VI 1200 SHL is as follows...
`ups.conf`:
```
[ups0]
    driver = "usbhid-ups"
    port = "auto"
    vendorid = "06DA"
    productid = "FFFF"
```

`upsc ups0`:
```
Init SSL without certificate database
battery.charge: 100
battery.runtime: 7200
battery.type: PbAc
battery.voltage: 27.00
device.mfr: PPC
device.model: Offline UPS
device.serial: 000000000   
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: FFFF
driver.parameter.synchronous: auto
driver.parameter.vendorid: 06DA
driver.version: 2.8.0
driver.version.data: Phoenixtec/Liebert HID 0.41
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.voltage: 237.0
output.frequency: 49.80
output.voltage: 237.0
ups.load: 0
ups.mfr: PPC
ups.model: Offline UPS
ups.productid: ffff
ups.serial: 000000000   
ups.status: OL CHRG
ups.vendorid: 06da
```

`upsrw -l ups0`:
```
(empty)
```

`upscmd -l ups0`:
```
Instant commands supported on UPS [ups0]:
(empty)
```

Commands and variables are now not available any more. For example, with the `shutdown.return` command missing, UPS shutdown is not possible.

I guess (as I did not test) that the fixes implemented by #1357 made support for Phoenixtec 0x06DA:0xFFFF devices in general poorer.

jimklimov, 2023-06-16:
Thanks for the report! it would really be ibteresting to know if that PR is the culprit - e.g. if its disablement for non-AEG branded devices should be expanded to more names, or made a config toggle...

blazzakrajsek, 2023-06-17:
Glad to help. Indeed the commit `d30ffffb8f` changed the driver for this device.

---
### `d30ffffb8f`:

`upsc ups0`: 
``` 
battery.charge: 100
battery.runtime: 7200
battery.type: PbAc
battery.voltage: 27.00
device.mfr: PPC
device.model: Offline UPS
device.serial: 000000000
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: FFFF
driver.parameter.synchronous: auto
driver.parameter.vendorid: 06DA
driver.version: 2.8.0-rc1-34-gd30ffffb8
driver.version.data: Phoenixtec/Liebert HID 0.4
driver.version.internal: 0.46
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.voltage: 238.0
output.frequency: 49.80
output.voltage: 238.0
ups.load: 0
ups.mfr: PPC
ups.model: Offline UPS
ups.productid: ffff
ups.serial: 000000000
ups.status: CHRG
ups.vendorid: 06da
```

`upsrw ups0`:
```
(empty)
```

`upscmd -l ups0`:
```
Instant commands supported on UPS [ups0]:
(empty)
```

---
### `51e6b886a`:

`upsc ups0`:
```
battery.charge: 100
battery.runtime: 7200
battery.type: PbAc
device.mfr: PPC
device.model: Offline UPS HID UPS
device.serial: 000000000   
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: FFFF
driver.parameter.synchronous: auto
driver.parameter.vendorid: 06DA
driver.version: 2.8.0-rc1-33-g51e6b886a
driver.version.data: MGE HID 1.45
[... 151 lines cut ...]
Maximum length: 10
Value: 30
```

`upscmd -l ups0`:
```
Instant commands supported on UPS [ups0]:

beeper.toggle - Toggle the UPS beeper
driver.killpower - Description unavailable
driver.reload - Description unavailable
driver.reload-or-error - Description unavailable
driver.reload-or-exit - Description unavailable
load.off - Turn off the load immediately
load.off.delay - Turn off the load with a delay (seconds)
load.on - Turn on the load immediately
load.on.delay - Turn on the load with a delay (seconds)
shutdown.return - Turn off the load and return when power is back
shutdown.stayoff - Turn off the load and remain off
shutdown.stop - Stop a shutdown in progress
test.battery.start - Start a battery test
```

As you suggested, I believe the best course of action would be to add a subdriver or some other option to be able to better control the options available for the exact UPS, since the VID/PID 0x06DA:0xFFFF is obviously used by many different devices.
I am open to suggestions and let me know if I can help any further. :)

jimklimov, 2023-06-17:
Noting for posterity: some progress about #1369 (essentially, adding `subdriver=...` support to `usbhid-ups`) could also help here.

blazzakrajsek, 2023-06-27:
I have renamed the ticket to better reflect the issue. Closing issue as resolved with #1968.

closed 2023-06-27 by blazzakrajsek