networkupstools/nut #1969: [Cyberpower PR1500ELCD] Various regression with latest updates

sblantipodi, 2023-06-16:
hello... I'm using the NUT addon since years now.
it works super awesome on my CyberPower PR1500ELCD using USB connected to my Pi4 running Home Assistant OS

Since latest udpates some entities stopped working.
Most entities works ok but
sensor.myups_output_voltage
is fixed at 265V.

it worked before, now it shows 265V no matter what the real output voltage is.

sblantipodi, 2023-06-16:
can this fix 
https://github.com/networkupstools/nut/issues/439
helps on this issue?

jimklimov, 2023-06-17:
I think there were some changes to that area after 2.8.0 release. Did you install a custom build or the release?

sblantipodi, 2023-06-17:
@jimklimov I'm using the one inside the Home Assistant NUT addon:
Network UPS Tools - Generic HID driver 0.47 (2.8.0)
but there are other users complaining about the same issue on a non custom build.

jimklimov, 2023-06-17:
Thanks. The practical side of the question is whether that issue did get fixed in existing NUT master-branch codebase, as I suspect it might have been after 2.8.0 release. Eventually when other bug-fixes are collected and 2.8.1 is cut, it would get solved also for other people's "non custom builds" :)

So if you would have a chance to check with newer NUT (whether by building a custom HA plugin binary - see wiki page https://github.com/networkupstools/nut/wiki/Building-NUT-integration-for-Home-Assistant for community-contributed guidelines, or in its host OS, or by dedicating a computer for a while to make and run the build) - it would be much appreciated. I have no experience with HA itself so can't help much there. With "common" builds in an OS, you can avoid installing the results and can run the driver from the build workspace to check if it works better than the older release snapshot - see https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests for details and suggestions about that.

sblantipodi, 2023-06-18:
hi @jimklimov it seems that the addon I'm using uses the nut=2.8.0-7 from the apt repo.
https://github.com/hassio-addons/addon-nut/blob/eb23aec5b591aab90a10a8189bd039c954ae5ab5/nut/Dockerfile#LL10C9-L10C20

I'm not able to build an addon currently and I have no time to learn it in this moment... hope to see someone else that could help here :)

ingcicala, 2023-06-23:
I reported the same bug on HA forum

KameDomotics, 2023-07-26:
I have the same problem, load sensor and output voltage sensor does not work. 
How can I fix it please?
(Home Assistant 2023.7.3 / NUT add on 0.12.0)
Thanks

sblantipodi, 2023-10-24:
I switched to docker compose image and same problem:
```
  nut-upsd:
    container_name: nut-upsd
    image: instantlinux/nut-upsd
    environment:
    - "PORT=/dev/bus/usb/001/003"
    - "NAME=Cyberpower_PR1500ELCD"
    - "DESCRIPTION=Cyberpower_PR1500ELCD"
    - "DRIVER=usbhid-ups"
    - "API_USER=admin"
    - "API_PASSWORD=mypwd"
    privileged: true
    ports:
      - 3493:3493
    restart: always
```

is there any interest in fixing this?

ingcicala, 2023-10-24:
> I switched to docker compose image and same problem:
>
> ```
>   nut-upsd:
>     container_name: nut-upsd
>     image: instantlinux/nut-upsd
>     environment:
>     - "PORT=/dev/bus/usb/001/003"
>     - [...]

I'm waiting for fix .....

jimklimov, 2023-10-24:
I am a bit perplexed that none of the reports seem to be from systems running NUT natively (containers are proposed by third parties, not by the NUT project itself). After all, if you want power for the host managed, usually a guest/container is not sufficiently privileged to shut down the host, right?

All that said, it is probably just a curiosity with regard to the issue with some readings being wrong.

> is there any interest in fixing this?

Is anyone with the device working on one? NUT sources and build procedures are available, drivers can be run from the build tree so not "corrupting" the original system with an installation (see wiki link for "in-place" builds above as a most-recent write-up on this).

Per original report this issue is a regression, so if you can find a version where it works and one where it does not (e.g. via `git bisect` or some better educated guesses), you can discover what change impacted this, and we can go from there about trying fixes.

A first step however would be to check if the current NUT master codebase still has the problem. I don't think anyone confirmed or denied that above, yet.

sblantipodi, 2023-10-24:
it's so weird...
I'm using an old image and it still show a wrong values on the output voltage...

docker pull instantlinux/nut-upsd:2.8.0-r0
it an image created a year ago...

```
/ # upsc Cyberpower_PR1500ELCD
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 4690
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 24.0
battery.voltage.nominal: 24
device.mfr: CPS
device.model: PR1500ELCD
device.serial: PRTGN2000005
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/bus/usb/001/003
driver.parameter.synchronous: auto
driver.version: 20220328-3481-gccbf3eb6c6
driver.version.data: CyberPower HID 0.6
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.24 (API: 0x1000109)
input.voltage: 236.0
input.voltage.nominal: 230
------ >output.voltage: 265.0 < -------------
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 16
ups.mfr: CPS
ups.model: PR1500ELCD
ups.productid: 0601
ups.realpower.nominal: 1350
ups.serial: PRTGN2000005
ups.status: OL
ups.test.result: Done and passed
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
```
I will try to follow the @jimklimov but it will take time...

sblantipodi, 2023-11-07:
@jimklimov 
I'm pretty sure that this is the offending pull request.
https://github.com/networkupstools/nut/pull/1497/files#diff-935d101f3ea64117ab54f95b7de69a3c261a7ee0528a2563292ef9c8b4f66d5eR285

since my Cyberpower is the one that is excluded there...
ups.productid: 0601

sblantipodi, 2023-11-07:
is this the only things needed to build the sources?
```
:; cd /tmp
:; git clone -b master https://github.com/networkupstools/nut
:; cd nut
:; ./ci_build.sh inplace
```

jimklimov, 2023-11-07:
https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests

has a chapter on prerequisites

sblantipodi, 2023-11-08:
ok, compilation is successful, I can see all the executable in the server and client config.
I compiled it with the inplance attribue:
./ci_build.sh inplace

how can I get the first reading now?

this command give me an error:
[image]

thanks for the support and for your great work here guys.

sblantipodi, 2023-11-08:
> After all, if you want power for the host managed, usually a guest/container is not sufficiently privileged to shut down the host, right?

Just to answer your question. With the combination of HA + Nut you can do very complex things...
I have a rule that sends me a Telegram message is there is an outage, and then sends me updates when the outage is finished, I can monitor the UPS status from outside the home, I have another rule that gracefully shut down the connected peripherals if the battery is less than 20% if I'm home, it shut down peripherals when battery is less than 50% if I'm not home... this is just to give you some basic examples of how powerful can be a smart home system :)
this obviously works well even if NUT is configured inside a docker container with the right tricks.

jimklimov, 2023-11-09:
Hm, it seems you would need to create that state directory (owned by nut) e.g. after installation with systemd integration, the `nut-common-tmpfiles.conf` should ensure that `systemd-tmpfiles --create` makes it as needed.

During the tests from the build area you can instead `export NUT_STATEPATH=/tmp` or similar.

Just to be sure, `0601` is the `ups.conf` section name for your UPS (emulated by `-s`)?

I think you are after something like:
````
:; NUT_STATEPATH=/tmp ./drivers/usbhid-ups -s test -x port=auto -x vendorid=0764 -x productid=0601 -DDDDDD -d1
````

sblantipodi, 2023-11-10:
ok, I built the latest main branch and it works well natively on my PC.

I also created a docker image and it obviously work well there too...

Official addons from HA are old and based on version 2.8, 
2.8 contains the issue, latest branch fixed the issue.

I think that we can close the issue here.

Thank you for all your support @jimklimov and for your awesome work here.

closed 2023-11-10 by sblantipodi

jimklimov, 2023-11-10:
See also NUT Github wiki for contributed hints about custom-building a HA container with newer NUT.

sblantipodi, 2023-11-10:
> See also NUT Github wiki for contributed hints about custom-building a HA container with newer NUT.

yes, what I have done with docker is more or less what is written on the wiki with my customizations and for my personal taste :D
at the end, data is consumed by HA in my use case too.
I think that people who are searching for help can find in that wiki a good start.

thank you! awesome project.

jimklimov, 2023-11-10:
Thanks for the kind words, much appreciated :)

jimklimov, 2024-12-18:
PR #2718 is presumed to have fixed the broken CPS input/output voltage reports mentioned in this issue (although it did not focus on cases of "voltage too high" like here).

@sblantipodi : you mentioned that this change https://github.com/networkupstools/nut/pull/1497/files#diff-935d101f3ea64117ab54f95b7de69a3c261a7ee0528a2563292ef9c8b4f66d5eR285 broke it for you? I'm curious now, after recent digging in nearby code.

Did you fix it for your build? How? Is that a code change applicable for upstreaming, or did you make use of e.g. `disable_fix_report_desc` flag setting?

As I read it, originally the descriptor fix (introduced by PR #1245 in NUTv2.8.0) applied to exactly VID=0x0764 and PID=0x0501, skipped for others. With this commit, it also applies to PID=0x0601, which your device is indeed.

The actual fix in #1245 (and updated in #2718) fires...
```
if (hvt_logmin == output_logmin && hvt_logmax == output_logmax)
```
...and adjusts LogMin/LogMax for those output readings (input too, with the newer fix), not the readings themselves directly.

So either the device lies with the original readings of actual values, or I misunderstand something else here :D 

Can you please custom-build the git master branch following https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests instructions to check if the current codebase misbehaves (and what debug-logs it spews then, to help track this down)?

KameDomotics, 2024-12-20:
I also have problems with the output_voltage sensor, I always get the wrong value, as I described here:

[https://github.com/hassio-addons/addon-nut/issues/383]()

jimklimov, 2024-12-20:
@KameDomotics : Are you using a build of current master branch with that fix from 2 days ago?

KameDomotics, 2024-12-21:
> @KameDomotics : Are you using a build of current master branch with that fix from 2 days ago?
> 
> 
> 
> Note: there's a Wiki article that purports to describe building your own HA plugin.

No, I am using the latest official version downloaded from Supervisor. How do I install yours? Which repository should I add?
Thanks

jimklimov, 2024-12-21:
No idea if anyone built it for HA yet, so you'd generally start from https://github.com/networkupstools/nut/wiki/Building-NUT-integration-for-Home-Assistant and https://github.com/networkupstools/nut/pull/2718#issuecomment-2541291296 - shake, mix, hash, serve..

KameDomotics, 2024-12-21:
> No idea if anyone built it for HA yet, so you'd generally start from https://github.com/networkupstools/nut/wiki/Building-NUT-integration-for-Home-Assistant and maybe revise https://github.com/networkupstools/nut/pull/2718#issuecomment-2541291296 about the recent intentional changes - shake, mix, hash, serve..

I don't have such advanced knowledge, I am not able to understand or fix the problem. 
I created a new template sensor.output_voltage_2 that adds +230V to the original sensor sensor_output_voltage. 
I am not able to do anything else with my skills