networkupstools/nut #1970: APC BK650M2-CH with onlinedischarge bug

dominicx, 2023-06-20:
First of all, everything works fine under nut 2.7.4. and I upgrade to 2.8.0-7 in debian 

upsc ups:
```
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 50
battery.mfr.date: 2001/01/01
battery.runtime: 3618
battery.runtime.low: 300
battery.type: PbAc
battery.voltage: 13.5
battery.voltage.nominal: 12.0
device.mfr: American Power Conversion
device.model: BK650M2-CH
device.serial: 9B2019A14274
device.type: ups
driver.name: usbhid-ups
driver.parameter.bus: 001
driver.parameter.pollfreq: 40
driver.parameter.pollinterval: 10
driver.parameter.port: auto
driver.parameter.product: BK650M2-CH
driver.parameter.productid: 0002
driver.parameter.serial: 9B2019A14274
driver.parameter.synchronous: auto
driver.parameter.vendor: American Power Conversion
driver.parameter.vendorid: 051D
driver.version: 2.8.0
driver.version.data: APC HID 0.98
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.24 (API: 0x1000108)
input.sensitivity: low
input.transfer.high: 278
input.transfer.low: 160
input.voltage: 232.0
input.voltage.nominal: 220
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.load: 5
ups.mfr: American Power Conversion
ups.mfr.date: 2020/05/08
ups.model: BK650M2-CH
ups.productid: 0002
ups.realpower.nominal: 390
ups.serial: xxxx
**ups.status: OL DISCHRG**
ups.test.result: Done and passed
ups.timer.reboot: 0
ups.timer.shutdown: -1
ups.vendorid: 051d
```

I got "Jun 20 18:03:30 pve usbhid-ups[2469246]: ups_status_set: seems that UPS [ups] is in OL+DISCHRG state now. Is it calibrating or do you perhaps want to set 'onlinedischarge' option? Some UPS models (e.g. CyberPower UT series) emit OL+DISCHRG when offline.
 in system log file.

I read the manual , and I find the 'onlinedischarge' settings and I add the setting to /etc/nut/ups.conf . and I got the message 'UPS ups@localhost on battery’.

the upsc ups gived me 'ups.status: OB DISCHRG'.

so, the APC BK650M2-CH ups  has been designed with ups.status: OL DISCHRG is online
and ups.status: OB DISCHRG is on battery.

dominicx, 2023-06-21:
Additional information:
usbhid-ups -DDD -u root -x explore -x vendorid=051d -x port=auto -s ups -d 1 2>&1 

[info.log](https://github.com/networkupstools/nut/files/11814733/info.log)

xsj684135, 2023-12-05:
老哥你解决了吗？我也是这么个情况 当充电到100时会触发ONBATT关机

jimklimov, 2023-12-05:
@dominicx : per #2212 these messages are sent when the `battery.charge` reaches 100% with a similar device - can you confirm that? e.g. if you discharge the UPS somewhat and plug the wall power back, are these messages gone until it gets fully recharged?

NUT v2.8.1 introduced a new config flag `onlinedischarge_calibration` for devices which use this `OL+DISCHRG` state-combo to report they are in fact calibrating; incidentally also APC's. So that is another thing to check for - but I suppose the charge would diminish over time as the UPS is discharging the battery while calibrating.

Perhaps it is time to add another such tweak, roughly speaking, for `OL+DISCHRG+battery.charge=100%` -- so when a well-fed UPS says it is online and yet discharging, we would not treat it as a problem to notify about. I can only guess it would be similar to charging strategies many phones/laptops/cars have nowadays, to keep batteries topped to just 90-95% of their capacity so the chemicals live longer.

PR #2216 (2023-12-10): `usbhid-ups`: allow to throttle or suppress logged messages about `OL+DISCHRG` situation
Closes: #1970
Closes: #2212
Closes: #2213
Closes: #2214
Closes: #2215

Introduces a number of new configurable toggles with hopefully reasonable defaults, which should help to reduce log noise on systems where an UPS reports "online" and "discharging" simultaneously (so far some CyberPower and APC models were reported to behave so, with different physical situations).