networkupstools/nut #1972: nut-monitor starts fine manually; will not start on boot.

Ramias1, 2023-06-24:
Trying to get NUT to work on Fedora 38.  UPS is plugged in to my QNAP NAS which acts as the NUT Server; I have the client already working on a Mac on that same UPS.  Trying to get this wrorking on Fedora.

/etc/ups/nut.conf has:
`MODE=netclient`

/etc/ups/upsmon.conf is all defaults plus this line:

`MONITOR qnapups@10.51.2.10 1 admin 123456 secondary`

After a reboot, **systemctl status nut-monitor** shows:
```
○ nut-monitor.service - Network UPS Tools - power device monitor and shutdown controller
     Loaded: loaded (/usr/lib/systemd/system/nut-monitor.service; enabled; preset: disabled)
    Drop-In: /usr/lib/systemd/system/service.d
             └─10-timeout-abort.conf
     Active: inactive (dead)
```

but I can then easily issue the start command (systemctl start nut-monitor), it starts, **upsc -c qnapups** on the QNAP shows it connected.

So the question: Why does this not work automatically on boot with Fedora 38?  I did have this working with Ubuntu but had to switch to Fedora for other reasons.

Thank you

Ramias1, 2023-06-24:
had to install full nut and enable nut.target; rebooted and now things work.

closed 2023-06-24 by Ramias1

jimklimov, 2023-06-24:
I wonder what is in 

````
Drop-In: /usr/lib/systemd/system/service.d
              └─10-timeout-abort.conf
````

My guess would be that it tweaks timeouts for starts, so if Fedora `upsmon` fails to connect to QNAP `upsd` (started too early, no networking yet, firewalls, etc), or otherwise does not come ibto the state where systemd considers it started, it gets killed off.

What did the service unit journal say on this?

Generally seems like a distro/packaging "problem", maybe worth bringing up in their tracker too at least to keep count of if it hits someone else.

Ramias1, 2023-06-28:
Issue was I was only enabling nut-monitor.service and not also nut.target