networkupstools/nut #1973: Output voltage is now wrong - off by about 220V

jonandel, 2023-06-25:
Somehint is underreportng the output voltage of my UPS - Im sure it used to be correct, when I set it up about 18 months ago...
See below - input voltage is 230V, output 10.

Im using NUT Tools in Homeassistant Hassos.

Homeassistanst community addon Nut Tools.

> ##NUT Tools config:
> - config: []
>   driver: usbhid-ups
>   name: myups
>   port: auto

##Home assistant Yaml showing NUT sensor for correct Powerwalker device.
> sensor:
>   # UPS PowerWalker Device
>   - platform: nut
>     name: "PowerWalker VI USB 6500"
>     host: a0d7b954-nut
>     username: !secret nutups_user
>     password: !secret nutups_password
>     resources:
>       - battery.charge
>       - battery.runtime
>       - ups.load
>       - ups.status

## Homeassistant Integration showing data from the Nut Tools device - note seems to get the vendor and device type OK...
"integration_manifest": {
"domain": "nut",
"name": "Network UPS Tools (NUT)",
"codeowners": [
"@bdraco",
"@ollo69",
"@pestevez"
],
"config_flow": true,
"documentation": "https://www.home-assistant.io/integrations/nut",
"integration_type": "device",
"iot_class": "local_polling",
"loggers": [
"pynut2"
],
"requirements": [
"pynut2==2.1.2"
],
"zeroconf": [
"_nut._tcp.local."
],
"is_built_in": true
},
"data": {
"entry": {
"entry_id": "96848fcb81c918f44c4b066b12ebe4e1",
"version": 1,
"domain": "nut",
"title": "a0d7b954-nut:3493",
"data": {
"host": "a0d7b954-nut",
"port": 3493,
"password": "REDACTED",
"username": "REDACTED"
},
"options": {
"scan_interval": 60
},
"pref_disable_new_entities": false,
"pref_disable_polling": false,
"source": "user",
"unique_id": null,
"disabled_by": null
},
"nut_data": {
"ups_list": {
"myups": "Description unavailable"
},
"status": {
"battery.charge": "100",
"battery.charge.low": "10",
"battery.charge.warning": "20",
"battery.mfr.date": "1 ",
"battery.runtime": "2520",
"battery.runtime.low": "300",
"battery.type": "PbAcid",
"battery.voltage": "13.7",
"battery.voltage.nominal": "12",
"device.mfr": "1 ",
"device.model": " 650",
"device.serial": " ",
"device.type": "ups",
"driver.name": "usbhid-ups",
"driver.parameter.pollfreq": "30",
"driver.parameter.pollinterval": "2",
"driver.parameter.port": "auto",
"driver.parameter.synchronous": "auto",
"driver.version": "2.8.0",
"driver.version.data": "CyberPower HID 0.6",
"driver.version.internal": "0.47",
"driver.version.usb": "libusb-1.0.26 (API: 0x1000109)",
"input.frequency": "50.2",
"input.transfer.high": "290",
"input.transfer.low": "165",
"input.voltage": "241.3",
"input.voltage.nominal": "230",
"output.frequency": "50.2",
"output.voltage": "10.9",
"ups.beeper.status": "enabled",
"ups.delay.shutdown": "20",
"ups.delay.start": "30",
"ups.load": "12",
"ups.mfr": "1 ",
"ups.model": " 650",
"ups.productid": "0601",
"ups.realpower.nominal": "360",
"ups.serial": " ",
"ups.status": "OL",
"ups.timer.shutdown": "-60",
"ups.timer.start": "-60",
"ups.vendorid": "0764"
}
},......

I posted details in the HA Core - suggestion there that it may be a NUT tools/driver issue...hence posting here for ideas...
https://github.com/home-assistant/core/issues/95157

jonandel, 2023-06-25:
I notice this in the addon log though - is it significant ?
[10:34:42] INFO: Configuring Device named myups...
[10:34:42] INFO: Starting the UPS drivers...
**Network UPS Tools - Generic HID driver 0.47 (2.8.0)
HIDParse: LogMax is less than LogMin. Vendor HID report descriptor may be incorrect; interpreting LogMax -1 as 65535 in ReportID: 0x0f
HIDParse: LogMax is less than LogMin. Vendor HID report descriptor may be incorrect; interpreting LogMax -1 as 255 in ReportID: 0x12
HIDParse: LogMax is less than LogMin. Vendor HID report descriptor may be incorrect; interpreting LogMax -1 as 255 in ReportID: 0x13**
Using subdriver: CyberPower HID 0.6
USB communication driver (libusb 1.0) 0.43
Network UPS Tools - UPS driver controller 2.8.0
cont-init: info: /etc/cont-init.d/nut.sh exited 0
cont-init: info: running /etc/cont-init.d/nutclient.sh
cont-init: info: /etc/cont-init.d/nutclient.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun upsd (no readiness notification)
services-up: info: copying legacy longrun upsmon (no readiness notification)
[10:34:42] INFO: Starting the UPS information server...

jimklimov, 2023-06-26:
Hello, thanks for the post. It *may* be significant - one of the changes in NUT v2.8.0 release was about detecting malformed protocol information and amending that (stemming from issue #1024 and later issues and PRs on this matter; another related thread is #439). Looking at related key words, I think #1969 is quite similar to yours (although there the bogus value is larger and does not change over time).

It is possible that some of that did misfire and better amendments could have been made, so in the NUT master branch there were some fixes later, e.g. #1338 or #1497

jonandel, 2023-07-01:
thanks - yes I saw several that were similar - interestingly #1512 reports the same problem but with a slightly different product - - mine is a 650 with no LCD, rather than a 500 with LCD.
However, I see NUT is reporting same vendor and product ID's....

Dunuin, 2023-08-05:
Same problem here with my PowerWalker STL 800. Worked perfectly fine with latest NUT shippen with Debian 11, then I upgraded Debian to 12 and the output voltage directly dropped from ~235V to ~5V and is now fluctuating between 2V and 10V for 3 weeks.

jimklimov, 2023-08-05:
Cheers, thanks for the reports. Are any of you in position to try custom-building NUT to query the device with current codebase (to check if this was solved after 2.8.0 release)? This can be done from a build workspace without compromising the packaged installation, see https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests

jonandel, 2023-08-19:
Im running HASOS, so will need to pull out a rPi and setup a test rig....give me a week or so - and Ill give it a go.

jonandel, 2023-08-21:
Jim,  Ive fired up a rPi and the master branch built (2.8.0.1)... which build do you suggest I try ?

jimklimov, 2023-08-21:
I think `master` should do it for tests. And then fiddle with driver options: what it does by default now, if it reports better again or still wrong as in 2.8.0 vs. 2.7.4; I think there was a toggle to disable the fix for presumed bad usb logical min/max encoding... can't remember quickly now - but should be in man pages (to see if that is the culprit).

jonandel, 2023-08-21:
Erm OK. Remember though, I'm seeing the error in 2.8.0 pulled through in the Home Assistant add in an if about a month ago...but I don't know which version was used previously (which I don't think had the issue)....
Let me play around for a few days..and ill come back....

jimklimov, 2023-08-22:
Thanks! Note that Home Assistant builds generally include a version of NUT from packaging, e.g. Debian's 2.7.4 or 2.8.0 release plus their tweaks and backports over time. So even a 2.8.0 is a bit over a year old, and some fixes have landed since, and maybe something among these impacts the handling of your device.

Thinking of it, you might want to build a custom HA plugin with current NUT. Never did that myself, but there is a contributed post on the Wiki about it.

jonandel, 2023-08-28:
Hey Jim,  Im not quite sure what Im doing frankly - but I *think* I have the mast branch installed and running.  (although tools such as nut-scanner complain about a missing libusb...
anywhooo... when I probe with usb-hid-ups driver it seems to be communicating with the USB, and I get the below output.

Is this any use? What should I do next ?

pi@raspberrypi:~/nut $ sudo drivers/usbhid-ups -DD -u root -x explore -x vendorid=0764 -x port=auto -s ups
Network UPS Tools - Generic HID driver 0.50 (2.8.0-2306-g3d9630e0f)
USB communication driver (libusb 0.1) 0.44
   0.000005     [D1] Built-in default or configured user for drivers 'nobody' was ignored due to 'root' specified on command line
   0.000155     [D1] Network UPS Tools version 2.8.0-2306-g3d9630e0f (release/snapshot of 2.8.0.1) built with gcc (Raspbian 10.2.1-6+rpi1) 10.2.1 20210110 and configured with flags: --with-dev
   0.000208     [D1] debug level is '2'
   0.002482     [D1] Succeeded to become_user(root): now UID=0 GID=0
   0.002653     [D2] Initializing an USB-connected UPS with library libusb-0.1 (or compat) (NUT subdriver name='USB communication driver (libusb 0.1)' ver='0.44')
   0.002711     [D1] upsdrv_initups (non-SHUT)...
   0.075475     [D2] Checking device (1D6B/0003) (002/001)
   0.119071     [D2] - VendorID: 1d6b
   0.119141     [D2] - ProductID: 0003
   0.119200     [D2] - Manufacturer: Linux 6.1.21-v8+ xhci-hcd
   0.119257     [D2] - Product: xHCI Host Controller
   0.119314     [D2] - Serial Number: 0000:01:00.0
   0.119370     [D2] - Bus: 002
   0.119426     [D2] - Device: 001
   0.119483     [D2] - Device release number: 0601
   0.119540     [D2] Trying to match device
   0.119598     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.119681     [D2] match_function_regex: failed match of VendorID: 1d6b
   0.119740     [D2] Device does not match - skipping
   0.120196     [D2] Checking device (0764/0601) (001/003)
   0.125553     [D2] - VendorID: 0764
   0.125620     [D2] - ProductID: 0601
   0.125675     [D2] - Manufacturer: 1
   0.125731     [D2] - Product:  650
   0.125788     [D2] - Serial Number:
   0.125844     [D2] - Bus: 001
   0.125900     [D2] - Device: 003
   0.125957     [D2] - Device release number: 0200
   0.126013     [D2] Trying to match device
   0.126070     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.126172     [D2] Device matches
   0.126262     [D2] failed to claim USB device: could not claim interface 0: Device or resource busy
   0.127452     [D2] detached kernel driver from USB device...
   0.128336     [D2] Retrieved HID descriptor (expected 9, got 9)
   0.128454     [D2] HID descriptor length 739
   0.133944     [D2] Report Descriptor size = 739
   0.134557     HIDParse: LogMax is less than LogMin. Vendor HID report descriptor may be incorrect; interpreting LogMax -1 as 65535 in ReportID: 0x0f
   0.135343     HIDParse: LogMax is less than LogMin. Vendor HID report descriptor may be incorrect; interpreting LogMax -1 as 255 in ReportID: 0x12
   0.135413     HIDParse: LogMax is less than LogMin. Vendor HID report descriptor may be incorrect; interpreting LogMax -1 as 255 in ReportID: 0x13
   0.135822     Using subdriver: EXPLORE HID 0.2
   0.136607     [D1] 73 HID objects found
   0.139079     [D1] Path: UPS.PowerSummary.iProduct, Type: Feature, ReportID: 0x01, Offset: 0, Size: 8, Value: 1
   0.141339     [D1] Path: UPS.PowerSummary.iSerialNumber, Type: Feature, ReportID: 0x02, Offset: 0, Size: 8, Value: 2
   0.144096     [D1] Path: UPS.PowerSummary.iOEMInformation, Type: Feature, ReportID: 0x03, Offset: 0, Size: 8, Value: 3
   0.146844     [D1] Path: UPS.PowerSummary.iDeviceChemistry, Type: Feature, ReportID: 0x04, Offset: 0, Size: 8, Value: 4
   0.149587     [D1] Path: UPS.PowerSummary.Rechargeable, Type: Feature, ReportID: 0x05, Offset: 0, Size: 8, Value: 1
   0.152337     [D1] Path: UPS.PowerSummary.CapacityMode, Type: Feature, ReportID: 0x06, Offset: 0, Size: 8, Value: 2
   0.155088     [D1] Path: UPS.PowerSummary.DesignCapacity, Type: Feature, ReportID: 0x07, Offset: 0, Size: 8, Value: 100
   0.155170     [D1] Path: UPS.PowerSummary.CapacityGranularity1, Type: Feature, ReportID: 0x07, Offset: 8, Size: 8, Value: 10
   0.155244     [D1] Path: UPS.PowerSummary.CapacityGranularity2, Type: Feature, ReportID: 0x07, Offset: 16, Size: 8, Value: 10
   0.155319     [D1] Path: UPS.PowerSummary.WarningCapacityLimit, Type: Feature, ReportID: 0x07, Offset: 24, Size: 8, Value: 20
   0.155483     [D1] Path: UPS.PowerSummary.FullChargeCapacity, Type: Feature, ReportID: 0x07, Offset: 32, Size: 8, Value: 100
   0.155685     [D1] Path: UPS.PowerSummary.RemainingCapacityLimit, Type: Feature, ReportID: 0x07, Offset: 40, Size: 8, Value: 10
   0.158093     [D1] Path: UPS.PowerSummary.RemainingCapacity, Type: Input, ReportID: 0x08, Offset: 0, Size: 8, Value: 100
   0.158177     [D1] Path: UPS.PowerSummary.RemainingCapacity, Type: Feature, ReportID: 0x08, Offset: 0, Size: 8, Value: 100
   0.158254     [D1] Path: UPS.PowerSummary.RunTimeToEmpty, Type: Input, ReportID: 0x08, Offset: 8, Size: 16, Value: 2520
   0.158368     [D1] Path: UPS.PowerSummary.RunTimeToEmpty, Type: Feature, ReportID: 0x08, Offset: 8, Size: 16, Value: 2520
   0.158569     [D1] Path: UPS.PowerSummary.RemainingTimeLimit, Type: Input, ReportID: 0x08, Offset: 24, Size: 16, Value: 300
[... cut ...]
  10.282173     [D1] Quick update...
  12.266427     [D1] upsdrv_updateinfo...
  12.283767     [D1] Got 3 HID objects...
  12.283872     [D2] Path: UPS.PowerSummary.RemainingCapacity, Type: Input, ReportID: 0x08, Offset: 0, Size: 8, Value: 100
  12.283965     [D2] Path: UPS.PowerSummary.RunTimeToEmpty, Type: Input, ReportID: 0x08, Offset: 8, Size: 16, Value: 2220
  12.284053     [D2] Path: UPS.PowerSummary.RemainingTimeLimit, Type: Input, ReportID: 0x08, Offset: 24, Size: 16, Value: 300
  12.284128     [D1] Quick update...

jonandel, 2023-08-28:
PS, any ideas why this doesnt work ? Whats the subdriver ?

(I used the quoted subdriver thats 'discovered' too - no luck 'EXPLORE HID 0.2'

pi@raspberrypi:~/nut $ sudo ./drivers/nutdrv_qx -s tempups -d1 -DDDDDD -x port=auto -x vendorid=0764 -x productid=0601 -x subdriver="CyberPower HID 0.6"
Network UPS Tools - Generic Q* USB/Serial driver 0.33 (2.8.0-2306-g3d9630e0f)
USB communication driver (libusb 0.1) 0.44
   0.000004     [D3] main_arg: var='port' val='auto'
   0.000341     [D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=auto, reloadable=0, reload_flag=0
   0.000397     [D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000463     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000937     [D3] main_arg: var='vendorid' val='0764'
   0.001016     [D5] send_to_all: SETINFO driver.parameter.vendorid "0764"
   0.001072     [D3] main_arg: var='productid' val='0601'
   0.001495     [D5] send_to_all: SETINFO driver.parameter.productid "0601"
   0.001553     [D3] main_arg: var='subdriver' val='CyberPower HID 0.6'
   0.001619     [D5] send_to_all: SETINFO driver.parameter.subdriver "CyberPower HID 0.6"
   0.002058     [D1] Network UPS Tools version 2.8.0-2306-g3d9630e0f (release/snapshot of 2.8.0.1) built with gcc (Raspbian 10.2.1-6+rpi1) 10.2.1 20210110 and configured with flags: --enable-check-NIT
   0.002124     [D1] debug level is '6'
   0.002186     [D5] send_to_all: SETINFO driver.debug "6"
   0.002243     [D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.005496     [D1] Succeeded to become_user(nobody): now UID=65534 GID=65534
   0.005588     [D5] send_to_all: SETINFO device.type "ups"
   0.005650     [D5] send_to_all: SETINFO driver.state "init.device"
   0.005703     [D1] upsdrv_initups...
   0.006128     Subdriver 'CyberPower HID 0.6' not found!
   0.006203     [D5] send_to_all: SETINFO driver.state "cleanup.exit"
pi@raspberrypi:~/nut $

jimklimov, 2023-08-29:
Try `usbhid-ups` without the `-x explore` flag (it discovers and reports known data points to help write new sub-drivers, but does not interpret them).

At least, according to the initial report the device was handled with `usbhid-ups` which is about a different protocol than what `nutdrv_qx` uses (that said, its subdriver would be like plain `cyberpower` - single token, without versions).

jonandel, 2023-08-31:
ok done... not sure it particularly helpful... gives the same as nutdrv_qx...:

pi@raspberrypi:~/nut $ sudo ./drivers/usbhid-ups -s 0764 -x port=auto
Network UPS Tools - Generic HID driver 0.50 (2.8.0-2306-g3d9630e0f)
USB communication driver (libusb 0.1) 0.44
HIDParse: LogMax is less than LogMin. Vendor HID report descriptor may be incorrect; interpreting LogMax -1 as 65535 in ReportID: 0x0f
HIDParse: LogMax is less than LogMin. Vendor HID report descriptor may be incorrect; interpreting LogMax -1 as 255 in ReportID: 0x12
HIDParse: LogMax is less than LogMin. Vendor HID report descriptor may be incorrect; interpreting LogMax -1 as 255 in ReportID: 0x13
Using subdriver: CyberPower HID 0.8

****

jonandel, 2023-08-31:
either way - it shows the current Master has the same problem as reported: ie
0.171086 [D1] Path: UPS.Input.Voltage, Type: Feature, ReportID: 0x0f, Offset: 0, Size: 16, Value: **245.5**
0.182088 [D1] Path: UPS.Output.Voltage, Type: Feature, ReportID: 0x12, Offset: 0, Size: 16, Value: **15.1**

So probably confirmed its still a problem !

jonandel, 2023-09-03:
@jimklimov Any ideas where to start looking ?  I see the cyberpower driver was updated about a year ago - but I dont see those HID Report desciptors having changed.
I pulled the patest mast yesterday - as I see a few changes to the cyberpower-mib recently.  I rebuilt, but still get the same incorrect reporting value for output voltage.
Also, in my original post, the sub-driver version is being reported as "driver.version.data": "CyberPower HID 0.6" - but current version here is 0.54 ... Where should I start looking ?

jimklimov, 2024-12-18:
PR #2718 is presumed to have fixed the broken CPS input/output voltage reports mentioned in this issue.

closed 2024-12-18 by jimklimov

jonandel, 2024-12-19:
great thats to all. I have to admit I gave up after tinkering for a while.
Now I only have to wait until Home Assistant Code team integrate these changes !

jimklimov, 2024-12-19:
There's a wiki article about building your own HA plugin... I don't use it yet, so can't help directly.