networkupstools/nut #1974: Is OR2200LCDRTXL2U supported?

apenngrace, 2023-06-29:
I found [issue #603](https://github.com/networkupstools/nut/issues/603) that suggests that CyberPower OR2200LCDRTXL2U is supported by nut.  However, I do not find that product in the product compatibility database.  When using TrueNAS Core (TrueNAS-13.0-U5.1), that model does not appear in the list of available UPS devices.  I tried selecting one that starts with OR22... but I am getting "Connection Refused" errors.

jimklimov, 2023-06-29:
Trail from that issue leads to fixes in powerpanel driver which were released as part of NUT 2.8.0. Does that match the version and driver you are trying?

apenngrace, 2023-06-30:
I checked and it looks like TrueNAS Core 13 is using NUT 2.7.4.  I guess that explains it.

closed 2023-06-30 by apenngrace