networkupstools/nut #1978: Problem with  NUTDRV_QX, subdriver  armac - unable start driver

bdkacz, 2023-07-02:
Hello

I have this UPS  http://armac.pl/oferta/ups-office-line-interactive-o850elcd/

I compiled the latest nut-2.8.0 from sources  from file [nut-2.8.0.tar.gz](https://www.networkupstools.org/source/2.8/nut-2.8.0.tar.gz)

Here is my lsusb output:

```
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 005 Device 002: ID 2357:0120 TP-Link Archer T2U PLUS [RTL8821AU]
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 002: ID 0925:1234 Lakeview Research UPS USB MON V1.4
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```

Then I try this settings in ups.conf :

```
[poddasze]
    driver = nutdrv_qx
    port   = auto
    vendorid = 0925
    productid = 1234
    subdriver ="armac"
    product = "UPS USB MON V1.4"
    bus = "006"

```

But when I try   upsdrvctl start I get:

```
Network UPS Tools - UPS driver controller 2.8.0
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43
 upsdrvctl startUsing protocol: Megatec 0.06
Can't initialise data from the UPS
Driver failed to start (exit status=1)

```
So I try to discover more and run   ./nutdrv_qx -DDD -a poddasze and get :

```
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D3] do_global_args: var='maxretry' val='3'
   0.000953     [D3] main_arg: var='driver' val='nutdrv_qx'
   0.001053     [D3] main_arg: var='port' val='auto'
   0.001157     [D3] main_arg: var='vendorid' val='0925'
   0.001255     [D3] main_arg: var='productid' val='1234'
   0.001351     [D3] main_arg: var='subdriver' val='armac'
   0.001448     [D3] main_arg: var='product' val='UPS USB MON V1.4'
   0.001536     [D3] main_arg: var='bus' val='006'
   0.001699     [D1] debug level is '3'
   0.003620     [D1] upsdrv_initups...
   0.085404     [D2] Checking device 1 of 10 (1D6B/0001)
   0.085827     [D1] Failed to open device (1D6B/0001), skipping: Access denied (insufficient permissions)
   0.085941     [D2] Checking device 2 of 10 (2357/0120)
   0.086079     [D1] Failed to open device (2357/0120), skipping: Access denied (insufficient permissions)
   0.086163     [D2] Checking device 3 of 10 (1D6B/0002)
   0.086289     [D1] Failed to open device (1D6B/0002), skipping: Access denied (insufficient permissions)
   0.086365     [D2] Checking device 4 of 10 (1D6B/0001)
   0.086485     [D1] Failed to open device (1D6B/0001), skipping: Access denied (insufficient permissions)
   0.086560     [D2] Checking device 5 of 10 (1D6B/0002)
   0.086683     [D1] Failed to open device (1D6B/0002), skipping: Access denied (insufficient permissions)
   0.086755     [D2] Checking device 6 of 10 (0925/1234)
   0.095368     [D2] - VendorID: 0925
   0.095425     [D2] - ProductID: 1234
   0.095456     [D2] - Manufacturer: ?
   0.095484     [D2] - Product: UPS USB MON V1.4
   0.095511     [D2] - Serial Number: unknown
   0.095542     [D2] - Bus: 006
   0.095568     [D2] - Device: unknown
   0.095593     [D2] - Device release number: 0001
   0.095618     [D2] Trying to match device
   0.095649     [D3] match_function_regex: matching a device...
   0.096184     [D2] Device matches
   0.096234     [D2] Reading first configuration descriptor
   0.096300     [D3] libusb_kernel_driver_active() returned 0
   0.096413     [D2] Claimed interface 0 successfully
   0.096457     [D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   0.096551     [D2] armac: unsupported cmd: QGS
   0.096596     [D2] qx_process_answer: short reply (input.voltage)
   0.096632     [D2] armac: unsupported cmd: QGS
   0.096664     [D2] qx_process_answer: short reply (input.voltage)
   0.096697     [D2] armac: unsupported cmd: QGS
   0.096774     [D2] qx_process_answer: short reply (input.voltage)
   0.096822     [D2] armac: unsupported cmd: M
   0.096859     [D2] voltronic_qs_protocol: invalid protocol [M]
   0.096891     [D2] armac: unsupported cmd: M
   0.096925     [D2] voltronic_qs_protocol: invalid protocol [M]
   0.096957     [D2] armac: unsupported cmd: M
   0.096990     [D2] voltronic_qs_protocol: invalid protocol [M]
   0.097022     [D2] armac: unsupported cmd: M
   0.097060     [D2] voltronic_qs_hex_protocol: invalid protocol [M]
   0.097092     [D2] armac: unsupported cmd: M
   0.097127     [D2] voltronic_qs_hex_protocol: invalid protocol [M]
   0.097160     [D2] armac: unsupported cmd: M
   0.097193     [D2] voltronic_qs_hex_protocol: invalid protocol [M]
   0.097225     [D2] armac: unsupported cmd: QS
   0.097258     [D2] qx_process_answer: short reply (input.voltage)
   0.097289     [D2] armac: unsupported cmd: QS
   0.097320     [D2] qx_process_answer: short reply (input.voltage)
   0.097350     [D2] armac: unsupported cmd: QS
   0.097381     [D2] qx_process_answer: short reply (input.voltage)
   0.097413     [D2] armac: unsupported cmd: D
   0.097444     [D2] qx_process_answer: short reply (input.voltage)
   0.097476     [D2] armac: unsupported cmd: D
   0.097507     [D2] qx_process_answer: short reply (input.voltage)
   0.097538     [D2] armac: unsupported cmd: D
   0.097568     [D2] qx_process_answer: short reply (input.voltage)
   1.306422     [D3] armac command Q1 response read: '(230.0 000.0 233.0 008 49.0 13.7 30.8 00001'
   2.494469     [D3] armac command ID response read: '#                           V2.63     '
   3.702449     [D3] armac command Q1 response read: '(231.0 000.0 232.0 008 49.0 13.7 30.8 00001'
   4.890439     [D3] armac command ID response read: '#                           V2.63     '
   6.098435     [D3] armac command Q1 response read: '(231.0 000.0 233.0 009 49.0 13.7 30.8 00001'
   7.287404     [D3] armac command ID response read: '#                           V2.63     '
   8.495400     [D3] armac command Q1 response read: '(228.0 000.0 231.0 008 49.0 13.7 30.8 00001'
   8.495640     [D2] armac: unsupported cmd: QPI
   8.495718     [D2] qx_process_answer: short reply (ups.firmware.aux)
   9.713366     [D3] armac command Q1 response read: '(229.0 000.0 231.0 008 49.0 13.7 30.8 00001'
   9.713591     [D2] armac: unsupported cmd: QPI
   9.713670     [D2] qx_process_answer: short reply (ups.firmware.aux)
  10.931370     [D3] armac command Q1 response read: '(229.0 000.0 230.0 009 50.0 13.7 30.8 00001'
  10.931597     [D2] armac: unsupported cmd: QPI
  10.931674     [D2] qx_process_answer: short reply (ups.firmware.aux)
  12.149365     [D3] armac command Q1 response read: '(231.0 000.0 233.0 009 49.0 13.7 30.8 00001'
  13.337335     [D3] armac command I response read: '#                           V2.63     '
  13.337526     Using protocol: Megatec 0.06
  13.337683     [D1] upsdrv_initinfo...
  14.555319     [D3] armac command Q1 response read: '(231.0 000.0 233.0 008 49.0 13.7 30.8 00001'
  14.555554     [D2] qx_process_answer: short reply (output.voltage)
  14.555647     [D2] qx_process_answer: short reply (ups.load)
  14.555724     [D2] qx_process_answer: short reply (input.frequency)
  14.555801     [D2] qx_process_answer: short reply (battery.voltage)
  14.555880     [D2] qx_process_answer: short reply (ups.temperature)
  14.555955     [D2] qx_process_answer: short reply (ups.status)
  14.556023     Can't initialise data from the UPS
  14.556091     [D1] upsdrv_cleanup...
```

So it get some data about voltages  , but not enought to start the driver :( 

Please help

jimklimov, 2023-07-05:
Hello, sorry to hear about that, perhaps your device is different from ones handled by the sub-driver (could be a case of "same ID, unrelated OEM").

FWIW, it was added in #1250 so possibly @blaa could troubleshoot this better. Could there be some bug in the new code?..

The subdriver comments suggest looking at `richcomm_usb`, and its replies to at least `Q1` give hope that it talks some variant of Megatec Qx protocols -- so my next suggestion would be to go "lock-picking" to try different subdrivers and protocols of `nutdrv_qx`.

bdkacz, 2023-07-05:
But the same UPS working ok,when I connect it to Asus router with asus-merlin firmware and nut installed via entware-ng ;)

jimklimov, 2023-07-05:
Ok, that is a very new piece of information, I believe :)

Can you check if they ship same or different NUT and driver versions? Looking at https://github.com/Entware/entware-ng-packages/blob/master/net/nut/Makefile I see it has 2.7.4 (so perhaps falls back to another subdriver/protocol?) and the repo is archived...

Also, if you built your own NUT - try just in case with the master branch. Some fixes did accumulate over the past year...

bdkacz, 2023-07-05:
Ok on armbian I will try to compile via git clone :)

About version on asus - actually I  re-flashed this router so I'm unable to check version of NUT . I copy 1:1 config files from asus to armbian and  as on first  post - it doesn't work. 

So i will check with version via git clone, and if no efftects  maybe I try build 2.7.4  on Armbian from sources

jimklimov, 2023-07-05:
Well, can't recommend 2.7.4 nowadays - it is over 7 years old now. My guess would rather be that if the UPS worked with it, there could be a different subdriver/protocol involved -- and you could try *its* 2.8.0+ version instead of `armac`.

At least, for the "1:1 config" part, the 2.7.4 driver would have probably refused to start with a then-unknown config option, I hope. At worst it might have fallen back to auto-detection (as if no option - which is another thing you can try), but I think it should have errored out.

bdkacz, 2023-07-05:
1:1 config I men copy driver name etc from old config, not copy 1:1 files :)

blaa, 2023-07-05:
`[D3] armac command Q1 response read: '(231.0 000.0 233.0 008 49.0 13.7 30.8 00001'`

Seems like a correctly parsed message.  (They get send in chunks of ~4 bytes in this protocol, so it means the proto mostly works fine).

The Armac subdriver did not parse the contents of this string though; only glued it together from USB and passed to already existing driver. That worked for my UPS/nut version.

Maybe something changed in Qx (?) driver that causes it to fail. 

Anywho - I can try to replicate on my hardware and make some fix, but it might take me 2 weeks to get started. Drop me a note if you manage to get it running (and how. ;d) or if you need some assistance. 

I'm unsure which version I have running, and I have a bit different UPS.

bdkacz, 2023-07-05:
So, after reinstall nut  via clone repository it works :)
With very simple config

```
[poddasze]
    driver = nutdrv_qx
    port   = auto

```

convicte, 2023-07-25:
Has there by any chance been progress on this?

I've purchased the 'new' Armac PF1 unit and tried getting it hooked up to UNRAID via NUT, but all three recommended drivers get me nowhere.

My `lsusb` output:
`Bus 001 Device 076: ID 0925:1234 Lakeview Research STD UPS MON V2.0`

On version 2023.07.24a for UNRAID I tried:
```
[ups]
    driver = nutdrv_qx
    port   = auto
    vendorid = "0925"
	productid = "1234"
    product = STD UPS MON V2.0
 ```
 and I get:
 
 ```
 Network UPS Tools - UPS driver controller 2.7.4.1
Network UPS Tools - Generic Q* USB/Serial driver 0.39 (2.7.4.1)
USB communication driver 0.43
No supported devices found. Please check your device availability with 'lsusb'
and make sure you have an up-to-date version of NUT. If this does not help,
try running the driver with at least 'subdriver', 'vendorid' and 'productid'
options specified. Please refer to the man page for details about these options
(man 8 nutdrv_qx).

Driver failed to start (exit status=1)
```
 
By trying to add `subdrivers = armac, I get the `not found` error:

```root@Server:~# upsdrvctl start
Network UPS Tools - UPS driver controller 2.7.4.1
Network UPS Tools - Generic Q* USB/Serial driver 0.39 (2.7.4.1)
USB communication driver 0.43
Subdriver 'armac' not found!
Driver failed to start (exit status=1)
```

It looks like it should have been added https://github.com/networkupstools/nut/blob/master/drivers/nutdrv_qx.c, but it obviously didn't or the Unraid version is not yet up to speed.

Any assistance would be greatly appreciated!

jimklimov, 2023-07-26:
@convicte : Hello again. What sort of "progress" here are you asking about? Driver is present, problem solved on NUT side. Poke the distro (unRaid) to package a newer NUT version.

Or am I missing something?

PS: your `product` value has spaces so should be in double quotes.

convicte, 2023-07-26:
Thank you, so very much for the prompt reply. I am not sure, we've interacted before, but I ask for so much assistance here and there to get my strange creations working, there is a good chance we did.

1) I must admit, I am a bit confused by how things are currently maintained, but is seams @SimonFair is the gracious host of the latest UNRAID NUT releases, and is pushing out a lot of updates with pending PRs
https://github.com/SimonFair/NUT-unRAID, regardless of the fact the original thread indeed points to a supper old NUT distro here https://forums.unraid.net/topic/60217-plugin-nut-v2-network-ups-tools/
[image]

I don't know honestly who is ahead of who, and thus I decided to start at the source. If you feel everything is in place for the vanilla NUT to have the `armac` driver working, I'll ask Simon then.

2) A side note - is there any reason this UPS, even though an `armac` brand would indicate in the NUT 'autodetect` function that it's a `richcomm` driver, which was the name of the company that wrote `Power Manager II` that is a software shipped with many of these UPSs. 
[image]
Is this just a false classification or am I barking at the wrong tree and when release, this driver will not work for me either.
BTW, when connected to this ancient software via a Win11 VM, I get the readings, so it's not dead, nor is the USB cable (from a 1997 printer) faulty, which I assumed to be the case.

Any assistance would be greatly appreciated!

blaa, 2023-07-26:
From your logs this looks like nut 2.7.4.1. 
2.8.0 was released a year ago and is available for example in Debian Stable (known for pretty old packages).

Unraid should upgrade it's base version.

convicte, 2023-07-26:
Working on that now!
Got completely lost in versioning and assumed I was using the latest, but now learned the UNRAID plugin has two branches, one based on 2.7.4.1 and another on 2.8.0, while latest PR are effective on both.
The one based on 2.7.4.1 is the main one in the community store, hence my logs.

convicte, 2023-07-26:
We've seemingly exhausted all the improvements required on the Unraid side to get the driver to work but still not joy.
The permission issue and other such problems were solved with some tinkering and the latest PRs, but the device is not recognized by the driver, which I believe puts me back on the NUT side of thing:

[image]

@blaa would you be able to point me to what I need to provide to receive assistance in that regard?
It pertains to this specific unit http://armac.pl/en/oferta/ups-rack-r3000ipf1-on-line/, which is what they indicate as their latest (came out 3-4 months ago) but is still fully recognized by the ancient Power Manager software just fine

convicte, 2023-07-26:
BTW, here is everything we've tested today: https://forums.unraid.net/topic/60217-plugin-nut-v2-network-ups-tools/page/32/

blaa, 2023-07-26:
So, let's sum up.
@bdkacz managed to get his UPS working just fine after installing new version.
@convicte is currently on 2.8.0 has device with matching USB definition that works with powermanager I used to write the armac driver.  It doesn't work on Unraid and no other devices/systems were tried. Unless this ancient software has some other mode of operation it should work fine. 

1) Make certain you tried subdriver = armac in config file (I've skimmed the unraid.net but I'm unsure you did, here certainly there's no output with it). 'product = ' line should not be necessary.

Recheck anyhow, because under 2.7.x this could not have worked.

2) Let's debug a big.

We here have pretty no idea what rc.nut start exactly does, so starting some more basic commands + current config file would certainly help. 

Try starting the driver as bdkacz did with -DDD:

 ./nutdrv_qx -DDD -a poddasze

Try to interpret the output, maybe fix something and paste here the result. 
-a marks a UPS name and needs to be tweaked to match the config.

2) Running it on Windows and dumping USB traffic with Wireguard could help make sure the protocol is the same, but lets leave that for later. Implementing driver without a device won't be easy anyway (but could suggest other driver).

convicte, 2023-07-26:
@blaa, thank you so much for the assistance thus far.
It took me some time to locate the driver file in the folder structure, but when I did, it all clicked together.

Here is the output of the `/usr/libexec/nut# ./nutdrv_qx -DDD -a ups` where `ups` is my config device name:

```
root@Server:/usr/libexec/nut# ./nutdrv_qx -DDD -a ups
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0-signed)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D3] main_arg: var='driver' val='nutdrv_qx'
   0.000014     [D3] main_arg: var='port' val='auto'
   0.000020     [D3] main_arg: var='vendorid' val='0925'
   0.000025     [D3] main_arg: var='productid' val='1234'
   0.000030     [D3] main_arg: var='subdriver' val='armac'
   0.000044     [D1] debug level is '3'
   0.168481     [D1] upsdrv_initups...
   0.169919     [D2] Checking device 1 of 9 (090C/1000)
   0.169934     [D1] Failed to open device (090C/1000), skipping: Access denied (insufficient permissions)
   0.169939     [D2] Checking device 2 of 9 (0578/0578)
   0.169945     [D1] Failed to open device (0578/0578), skipping: Access denied (insufficient permissions)
   0.169950     [D2] Checking device 3 of 9 (1D6B/0003)
   0.169957     [D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.169961     [D2] Checking device 4 of 9 (1D6B/0002)
   0.169966     [D1] Failed to open device (1D6B/0002), skipping: Access denied (insufficient permissions)
   0.169969     [D2] Checking device 5 of 9 (18D1/9302)
   0.169975     [D1] Failed to open device (18D1/9302), skipping: Access denied (insufficient permissions)
   0.169978     [D2] Checking device 6 of 9 (1D6B/0003)
   0.169983     [D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.169986     [D2] Checking device 7 of 9 (0925/1234)
   0.191051     [D2] - VendorID: 0925
   0.191067     [D2] - ProductID: 1234
   0.191072     [D2] - Manufacturer: unknown
   0.191074     [D2] - Product: STD UPS MON V2.0
   0.191081     [D2] - Serial Number: J8D5201395FBC47
   0.191084     [D2] - Bus: 001
   0.191086     [D2] - Device: unknown
   0.191088     [D2] - Device release number: 0001
   0.191091     [D2] Trying to match device
   0.191094     [D3] match_function_regex: matching a device...
   0.191127     [D2] Device matches
   0.191131     [D2] Reading first configuration descriptor
   0.191141     [D3] libusb_kernel_driver_active() returned 0
   0.191158     [D2] Claimed interface 0 successfully
   0.191162     [D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   0.191189     [D2] armac: unsupported cmd: QGS
   0.191200     [D2] qx_process_answer: short reply (input.voltage)
   0.191205     [D2] armac: unsupported cmd: QGS
   0.191208     [D2] qx_process_answer: short reply (input.voltage)
   0.191214     [D2] armac: unsupported cmd: QGS
   0.191216     [D2] qx_process_answer: short reply (input.voltage)
   0.191220     [D2] armac: unsupported cmd: M
   0.191228     [D2] voltronic_qs_protocol: invalid protocol [M]
   0.191234     [D2] armac: unsupported cmd: M
   0.191238     [D2] voltronic_qs_protocol: invalid protocol [M]
   0.191241     [D2] armac: unsupported cmd: M
   0.191244     [D2] voltronic_qs_protocol: invalid protocol [M]
   0.191247     [D2] armac: unsupported cmd: M
   0.191257     [D2] voltronic_qs_hex_protocol: invalid protocol [M]
   0.191260     [D2] armac: unsupported cmd: M
   0.191263     [D2] voltronic_qs_hex_protocol: invalid protocol [M]
   0.191269     [D2] armac: unsupported cmd: M
   0.191272     [D2] voltronic_qs_hex_protocol: invalid protocol [M]
   0.191275     [D2] armac: unsupported cmd: QS
   0.191288     [D2] qx_process_answer: short reply (input.voltage)
   0.191292     [D2] armac: unsupported cmd: QS
   0.191295     [D2] qx_process_answer: short reply (input.voltage)
   0.191299     [D2] armac: unsupported cmd: QS
   0.191306     [D2] qx_process_answer: short reply (input.voltage)
   0.191309     [D2] armac: unsupported cmd: D
   0.191317     [D2] qx_process_answer: short reply (input.voltage)
[... 35 lines cut ...]
  I am certain I tried with both generic configuration, as discussed above, and `subdriver:armac`
  Your last config matches with the one I tried several times and had the `Device not supported` error.
Funnily enough, the automatic matching seems to work and the red parameters of my device are the same as yours, where it comes to `productID` and `vendorID`.

Here is my config, though I constantly wonder if syntax matters here, such as indentation and "" around various variables, since even in this there you indicated yours differently than the original reported and both seem to work.

[image]

I've tried both with the entries line by line, and with this gap (since it's said to have been a bug which required this spacing to work).
[image]

[image]

Default passwords for the moment:

[image]

[image]

[image]

I hope this helps!
Please let me know if I could assist in any way.

Thank you in advance!!!

convicte, 2023-07-26:
I've increased debug logging even further  to -DDDD and there is clearly sensible data flowing, where I can spot the high input voltage around 245V, the output regulated voltage of 230V, the 50Hz frequency, where the rest is probable usage at 0 (since the draw is disconnected), etc. This said, the driver struggles to parse and process it correctly - though I have no idea why.

```
0.226990     [D2] qx_process_answer: short reply (input.voltage)
   0.226993     [D4] armac command Q1
   0.268745     [D4] read: ret 6 buf a6: 28 32 34 35 2e  >(245.<
   0.289742     [D4] read: ret 6 buf 86: 33 20 30 30 30  >3 000<
   0.310756     [D4] read: ret 6 buf a6: 2e 30 20 32 32  >.0 22<
   0.331749     [D4] read: ret 6 buf 86: 39 2e 36 20 30  >9.6 0<
   0.352745     [D4] read: ret 6 buf a6: 30 30 20 34 39  >00 49<
   0.373748     [D4] read: ret 6 buf 86: 2e 39 20 32 2e  >.9 2.<
   0.393752     [D4] read: ret 6 buf a6: 32 35 20 35 30  >25 50<
   0.414746     [D4] read: ret 6 buf 86: 2e 30 20 30 30  >.0 00<
   0.435756     [D4] read: ret 6 buf a6: 30 30 30 30 30  >00000<
000<.939819     [D4] read: ret 6 buf 83: 31 0d 30 30 30  >1
   1.940207     [D1] interrupt read error: Operation timed out (-7)
```

It's a really large data dump, so I moved it to a pastbin not to generate an even worse code wall:
https://pastebin.com/2DRprKpa

Looking forward to everyone's input.

blaa, 2023-07-26:
Good you tried -D; in the previous dump I was wondering why the Q1 command is not being tested at all. And well - it is executed, but at the end of reading, when the almost whole message is already read the driver gets `Operation timed out`.

Relevant part:
```c
        memset(buf, 0, buflen);

        bufpos = 0;
        while (bufpos + 6 < buflen) {
                size_t bytes_available;

                /* Read data in 6-byte chunks */
                ret = usb_interrupt_read(udev,
                        0x81,
                        (usb_ctrl_charbuf)tmpbuf, 6, 1000);

                /* Any errors here mean that we are unable to read a reply
                 * (which will happen after successfully writing a command
                 * to the UPS) */
                if (ret != 6) {
                        upsdebugx(1,  // <-- We end up here.
                                "interrupt read error: %s (%d)",
                                ret ? nut_usb_strerror(ret) : "timeout",
                                ret);
                        return ret;
                }

                upsdebugx(4,
                        "read: ret %d buf %02hhx: %02hhx %02hhx %02hhx %02hhx %02hhx  >%c%c%c%c%c<",
                        ret,
                        tmpbuf[0], tmpbuf[1], tmpbuf[2], tmpbuf[3], tmpbuf[4], tmpbuf[5],
                        tmpbuf[1], tmpbuf[2], tmpbuf[3], tmpbuf[4], tmpbuf[5]);

                bytes_available = (unsigned char)tmpbuf[0] & 0x0f;
                if (bytes_available == 0) {
                        /* End of transfer */
                        break;  // <-- This is not being hit (because supposedly there are still 3 bytes to be read)
                }

                memcpy(buf + bufpos, tmpbuf + 1, bytes_available);
                bufpos += bytes_available;

                if (bytes_available <= 2) { // <--- This won't be hit here (3), maaybe it should.
                        /* Slow down, let the UPS buffer more bytes */
                        usleep(15000);
                }
        }
```

convicte, 2023-07-26:
You seem to think I know what I am doing around code revision... a grave error on your part... lol
I really would like to help, but I wonder if the time needed to explain how to do it will be longer than a few iterations.

I am being honest when I am well over my head with something.
This is, unless you'd be so kind to invest some time in a real-time chat, to troubleshoot it this way.
We seem to both be from the same neck of the woods...

blaa, 2023-07-26:
We could try realtime talk tomorrow. Signal? Element/Matrix? Discord maybe? 

I've a simple idea for an additional check that would work with your UPS. But probably would have to build it myself and send you nutdrv_qx driver binary? Your device is a simple x86_64 for this unraid? Or something more exotic? 

Would be easier when I'm home and can test with a device.

This could be as easy as:
```
if (tmpbuf[bytes_available - 1] == 0x0d) {
    // This looks like end-of-line -> end of response.
    break;
}
```

But also interrupt timeout could be handled as "success" if some bytes were previously read.

convicte, 2023-07-26:
Discord works for me!
It's really annoying there is no messaging built into GitHub for sensitive information exchange.

I am working from home and on my PC, so have access to the hardware at a moment's notice.
Please let me know what works for you.

Thank you so very much for all the assistance!

blaa, 2023-07-26:
Here you have two binary ELF files (executables). Build on a Debian Stable.  
https://pliki.xyz/index.php/s/CNHZ9GPEiW2kaQe

The major problem might be that I have different library versions and they won't simply run on unraid. 
If they do though - we have a process to test.

convicte, 2023-07-26:
I am trying to put them into the right folder now. Lets see if I can run it the same way I did before just with the new file name.

blaa, 2023-07-27:
There are either a differences in protocol, or just protocol works as supposed but in this case the revengineered version falls short.

Following change seems to read the Q1 command just fine, but I think it won't work on other devices like this. ;) 
```
diff --git a/drivers/nutdrv_qx.c b/drivers/nutdrv_qx.c
index 350c555a0..70821c01f 100644
--- a/drivers/nutdrv_qx.c
+++ b/drivers/nutdrv_qx.c
@@ -1838,14 +1838,20 @@ static int      armac_command(const char *cmd, char *buf, size_t buflen)
                        tmpbuf[0], tmpbuf[1], tmpbuf[2], tmpbuf[3], tmpbuf[4], tmpbuf[5],
                        tmpbuf[1], tmpbuf[2], tmpbuf[3], tmpbuf[4], tmpbuf[5]);
 
+               /* This includes status/length byte on R/3000I/PF1 */
                bytes_available = (unsigned char)tmpbuf[0] & 0x0f;
                if (bytes_available == 0) {
                        /* End of transfer */
                        break;
                }
 
-               memcpy(buf + bufpos, tmpbuf + 1, bytes_available);
-               bufpos += bytes_available;
+               memcpy(buf + bufpos, tmpbuf + 1, bytes_available - 1);
+               bufpos += bytes_available - 1;
+
+               if (tmpbuf[bytes_available - 1] == 0x0d) {
+                       /* End of line is an end of the message as well */
+                       break;
+               }
 
                if (bytes_available <= 2) {
                        /* Slow down, let the UPS buffer more bytes */
```

Transmission sample:
```
   0.083301     [D4] armac command Q1
   0.164847     [D4] read: ret 6 buf a6: 28 32 34 31 2e  >(241.<
   0.184839     [D4] read: ret 6 buf 86: 35 20 30 30 30  >5 000<
   0.205851     [D4] read: ret 6 buf a6: 2e 30 20 32 33  >.0 23<
   0.226849     [D4] read: ret 6 buf 86: 30 2e 33 20 30  >0.3 0<
   0.247859     [D4] read: ret 6 buf a6: 30 30 20 34 39  >00 49<
   0.268862     [D4] read: ret 6 buf 86: 2e 39 20 32 2e  >.9 2.<
   0.289857     [D4] read: ret 6 buf a6: 32 35 20 34 38  >25 48<
   0.309866     [D4] read: ret 6 buf 86: 2e 30 20 30 30  >.0 00<
   0.330863     [D4] read: ret 6 buf a6: 30 30 30 30 30  >00000<
   0.827913     [D4] read: ret 6 buf 83: 31 0d 30 30 30  >1 000<
   0.827927     [D3] armac command Q1 response read: '(241.5 000.0 230.3 000 49.9 2.25 48.0 00000001'
   0.827954     [D4] armac command ID
  1.394985     [D4] read: ret 6 buf a5: 4e 41 4b 0d 30  >NAK  <
   1.395001     [D3] armac command ID response read: 'NAK'
```

When I'm home I can test it against my own and create a patch that should work on both just fine.

@bdkacz if you would be so kind to try to run the nutdrv_qx with -DDDD and paste your transmission fragment related to the Q1 command I'd be grateful. ;-) More examples, the better.

desertwitch, 2023-07-28:
We've since managed to get it working with @blaa 's patch for @convicte 🥇

jimklimov, 2023-07-28:
Would you please post a pull request so it also works after NUT upgrades for you and others? ;)

jimklimov, 2023-07-31:
Cheers, posted the PR with patch copied from above, moving the programming part of discussion there to #2003 - got a question...

Novocaine85, 2023-08-04:
I was wondering how this is even possible.
I was using this driver and subdriver on top of my proxmox machine. After a little bit of strugling I managed to make it work. 
I did an update to my machine and since there was a kernel updated a reboot was needed.
After the reboot I was no longer able to start nut-server, having the same exact issue as @bdkacz.

Tomorrow night I have a graveyard shift and i would like to try to build @blaa version in order to test it and see if it works.

really wired.

blaa, 2023-08-05:
@Novocaine85 which UPS do you have? This was reverse engineered protocol based on my device solely. It's nice that currently it's corrected with a debug info from other devices. Hope it works nice, and if not - send me -DDDD outputs from nutdrv_qx driver. ;)

Novocaine85, 2023-08-05:
> @Novocaine85 which UPS do you have? This was reverse engineered protocol based on my device solely. It's nice that currently it's corrected with a debug info from other devices. Hope it works nice, and if not - send me -DDDD outputs from nutdrv_qx driver. ;)

My UPS is a Vultech V2000 (https://www.vultech.it/en/serie-pro/640-ups-2000va-line-interactive-lcd-gruppo-di-continuita-8057284627203.html). 

Now: this is what i got from the UPS launching it from NUT installed from apt (debian bookworm, v2.8.0)
```
root@proxmox:~# /lib/nut/nutdrv_qx -a vultech -u root -D -D -D -x productid=1234 -x vendorid=0925 -x subdriver=armac
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D3] main_arg: var='productid' val='1234'
   0.000006     [D3] main_arg: var='vendorid' val='0925'
   0.000008     [D3] main_arg: var='subdriver' val='armac'
   0.000009     [D1] debug level is '3'
   0.000181     [D1] upsdrv_initups...
   0.001722     [D2] Checking device 1 of 9 (2109/0822)
   0.127399     [D2] - VendorID: 2109
   0.127408     [D2] - ProductID: 0822
   0.127409     [D2] - Manufacturer: VIA Labs, Inc.         
   0.127411     [D2] - Product: USB3.1 Hub             
   0.127412     [D2] - Serial Number: 000000001
   0.127413     [D2] - Bus: 002
   0.127414     [D2] - Device: unknown
   0.127416     [D2] - Device release number: 9013
   0.127417     [D2] Trying to match device
   0.127419     [D3] match_function_regex: matching a device...
   0.127468     [D2] match_function_regex: failed match of VendorID: 2109
   0.127470     [D2] Device does not match - skipping
   0.155436     [D2] Checking device 2 of 9 (1D6B/0003)
   0.155897     [D2] - VendorID: 1d6b
   0.155902     [D2] - ProductID: 0003
   0.155903     [D2] - Manufacturer: Linux 6.2.16-6-pve xhci-hcd
   0.155904     [D2] - Product: xHCI Host Controller
   0.155905     [D2] - Serial Number: 0000:00:14.0
   0.155906     [D2] - Bus: 002
   0.155907     [D2] - Device: unknown
   0.155908     [D2] - Device release number: 0602
   0.155909     [D2] Trying to match device
   0.155910     [D3] match_function_regex: matching a device...
   0.155913     [D2] match_function_regex: failed match of VendorID: 1d6b
   0.155914     [D2] Device does not match - skipping
   0.156226     [D2] Checking device 3 of 9 (1A86/7523)
   0.156656     [D2] - VendorID: 1a86
   0.156661     [D2] - ProductID: 7523
   0.156662     [D2] - Manufacturer: unknown
   0.156664     [D2] - Product: USB Serial
   0.156665     [D2] - Serial Number: unknown
   0.156666     [D2] - Bus: 001
   0.156667     [D2] - Device: unknown
   0.156669     [D2] - Device release number: 0264
   0.156670     [D2] Trying to match device
   0.156671     [D3] match_function_regex: matching a device...
   0.156674     [D2] match_function_regex: failed match of VendorID: 1a86
   0.156675     [D2] Device does not match - skipping
   0.156679     [D2] Checking device 4 of 9 (2109/2822)
   0.157912     [D2] - VendorID: 2109
   0.157919     [D2] - ProductID: 2822
   0.157920     [D2] - Manufacturer: VIA Labs, Inc.         
   0.157921     [D2] - Product: USB2.0 Hub             
   0.157923     [D2] - Serial Number: 000000001
   0.157924     [D2] - Bus: 001
   0.157925     [D2] - Device: unknown
   0.157926     [D2] - Device release number: 9013
   0.157928     [D2] Trying to match device
   0.157929     [D3] match_function_regex: matching a device...
   0.157932     [D2] match_function_regex: failed match of VendorID: 2109
   0.157933     [D2] Device does not match - skipping
   0.157939     [D2] Checking device 5 of 9 (0925/1234)
   0.160593     [D2] - VendorID: 0925
   0.160596     [D2] - ProductID: 1234
   0.160598     [D2] - Manufacturer: ?
[... 48 lines cut ...]
   3.083846     [D3] armac command Q1 response read: '(232.0 000.0 231.0 017 50.0 27.0 20.8 00001'
   3.083870     [D2] armac: unsupported cmd: QPI
   3.083873     [D2] qx_process_answer: short reply (ups.firmware.aux)
   4.292133     [D3] armac command Q1 response read: '(231.0 000.0 230.0 019 49.0 27.0 20.8 00001'
   4.292147     [D2] armac: unsupported cmd: QPI
   4.292149     [D2] qx_process_answer: short reply (ups.firmware.aux)
   5.500065     [D3] armac command Q1 response read: '(232.0 000.0 231.0 017 49.0 27.0 20.8 00001'
   5.500085     [D2] armac: unsupported cmd: QPI
   5.500088     [D2] qx_process_answer: short reply (ups.firmware.aux)
   6.708159     [D3] armac command Q1 response read: '(232.0 000.0 231.0 017 50.0 27.0 20.8 00001'
   7.896436     [D3] armac command I response read: '#                           V4.10     '
   7.896448     Using protocol: Megatec 0.06
   7.896454     [D1] upsdrv_initinfo...
   9.104342     [D3] armac command Q1 response read: '(232.0 000.0 231.0 019 50.0 27.0 20.8 00001'
   9.104369     [D2] qx_process_answer: short reply (output.voltage)
   9.104372     [D2] qx_process_answer: short reply (ups.load)
   9.104374     [D2] qx_process_answer: short reply (input.frequency)
   9.104376     [D2] qx_process_answer: short reply (battery.voltage)
   9.104378     [D2] qx_process_answer: short reply (ups.temperature)
   9.104380     [D2] qx_process_answer: short reply (ups.status)
   9.104382     Can't initialise data from the UPS
   9.104384     [D1] upsdrv_cleanup...
```

now i'll try to build your version.

Novocaine85, 2023-08-05:
```
root@proxmox:/usr/local/ups/bin# /usr/local/ups/bin/nutdrv_qx -a vultech -u root -D -D -D -x productid=1234 -x vendorid=0925 -x subdriver=armac
Network UPS Tools - Generic Q* USB/Serial driver 0.33 (2.8.0-2272-g3a11558e6)
USB communication driver (libusb 0.1) 0.44
   0.000000     [D3] main_arg: var='productid' val='1234'
   0.000007     [D3] main_arg: var='vendorid' val='0925'
   0.000009     [D3] main_arg: var='subdriver' val='armac'
   0.000011     [D1] Network UPS Tools version 2.8.0-2272-g3a11558e6 (release/snapshot of 2.8.0.1) built with gcc (Debian 12.2.0-14) 12.2.0 and configured with flags: --sysconfdir=/etc/nut --with-user=ups --with-group=root --with-usb
   0.000013     [D1] debug level is '3'
   0.000184     [D1] Succeeded to become_user(root): now UID=0 GID=0
   0.000190     [D1] upsdrv_initups...
   0.536371     [D3] usb_busses=0x557fed8970a0
   0.536385     [D2] Checking device (1D6B/0003) (004/001)
   0.703493     [D2] - VendorID: 1d6b
   0.703500     [D2] - ProductID: 0003
   0.703501     [D2] - Manufacturer: Linux 6.2.16-6-pve xhci-hcd
   0.703502     [D2] - Product: xHCI Host Controller
   0.703503     [D2] - Serial Number: 0000:00:0d.0
   0.703504     [D2] - Bus: 004
   0.703505     [D2] - Device: 001
   0.703506     [D2] - Device release number: 0602
   0.703506     [D2] Trying to match device
   0.703508     [D3] match_function_regex: matching a device...
   0.703511     [D2] match_function_regex: failed match of VendorID: 1d6b
   0.703512     [D2] Device does not match - skipping
   0.703536     [D2] Checking device (1D6B/0002) (003/001)
   0.703560     [D2] - VendorID: 1d6b
   0.703561     [D2] - ProductID: 0002
   0.703562     [D2] - Manufacturer: Linux 6.2.16-6-pve xhci-hcd
   0.703563     [D2] - Product: xHCI Host Controller
   0.703564     [D2] - Serial Number: 0000:00:0d.0
   0.703564     [D2] - Bus: 003
   0.703565     [D2] - Device: 001
   0.703566     [D2] - Device release number: 0602
   0.703567     [D2] Trying to match device
   0.703568     [D3] match_function_regex: matching a device...
   0.703569     [D2] match_function_regex: failed match of VendorID: 1d6b
   0.703570     [D2] Device does not match - skipping
   0.703581     [D2] Checking device (2109/0822) (002/002)
   0.831027     [D2] - VendorID: 2109
   0.831037     [D2] - ProductID: 0822
   0.831038     [D2] - Manufacturer: VIA Labs, Inc.         
   0.831040     [D2] - Product: USB3.1 Hub             
   0.831041     [D2] - Serial Number: 000000001
   0.831042     [D2] - Bus: 002
   0.831043     [D2] - Device: 002
   0.831045     [D2] - Device release number: 9013
   0.831046     [D2] Trying to match device
   0.831047     [D3] match_function_regex: matching a device...
   0.831063     [D2] match_function_regex: failed match of VendorID: 2109
   0.831065     [D2] Device does not match - skipping
   0.859295     [D2] Checking device (1D6B/0003) (002/001)
   0.859783     [D2] - VendorID: 1d6b
   0.859788     [D2] - ProductID: 0003
   0.859789     [D2] - Manufacturer: Linux 6.2.16-6-pve xhci-hcd
   0.859794     [D2] - Product: xHCI Host Controller
   0.859795     [D2] - Serial Number: 0000:00:14.0
   0.859796     [D2] - Bus: 002
   0.859797     [D2] - Device: 001
   0.859798     [D2] - Device release number: 0602
   0.859799     [D2] Trying to match device
   0.859800     [D3] match_function_regex: matching a device...
   0.859803     [D2] match_function_regex: failed match of VendorID: 1d6b
   0.859804     [D2] Device does not match - skipping
   0.860093     [D2] Checking device (1A86/7523) (001/005)
   0.860567     [D2] - VendorID: 1a86
   0.860573     [D2] - ProductID: 7523
   0.860574     [D2] - Manufacturer: unknown
   0.860576     [D2] - Product: USB Serial
   0.860577     [D2] - Serial Number: unknown
[... 82 lines cut ...]
   3.791914     [D3] armac command Q1 response read: '(229.0 000.0 229.0 013 50.0 27.0 20.8 00001'
   3.791939     [D2] armac: unsupported cmd: QPI
   3.791941     [D2] qx_process_answer: short reply (ups.firmware.aux)
   4.999937     [D3] armac command Q1 response read: '(232.0 000.0 232.0 014 49.0 27.0 20.8 00001'
   4.999952     [D2] armac: unsupported cmd: QPI
   4.999953     [D2] qx_process_answer: short reply (ups.firmware.aux)
   6.208120     [D3] armac command Q1 response read: '(233.0 000.0 233.0 014 50.0 27.0 20.8 00001'
   6.208135     [D2] armac: unsupported cmd: QPI
   6.208138     [D2] qx_process_answer: short reply (ups.firmware.aux)
   7.415341     [D3] armac command Q1 response read: '(232.0 000.0 232.0 014 50.0 27.0 20.8 00001'
   8.603255     [D3] armac command I response read: '#                           V4.10     '
   8.603271     Using protocol: Megatec 0.06
   8.603280     [D1] upsdrv_initinfo...
   9.811333     [D3] armac command Q1 response read: '(232.0 000.0 232.0 013 50.0 27.0 20.8 00001'
   9.811348     [D2] qx_process_answer: short reply (output.voltage)
   9.811349     [D2] qx_process_answer: short reply (ups.load)
   9.811351     [D2] qx_process_answer: short reply (input.frequency)
   9.811352     [D2] qx_process_answer: short reply (battery.voltage)
   9.811353     [D2] qx_process_answer: short reply (ups.temperature)
   9.811355     [D2] qx_process_answer: short reply (ups.status)
   9.811356     Can't initialise data from the UPS
   9.811359     [D1] upsdrv_cleanup...
```

what's wrong?

blaa, 2023-08-05:
Add another -D please! With four, it shows the messages as they are read from the UPS. I see no initial "cleaning" messeges, but maybe that's because of 3x D.

In general it failed because 3 initial Q1 queries failed, but the reason can't be seen.

Novocaine85, 2023-08-05:
```
root@proxmox:~# /usr/local/ups/bin/nutdrv_qx -a vultech -u root -D -D -D -D -x productid=1234 -x vendorid=0925 -x subdriver=armac
Network UPS Tools - Generic Q* USB/Serial driver 0.33 (2.8.0-2272-g3a11558e6)
USB communication driver (libusb 0.1) 0.44
   0.000000     [D3] main_arg: var='productid' val='1234'
   0.000008     [D3] main_arg: var='vendorid' val='0925'
   0.000010     [D3] main_arg: var='subdriver' val='armac'
   0.000012     [D1] Network UPS Tools version 2.8.0-2272-g3a11558e6 (release/snapshot of 2.8.0.1) built with gcc (Debian 12.2.0-14) 12.2.0 and configured with flags: --sysconfdir=/etc/nut --with-user=ups --with-group=root --with-usb
   0.000013     [D1] debug level is '4'
   0.000200     [D1] Succeeded to become_user(root): now UID=0 GID=0
   0.000206     [D1] upsdrv_initups...
   0.701533     [D3] usb_busses=0x55c8db8550a0
   0.701544     [D2] Checking device (1D6B/0003) (004/001)
   0.868439     [D2] - VendorID: 1d6b
   0.868447     [D2] - ProductID: 0003
   0.868448     [D2] - Manufacturer: Linux 6.2.16-6-pve xhci-hcd
   0.868448     [D2] - Product: xHCI Host Controller
   0.868449     [D2] - Serial Number: 0000:00:0d.0
   0.868450     [D2] - Bus: 004
   0.868451     [D2] - Device: 001
   0.868452     [D2] - Device release number: 0602
   0.868453     [D2] Trying to match device
   0.868454     [D3] match_function_regex: matching a device...
   0.868458     [D2] match_function_regex: failed match of VendorID: 1d6b
   0.868459     [D2] Device does not match - skipping
   0.868494     [D2] Checking device (1D6B/0002) (003/001)
   0.868659     [D2] - VendorID: 1d6b
   0.868664     [D2] - ProductID: 0002
   0.868665     [D2] - Manufacturer: Linux 6.2.16-6-pve xhci-hcd
   0.868666     [D2] - Product: xHCI Host Controller
   0.868667     [D2] - Serial Number: 0000:00:0d.0
   0.868668     [D2] - Bus: 003
   0.868669     [D2] - Device: 001
   0.868670     [D2] - Device release number: 0602
   0.868671     [D2] Trying to match device
   0.868672     [D3] match_function_regex: matching a device...
   0.868674     [D2] match_function_regex: failed match of VendorID: 1d6b
   0.868675     [D2] Device does not match - skipping
   0.868686     [D2] Checking device (2109/0822) (002/002)
   0.996306     [D2] - VendorID: 2109
   0.996312     [D2] - ProductID: 0822
   0.996313     [D2] - Manufacturer: VIA Labs, Inc.         
   0.996314     [D2] - Product: USB3.1 Hub             
   0.996315     [D2] - Serial Number: 000000001
   0.996316     [D2] - Bus: 002
   0.996316     [D2] - Device: 002
   0.996317     [D2] - Device release number: 9013
   0.996318     [D2] Trying to match device
   0.996319     [D3] match_function_regex: matching a device...
   0.996331     [D2] match_function_regex: failed match of VendorID: 2109
   0.996333     [D2] Device does not match - skipping
   1.024269     [D2] Checking device (1D6B/0003) (002/001)
   1.024762     [D2] - VendorID: 1d6b
   1.024767     [D2] - ProductID: 0003
   1.024768     [D2] - Manufacturer: Linux 6.2.16-6-pve xhci-hcd
   1.024769     [D2] - Product: xHCI Host Controller
   1.024770     [D2] - Serial Number: 0000:00:14.0
   1.024771     [D2] - Bus: 002
   1.024772     [D2] - Device: 001
   1.024773     [D2] - Device release number: 0602
   1.024774     [D2] Trying to match device
   1.024775     [D3] match_function_regex: matching a device...
   1.024777     [D2] match_function_regex: failed match of VendorID: 1d6b
   1.024778     [D2] Device does not match - skipping
   1.025097     [D2] Checking device (1A86/7523) (001/005)
   1.025903     [D2] - VendorID: 1a86
   1.025908     [D2] - ProductID: 7523
   1.025910     [D2] - Manufacturer: unknown
   1.025911     [D2] - Product: USB Serial
   1.025913     [D2] - Serial Number: unknown
[... 254 lines cut ...]
   8.383722     [D4] read: ret 6 buf 81: 20 33 2e 30 30  > 3.00<
   8.399664     [D4] read: ret 6 buf c1: 30 33 2e 30 30  >03.00<
   8.415653     [D4] read: ret 6 buf 83: 31 34 20 30 30  >14 00<
   8.416818     [D4] read: ret 6 buf c4: 34 39 2e 30 30  >49.00<
   8.421614     [D4] read: ret 6 buf 81: 20 39 2e 30 30  > 9.00<
   8.437632     [D4] read: ret 6 buf c1: 32 39 2e 30 30  >29.00<
   8.453569     [D4] read: ret 6 buf 83: 37 2e 30 30 30  >7.000<
   8.454636     [D4] read: ret 6 buf c4: 20 32 30 2e 30  > 20.0<
   8.458666     [D4] read: ret 6 buf 81: 38 32 30 2e 30  >820.0<
   8.474579     [D4] read: ret 6 buf c1: 20 32 30 2e 30  > 20.0<
   8.490705     [D4] read: ret 6 buf 83: 30 30 30 2e 30  >000.0<
   8.491750     [D4] read: ret 6 buf c4: 30 31 00 30 30  >01
   8.496653     [D4] read: ret 6 buf 81: 31 31 00 30 30  >11
1  8.512721     [D4] read: ret 6 buf c1: 0d 31 00 30 30  >
1  9.486714     [D4] read: ret 6 buf 80: 0d 31 00 30 30  >
   9.486727     [D3] armac command Q1 response read: '(233.0 000.0 233.0 014 49.0 27.0 20.8 00001'
   9.486740     [D2] qx_process_answer: short reply (output.voltage)
   9.486743     [D2] qx_process_answer: short reply (ups.load)
   9.486744     [D2] qx_process_answer: short reply (input.frequency)
   9.486747     [D2] qx_process_answer: short reply (battery.voltage)
   9.486749     [D2] qx_process_answer: short reply (ups.temperature)
   9.486750     [D2] qx_process_answer: short reply (ups.status)
   9.486752     Can't initialise data from the UPS
   9.486759     [D1] upsdrv_cleanup...
```

blaa, 2023-08-05:
Are you certain this has the latest changes? It reports 2.8.0-2272-g3a11558e6 - the changes are not yet merged. Latest are here: https://github.com/blaa/nut.git 

PR should land shortly, but tests like this will help. ;)

The problem I see in your log seems like something that SHOULD HAVE BEEN FIXED by this last PR, so if it's not, I'll need to dive in, but we must check if your version is right.

Novocaine85, 2023-08-05:
I confirm I've git cloned your version.

Novocaine85, 2023-08-05:
```
root@proxmox:/tmp/nut# /usr/local/ups/bin/nutdrv_qx -a vultech -u root -D -D -D -D -D -x productid=1234 -x vendorid=0925 -x subdriver=armac
Network UPS Tools - Generic Q* USB/Serial driver 0.33 (2.7.4-7674-gdb0272a94)
USB communication driver (libusb 0.1) 0.44
   0.000000     [D3] main_arg: var='productid' val='1234'
   0.000009     [D5] send_to_all: SETINFO driver.parameter.productid "1234"
   0.000012     [D3] main_arg: var='vendorid' val='0925'
   0.000014     [D5] send_to_all: SETINFO driver.parameter.vendorid "0925"
   0.000016     [D3] main_arg: var='subdriver' val='armac'
   0.000020     [D5] send_to_all: SETINFO driver.parameter.subdriver "armac"
   0.000022     [D1] Network UPS Tools version 2.7.4-7674-gdb0272a94 (release/snapshot of 2.8.0.1) built with gcc (Debian 12.2.0-14) 12.2.0 and configured with flags: --sysconfdir=/etc/nut --with-user=ups --with-group=root --with-usb
   0.000025     [D1] debug level is '5'
   0.000027     [D5] send_to_all: SETINFO driver.debug "5"
   0.000030     [D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.000218     [D1] Succeeded to become_user(root): now UID=0 GID=0
   0.000224     [D5] send_to_all: SETINFO device.type "ups"
   0.000226     [D5] send_to_all: SETINFO driver.state "init.device"
   0.000227     [D1] upsdrv_initups...
   0.692148     [D3] usb_busses=0x5642c4fc8000
   0.692159     [D2] Checking device (1D6B/0003) (004/001)
   0.862847     [D2] - VendorID: 1d6b
   0.862856     [D2] - ProductID: 0003
   0.862858     [D2] - Manufacturer: Linux 6.2.16-6-pve xhci-hcd
   0.862859     [D2] - Product: xHCI Host Controller
   0.862860     [D2] - Serial Number: 0000:00:0d.0
   0.862860     [D2] - Bus: 004
   0.862864     [D2] - Device: 001
   0.862865     [D2] - Device release number: 0602
   0.862866     [D2] Trying to match device
   0.862867     [D3] match_function_regex: matching a device...
   0.862877     [D2] match_function_regex: failed match of VendorID: 1d6b
   0.862878     [D2] Device does not match - skipping
   0.863932     [D2] Checking device (1D6B/0002) (003/001)
   1.027435     [D2] - VendorID: 1d6b
   1.027442     [D2] - ProductID: 0002
   1.027444     [D2] - Manufacturer: Linux 6.2.16-6-pve xhci-hcd
   1.027446     [D2] - Product: xHCI Host Controller
   1.027448     [D2] - Serial Number: 0000:00:0d.0
   1.027450     [D2] - Bus: 003
   1.027452     [D2] - Device: 001
   1.027453     [D2] - Device release number: 0602
   1.027456     [D2] Trying to match device
   1.027457     [D3] match_function_regex: matching a device...
   1.027461     [D2] match_function_regex: failed match of VendorID: 1d6b
   1.027462     [D2] Device does not match - skipping
   1.027542     [D2] Checking device (2109/0822) (002/002)
   1.158140     [D2] - VendorID: 2109
   1.158164     [D2] - ProductID: 0822
   1.158165     [D2] - Manufacturer: VIA Labs, Inc.         
   1.158166     [D2] - Product: USB3.1 Hub             
   1.158167     [D2] - Serial Number: 000000001
   1.158168     [D2] - Bus: 002
   1.158169     [D2] - Device: 002
   1.158169     [D2] - Device release number: 9013
   1.158170     [D2] Trying to match device
   1.158172     [D3] match_function_regex: matching a device...
   1.158182     [D2] match_function_regex: failed match of VendorID: 2109
   1.158183     [D2] Device does not match - skipping
   1.186831     [D2] Checking device (1D6B/0003) (002/001)
   1.186894     [D2] - VendorID: 1d6b
   1.186895     [D2] - ProductID: 0003
   1.186896     [D2] - Manufacturer: Linux 6.2.16-6-pve xhci-hcd
   1.186897     [D2] - Product: xHCI Host Controller
   1.186898     [D2] - Serial Number: 0000:00:14.0
   1.186899     [D2] - Bus: 002
   1.186900     [D2] - Device: 001
   1.186901     [D2] - Device release number: 0602
   1.186902     [D2] Trying to match device
   1.186903     [D3] match_function_regex: matching a device...
   1.186906     [D2] match_function_regex: failed match of VendorID: 1d6b
[... 667 lines cut ...]
  10.784124     [D4] read: ret 6 buf 82: 30 30 20 30 30  >00 00<
  10.794960     [D4] read: ret 6 buf c1: 30 30 20 30 30  >00 00<
  10.806028     [D4] read: ret 6 buf 82: 2e 30 20 30 30  >.0 00<
  10.817038     [D4] read: ret 6 buf c3: 20 32 33 30 30  > 2300<
  10.821008     [D4] read: ret 6 buf 83: 32 2e 30 30 30  >2.000<
  10.825038     [D4] read: ret 6 buf c1: 20 2e 30 30 30  > .000<
  10.836036     [D4] read: ret 6 buf 81: 30 2e 30 30 30  >0.000<
  10.846959     [D4] read: ret 6 buf c2: 31 34 30 30 30  >14000<
  10.857970     [D4] read: ret 6 buf 83: 20 34 39 30 30  > 4900<
  10.858929     [D4] read: ret 6 buf c2: 2e 30 39 30 30  >.0900<
  10.870126     [D4] read: ret 6 buf 81: 20 30 39 30 30  > 0900<
  10.881053     [D4] read: ret 6 buf c2: 32 37 39 30 30  >27900<
  10.891927     [D4] read: ret 6 buf 83: 2e 30 20 30 30  >.0 00<
  10.896051     [D4] read: ret 6 buf c3: 32 30 2e 30 30  >20.00<
  10.900026     [D4] read: ret 6 buf 81: 38 30 2e 30 30  >80.00<
  10.910931     [D4] read: ret 6 buf c1: 20 30 2e 30 30  > 0.00<
  10.922105     [D4] read: ret 6 buf 82: 30 30 2e 30 30  >00.00<
  10.933079     [D4] read: ret 6 buf c3: 30 30 31 30 30  >00100<
  10.934067     [D4] read: ret 6 buf 82: 00 30 31 30 30  >
  10.934077     [D3] null byte read - is UPS off?
  10.934081     [D2] qx_process_answer: short reply (input.voltage)
  10.934083     Device not supported!
  10.934084     Device not supported!
  10.934092     [D5] send_to_all: SETINFO driver.state "cleanup.exit"
```

blaa, 2023-08-05:
Haha, great. This is new version indeed. And - some new behaviour...

```
   1.191881     [D4] armac command Q1
   1.193102     [D4] armac cleanup ret i=0 ret=6 ctrl=c0
   1.194098     [D4] armac cleanup ret i=1 ret=6 ctrl=80  <- Those seem to work fine.
   1.323226     [D4] read: ret 6 buf c1: 28 31 00 30 30  >(1   <- start of the message is correct.
   1.334354     [D4] read: ret 6 buf 81: 32 31 00 30 30  >21
   1.345239     [D4] read: ret 6 buf c2: 33 33 00 30 30  >33
   1.356217     [D4] read: ret 6 buf 83: 2e 30 20 30 30  >.0 00<
   1.357349     [D4] read: ret 6 buf c2: 30 30 20 30 30  >00 00<
   1.368380     [D4] read: ret 6 buf 81: 30 30 20 30 30  >00 00<
   1.379368     [D4] read: ret 6 buf c2: 2e 30 20 30 30  >.0 00<
   1.390388     [D4] read: ret 6 buf 83: 20 32 33 30 30  > 2300<
   1.394382     [D4] read: ret 6 buf c3: 32 2e 30 30 30  >2.000<
   1.398387     [D4] read: ret 6 buf 81: 20 2e 30 30 30  > .000<
   1.409412     [D4] read: ret 6 buf c1: 30 2e 30 30 30  >0.000<
   1.420421     [D4] read: ret 6 buf 82: 31 34 30 30 30  >14000<
   1.431403     [D4] read: ret 6 buf c3: 20 34 39 30 30  > 4900<
   1.432399     [D4] read: ret 6 buf 82: 2e 30 39 30 30  >.0900<
   1.443277     [D4] read: ret 6 buf c1: 20 30 39 30 30  > 0900<
   1.454355     [D4] read: ret 6 buf 82: 32 37 39 30 30  >27900<
   1.465385     [D4] read: ret 6 buf c3: 2e 30 20 30 30  >.0 00< ".0 "
   1.469432     [D4] read: ret 6 buf 83: 32 30 2e 30 30  >20.00< "20."
   1.473367     [D4] read: ret 6 buf c1: 38 30 2e 30 30  >80.00< "8"
   1.484428     [D4] read: ret 6 buf 81: 20 30 2e 30 30  > 0.00< " "
   1.495430     [D4] read: ret 6 buf c2: 30 30 2e 30 30  >00.00< "00"
   1.506238     [D4] read: ret 6 buf 83: 30 30 31 30 30  >00100< "001"   --> ".0 20.8 00001"  Looks like message end.
   1.507157     [D4] read: ret 6 buf c2: 00 30 31 30 30  >   <- Here we got \x00 which was unexpected.
   1.507160     [D3] null byte read - is UPS off?
```

Looks like your UPS terminates messages with a null byte instead of \r (0x0d). I can add support for this behaviour tomorrow I guess. It should be easily compatible.

Novocaine85, 2023-08-05:
thank you for your support meanwhile :)

blaa, 2023-08-05:
@Novocaine85 I've even hacked a possible solution - `git pull` should get it. If I understand the behaviour correctly it should work. You can try when you have a moment.

Novocaine85, 2023-08-05:
```
root@proxmox:/tmp/nut# /usr/local/ups/bin/nutdrv_qx -a vultech -u root -D -D -D -D -D -x productid=1234 -x vendorid=0925 -x subdriver=armac
Network UPS Tools - Generic Q* USB/Serial driver 0.33 (2.7.4-7675-gd545f624a)
USB communication driver (libusb 0.1) 0.44
   0.000000     [D3] main_arg: var='productid' val='1234'
   0.000007     [D5] send_to_all: SETINFO driver.parameter.productid "1234"
   0.000009     [D3] main_arg: var='vendorid' val='0925'
   0.000011     [D5] send_to_all: SETINFO driver.parameter.vendorid "0925"
   0.000013     [D3] main_arg: var='subdriver' val='armac'
   0.000015     [D5] send_to_all: SETINFO driver.parameter.subdriver "armac"
   0.000021     [D1] Network UPS Tools version 2.7.4-7675-gd545f624a (release/snapshot of 2.8.0.1) built with gcc (Debian 12.2.0-14) 12.2.0 and configured with flags: --sysconfdir=/etc/nut --with-user=ups --with-group=root --with-usb
   0.000024     [D1] debug level is '5'
   0.000026     [D5] send_to_all: SETINFO driver.debug "5"
   0.000028     [D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.000207     [D1] Succeeded to become_user(root): now UID=0 GID=0
   0.000212     [D5] send_to_all: SETINFO device.type "ups"
   0.000214     [D5] send_to_all: SETINFO driver.state "init.device"
   0.000215     [D1] upsdrv_initups...
   0.540768     [D3] usb_busses=0x5654b48db000
   0.540779     [D2] Checking device (1D6B/0003) (004/001)
   0.708044     [D2] - VendorID: 1d6b
   0.708053     [D2] - ProductID: 0003
   0.708054     [D2] - Manufacturer: Linux 6.2.16-6-pve xhci-hcd
   0.708056     [D2] - Product: xHCI Host Controller
   0.708057     [D2] - Serial Number: 0000:00:0d.0
   0.708059     [D2] - Bus: 004
   0.708060     [D2] - Device: 001
   0.708061     [D2] - Device release number: 0602
   0.708063     [D2] Trying to match device
   0.708065     [D3] match_function_regex: matching a device...
   0.708073     [D2] match_function_regex: failed match of VendorID: 1d6b
   0.708075     [D2] Device does not match - skipping
   0.708109     [D2] Checking device (1D6B/0002) (003/001)
   0.708163     [D2] - VendorID: 1d6b
   0.708168     [D2] - ProductID: 0002
   0.708170     [D2] - Manufacturer: Linux 6.2.16-6-pve xhci-hcd
   0.708172     [D2] - Product: xHCI Host Controller
   0.708174     [D2] - Serial Number: 0000:00:0d.0
   0.708176     [D2] - Bus: 003
   0.708178     [D2] - Device: 001
   0.708179     [D2] - Device release number: 0602
   0.708181     [D2] Trying to match device
   0.708182     [D3] match_function_regex: matching a device...
   0.708185     [D2] match_function_regex: failed match of VendorID: 1d6b
   0.708186     [D2] Device does not match - skipping
   0.708197     [D2] Checking device (2109/0822) (002/002)
   0.842960     [D2] - VendorID: 2109
   0.842970     [D2] - ProductID: 0822
   0.842971     [D2] - Manufacturer: VIA Labs, Inc.         
   0.842972     [D2] - Product: USB3.1 Hub             
   0.842973     [D2] - Serial Number: 000000001
   0.842974     [D2] - Bus: 002
   0.842975     [D2] - Device: 002
   0.842977     [D2] - Device release number: 9013
   0.842978     [D2] Trying to match device
   0.842979     [D3] match_function_regex: matching a device...
   0.842987     [D2] match_function_regex: failed match of VendorID: 2109
   0.842989     [D2] Device does not match - skipping
   0.871924     [D2] Checking device (1D6B/0003) (002/001)
   0.871991     [D2] - VendorID: 1d6b
   0.871998     [D2] - ProductID: 0003
   0.872001     [D2] - Manufacturer: Linux 6.2.16-6-pve xhci-hcd
   0.872003     [D2] - Product: xHCI Host Controller
   0.872004     [D2] - Serial Number: 0000:00:14.0
   0.872006     [D2] - Bus: 002
   0.872008     [D2] - Device: 001
   0.872010     [D2] - Device release number: 0602
   0.872012     [D2] Trying to match device
   0.872014     [D3] match_function_regex: matching a device...
   0.872017     [D2] match_function_regex: failed match of VendorID: 1d6b
[... 687 lines cut ...]
  10.491307     [D4] read: ret 6 buf 81: 30 30 20 30 30  >00 00<
  10.502278     [D4] read: ret 6 buf c2: 2e 30 20 30 30  >.0 00<
  10.513309     [D4] read: ret 6 buf 83: 20 32 33 30 30  > 2300<
  10.514317     [D4] read: ret 6 buf c2: 33 2e 33 30 30  >3.300<
  10.525313     [D4] read: ret 6 buf 81: 30 2e 33 30 30  >0.300<
  10.536232     [D4] read: ret 6 buf c2: 20 30 33 30 30  > 0300<
  10.547296     [D4] read: ret 6 buf 83: 31 33 20 30 30  >13 00<
  10.551317     [D4] read: ret 6 buf c3: 34 39 2e 30 30  >49.00<
  10.555324     [D4] read: ret 6 buf 81: 30 39 2e 30 30  >09.00<
  10.566294     [D4] read: ret 6 buf c1: 20 39 2e 30 30  > 9.00<
  10.577287     [D4] read: ret 6 buf 82: 32 37 2e 30 30  >27.00<
  10.588228     [D4] read: ret 6 buf c3: 2e 30 20 30 30  >.0 00<
  10.592174     [D4] read: ret 6 buf 83: 32 30 2e 30 30  >20.00<
  10.596261     [D4] read: ret 6 buf c1: 38 30 2e 30 30  >80.00<
  10.607320     [D4] read: ret 6 buf 81: 20 30 2e 30 30  > 0.00<
  10.618362     [D4] read: ret 6 buf c2: 30 30 2e 30 30  >00.00<
  10.629301     [D4] read: ret 6 buf 83: 30 30 31 30 30  >00100<
  10.630332     [D4] read: ret 6 buf c2: 00 30 31 30 30  >
  10.630341     [D3] trailing bytes in serial transmission found: 1  copied out of 2
  10.630344     [D3] armac command Q1 response read: '(233.0 000.0 233.0 013 49.0 27.0 20.8 00001'
  10.630347     [D2] qx_process_answer: short reply (input.voltage)
  10.630349     Device not supported!
  10.630350     Device not supported!
  10.630357     [D5] send_to_all: SETINFO driver.state "cleanup.exit"
```

blaa, 2023-08-05:
So... now it seems to read the messeges OK each time, but the upper driver (qx) doesn't understand them. Maybe it NEEDS \r instead of \x00. Or... the \x00 should not be included in the length. Two options to try. Third one - it doesn't understand the output at all and some support for Qx is required, but as I understand it used to work.

I pushed the first solution - you can try it.

Novocaine85, 2023-08-05:
```
root@proxmox:/tmp/nut# /usr/local/ups/bin/nutdrv_qx -a vultech -u root -D -D -D -D -D -x productid=1234 -x vendorid=0925 -x subdriver=armac
Network UPS Tools - Generic Q* USB/Serial driver 0.33 (2.7.4-7675-gd48853820)
USB communication driver (libusb 0.1) 0.44
   0.000000     [D3] main_arg: var='productid' val='1234'
   0.000011     [D5] send_to_all: SETINFO driver.parameter.productid "1234"
   0.000013     [D3] main_arg: var='vendorid' val='0925'
   0.000015     [D5] send_to_all: SETINFO driver.parameter.vendorid "0925"
   0.000017     [D3] main_arg: var='subdriver' val='armac'
   0.000019     [D5] send_to_all: SETINFO driver.parameter.subdriver "armac"
   0.000022     [D1] Network UPS Tools version 2.7.4-7675-gd48853820 (release/snapshot of 2.8.0.1) built with gcc (Debian 12.2.0-14) 12.2.0 and configured with flags: --sysconfdir=/etc/nut --with-user=ups --with-group=root --with-usb
   0.000025     [D1] debug level is '5'
   0.000026     [D5] send_to_all: SETINFO driver.debug "5"
   0.000028     [D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.000252     [D1] Succeeded to become_user(root): now UID=0 GID=0
   0.000258     [D5] send_to_all: SETINFO device.type "ups"
   0.000260     [D5] send_to_all: SETINFO driver.state "init.device"
   0.000262     [D1] upsdrv_initups...
   0.535715     [D3] usb_busses=0x557a37939000
   0.535724     [D2] Checking device (1D6B/0003) (004/001)
   0.707218     [D2] - VendorID: 1d6b
   0.707227     [D2] - ProductID: 0003
   0.707228     [D2] - Manufacturer: Linux 6.2.16-6-pve xhci-hcd
   0.707229     [D2] - Product: xHCI Host Controller
   0.707229     [D2] - Serial Number: 0000:00:0d.0
   0.707230     [D2] - Bus: 004
   0.707231     [D2] - Device: 001
   0.707232     [D2] - Device release number: 0602
   0.707233     [D2] Trying to match device
   0.707235     [D3] match_function_regex: matching a device...
   0.707245     [D2] match_function_regex: failed match of VendorID: 1d6b
   0.707246     [D2] Device does not match - skipping
   0.707344     [D2] Checking device (1D6B/0002) (003/001)
   0.707594     [D2] - VendorID: 1d6b
   0.707598     [D2] - ProductID: 0002
   0.707599     [D2] - Manufacturer: Linux 6.2.16-6-pve xhci-hcd
   0.707600     [D2] - Product: xHCI Host Controller
   0.707601     [D2] - Serial Number: 0000:00:0d.0
   0.707602     [D2] - Bus: 003
   0.707603     [D2] - Device: 001
   0.707604     [D2] - Device release number: 0602
   0.707605     [D2] Trying to match device
   0.707606     [D3] match_function_regex: matching a device...
   0.707608     [D2] match_function_regex: failed match of VendorID: 1d6b
   0.707609     [D2] Device does not match - skipping
   0.707676     [D2] Checking device (2109/0822) (002/002)
   0.834389     [D2] - VendorID: 2109
   0.834402     [D2] - ProductID: 0822
   0.834403     [D2] - Manufacturer: VIA Labs, Inc.         
   0.834404     [D2] - Product: USB3.1 Hub             
   0.834405     [D2] - Serial Number: 000000001
   0.834406     [D2] - Bus: 002
   0.834407     [D2] - Device: 002
   0.834408     [D2] - Device release number: 9013
   0.834409     [D2] Trying to match device
   0.834410     [D3] match_function_regex: matching a device...
   0.834422     [D2] match_function_regex: failed match of VendorID: 2109
   0.834423     [D2] Device does not match - skipping
   0.862904     [D2] Checking device (1D6B/0003) (002/001)
   0.863159     [D2] - VendorID: 1d6b
   0.863161     [D2] - ProductID: 0003
   0.863163     [D2] - Manufacturer: Linux 6.2.16-6-pve xhci-hcd
   0.863165     [D2] - Product: xHCI Host Controller
   0.863166     [D2] - Serial Number: 0000:00:14.0
   0.863167     [D2] - Bus: 002
   0.863169     [D2] - Device: 001
   0.863170     [D2] - Device release number: 0602
   0.863171     [D2] Trying to match device
   0.863173     [D3] match_function_regex: matching a device...
   0.863175     [D2] match_function_regex: failed match of VendorID: 1d6b
[... 689 lines cut ...]
  10.464092     [D4] read: ret 6 buf c1: 30 30 20 30 30  >00 00<
  10.475091     [D4] read: ret 6 buf 82: 2e 30 20 30 30  >.0 00<
  10.486073     [D4] read: ret 6 buf c3: 20 32 33 30 30  > 2300<
  10.490137     [D4] read: ret 6 buf 83: 31 2e 30 30 30  >1.000<
  10.494127     [D4] read: ret 6 buf c1: 20 2e 30 30 30  > .000<
  10.505118     [D4] read: ret 6 buf 81: 30 2e 30 30 30  >0.000<
  10.516090     [D4] read: ret 6 buf c2: 31 35 30 30 30  >15000<
  10.527082     [D4] read: ret 6 buf 83: 20 35 30 30 30  > 5000<
  10.528092     [D4] read: ret 6 buf c2: 2e 30 30 30 30  >.0000<
  10.539001     [D4] read: ret 6 buf 81: 20 30 30 30 30  > 0000<
  10.550230     [D4] read: ret 6 buf c2: 32 37 30 30 30  >27000<
  10.561082     [D4] read: ret 6 buf 83: 2e 30 20 30 30  >.0 00<
  10.565123     [D4] read: ret 6 buf c3: 32 30 2e 30 30  >20.00<
  10.569148     [D4] read: ret 6 buf 81: 38 30 2e 30 30  >80.00<
  10.580061     [D4] read: ret 6 buf c1: 20 30 2e 30 30  > 0.00<
  10.591092     [D4] read: ret 6 buf 82: 30 30 2e 30 30  >00.00<
  10.602045     [D4] read: ret 6 buf c3: 30 30 31 30 30  >00100<
  10.602998     [D4] read: ret 6 buf 82: 00 30 31 30 30  >
  10.603001     [D3] trailing bytes in serial transmission found: 1  copied out of 2
  10.603003     [D3] armac command Q1 response read: '(232.0 000.0 231.0 015 50.0 27.0 20.8 00001'
  10.603006     [D2] qx_process_answer: short reply (input.voltage)
  10.603008     Device not supported!
  10.603011     Device not supported!
  10.603018     [D5] send_to_all: SETINFO driver.state "cleanup.exit"
```

blaa, 2023-08-05:
Still `short reply` - we can try tomorrow the other approach. Discord also would be probably better (or signal, or Matrix/Element).

Novocaine85, 2023-08-05:
add me on discord: icchan85

blaa, 2023-08-06:
I added a testcase locally with your data. 

The command is read correctly(?) after the change, but your UPS has less status bits (assuming 0x00 ends message)
00001 instead 00000000 (at the end of response) and Qx doesn't parse the response. Although it ALMOST could.

```
Your UPS:
(232.0 000.0 231.0 015 50.0 27.0 20.8 00001
Convicte's (Armac PF1):
(245.3 000.0 229.6 000 49.9 2.25 50.0 00000001
My UPS (Armac PSW, while on battery)
(000.0 140.0 227.0 002 00.0 46.8 56.0 10001000
```

Maybe previously it worked sometimes because the 0x00 wasn't terminating the message and it was longer for wrong reasons. Or something changed in NUT and master doesn't handle it, but previous version did (?).

At least table in nutdrv_qx_q1.c doesn't allow your response. 

It could be added probably ok, but we'd need a description of the bits (or maybe assuming the first work the same, and later should be ignored). For example which bits are enabled when on line, on battery, during the test.

PR #2005 (2023-08-17): drivers/nutdrv_qx.c: improve Armac subdriver
Based on a debug output from a newer device (*/PF1) we've improved understanding on how:
- those devices encode the length of a chunk of data.
- how the end of transmission can be marked / detected.

Related to:
https://github.com/networkupstools/nut/pull/2003
https://github.com/networkupstools/nut/issues/1978

Closes #1978 

I'll test this changes for "old" devices in september. *Maybe* earlier.
closed 2023-08-17 by jimklimov