networkupstools/nut #1985: riello_usb multiple UPS not possible

benny-kania-novatiq, 2023-07-10:
Hello,

I have 2 Riello SDH3000 connected via USB to one server. Here is the output from nut-scanner:
```
[nutdev1]
        driver = "riello_usb"
        port = "auto"
        vendorid = "04B4"
        productid = "5500"
        product = "USB to Serial"
        vendor = "Cypress Semiconductor"
        bus = "002"
[nutdev2]
        driver = "riello_usb"
        port = "auto"
        vendorid = "04B4"
        productid = "5500"
        product = "USB to Serial"
        vendor = "Cypress Semiconductor"
        bus = "002"
```

The devices are connected to:
Bus 002 Device 002
Bus 002 Device 003

The driver only connects to one of the two devices. The "bus", "device" and "allow_duplicates" are not supported by the driver. Any idea how I can overcome that?

Ubuntu 22.04 LTS
Network UPS Tools - Riello USB driver 0.03 (2.7.4)

All the best,
Benny

jimklimov, 2023-07-10:
Try a custom build of current master, e.g. following https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests

Some such behaviors (especially matching by both "device" and "bus") were addressed after 2.8.0 release and would be part of 2.8.1 after some things are completed.

Peruse also issues labeled as https://github.com/networkupstools/nut/issues?q=label%3AUSB-duplicate-devices (especially the closed ones) about some recent changes and config options for duplicate similar device support.

jimklimov, 2023-11-13:
Now that NUT v2.8.1 was released (and 2.8.2 is probably coming up soon), the solution would pop up in packages as well.

closed 2023-11-13 by jimklimov