networkupstools/nut #1986: usbhid-ups w/pollonly loops forever with "nut_libusb_get_report: No such device" when USB device is reset in 2.8.0

tofurky, 2023-07-11:
I've got a couple CyberPower CP1500PFCLCD devices that in the past required `pollonly` to be set, otherwise they'd give a stale data error.

Upon upgrading to 2.8.0 (via Debian 12 and Ubuntu 23.04), when the UPS switches back to line power from battery and the USB device resets (but immediately is reenumerated by the kernel), the usbhid-ups driver never recovers.

This was being logged over and over, filling up logs:
```
Jul 10 21:22:04 gauss usbhid-ups[2868]: nut_libusb_get_report: No such device (it may have been disconnected)
Jul 10 21:22:04 gauss usbhid-ups[2868]: nut_libusb_get_report: No such device (it may have been disconnected)
Jul 10 21:22:04 gauss usbhid-ups[2868]: nut_libusb_get_report: No such device (it may have been disconnected)
Jul 10 21:22:04 gauss usbhid-ups[2868]: nut_libusb_get_report: No such device (it may have been disconnected)
Jul 10 21:22:04 gauss usbhid-ups[2868]: nut_libusb_get_report: No such device (it may have been disconnected)
Jul 10 21:22:04 gauss usbhid-ups[2868]: nut_libusb_get_report: No such device (it may have been disconnected)
Jul 10 21:22:06 gauss usbhid-ups[2868]: nut_libusb_get_report: No such device (it may have been disconnected)
Jul 10 21:22:06 gauss usbhid-ups[2868]: nut_libusb_get_report: No such device (it may have been disconnected)
Jul 10 21:22:06 gauss usbhid-ups[2868]: nut_libusb_get_report: No such device (it may have been disconnected)
Jul 10 21:22:06 gauss usbhid-ups[2868]: nut_libusb_get_report: No such device (it may have been disconnected)
Jul 10 21:22:06 gauss usbhid-ups[2868]: nut_libusb_get_report: No such device (it may have been disconnected)
Jul 10 21:22:06 gauss usbhid-ups[2868]: nut_libusb_get_report: No such device (it may have been disconnected)
```

The behavior is pretty easy to reproduce by simply using pollonly and running `usbreset 0764:0501`, which causes a similar loop of `nut_libusb_get_report: Input/Output Error`.

I noticed the following changeset (https://github.com/networkupstools/nut/commit/c96f32b860fbd4f8e4e3344d7f0b85a0aa67c8b0) and saw some additional special cases when `pollonly` isn't set, so tried that. The UPS then reconnects automatically and doesn't spam the `nut_libusb_get_report` messages forever after a synthetic `usbreset`. I just tested yanking the power cable from the UPS, and the `nut_libusb_get_interrupt: No such device` appeared only once.

Disabling `pollonly` seems to work OK now - probably something was fixed in NUT in the several years since I first got the UPS - but I think there's a regression here between 2.7.4 and 2.8.0 with how the driver handles USB resets.

jimklimov, 2023-07-14:
I believe before the fix above, re-connections were very flaky in practice, so that detection was added for the use-case known how to detect and debug. Maybe something similar should just be added at least for the error you see :\

Also thanks for the repro case with `usbreset`, wasn't aware of that (doesn't even have a man page nor help text on my Linux box, but exists in PATH).

edt-xx, 2023-11-13:
had what looks to be the same issue with a "0d9f:0004 Powercom Co., Ltd HID UPS Battery".  Logs would get spammed as above.  It seems worst with linux 6.6.1,   in any case the solution was the same, removing pollonly from nut.conf looks to help.

edt-xx, 2023-11-13:
My ups is old, I've replaced the battery several times.  The ups is convinced that the battery is old and needs to be replaced and I've not been able to figure out how to tell it otherwise...  What I have found is that running "upscmd -ued -pxxx opti test.battery.start.quick" daily prevents the problem.  After removing pollonly, I get the problem mentioned in 1983.  Testing to see if use_resetter.py run a few seconds after the upscmd helps.