networkupstools/nut #1987: SMS Manager III Support

leleobhz, 2023-07-11:
Hello!

I'm trying to read SMS Manager III without sucess. I've built NUT for Docker and published at https://github.com/leleobhz/docker-nut (To use glibc-based compilation) and as stated at https://networkupstools.org/stable-hcl.html - This nobreak is served by [blazer_ser](https://networkupstools.org/docs/man/blazer_ser.html). Since this driver was superseed by [nutdrv_qx](https://networkupstools.org/docs/man/nutdrv_qx.html), I'm trying to use it. 

With the following configuration:

```bash
docker run -it --rm --name nut --user 0 --privileged --device /dev -e DRIVER="nutdrv_qx" -e DESCRIPTION="Nobreak Zenith" -e NAME="NobreakZenith" -e PORT="auto" -e SUBDRIVER="cypress" -e VENDORID="04b4" -e PRODUCTID="5500" -e SDORDER="-1" -e API_USER="upsmon" -e API_PASSWORD="aaaaa"  quay.io/pqatsi/docker-nut:v2.8.0-signed
```

Resulting this `ups.conf`:

```
[NobreakZenith]
        driver = nutdrv_qx
        port = auto
        desc = "Nobreak Zenith"
        subdriver = "cypress"
        vendorid = 04b4
        productid = "5500"
        sdorder = -1
```
Running driver directly returns:

/nut/bin/nutdrv_qx -V

```
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0-signed)
USB communication driver (libusb 1.0) 0.43
```

/nut/bin/nutdrv_qx -a NobreakZenith -DDD
```
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0-signed)
USB communication driver (libusb 1.0) 0.43
   0.000000	[D1] debug level is '3'
   0.002508	[D1] upsdrv_initups...
   0.024183	[D2] Checking device 1 of 4 (04B4/5500)
   0.039965	[D2] - VendorID: 04b4
   0.040428	[D2] - ProductID: 5500
   0.040754	[D2] - Manufacturer: Cypress Semiconductor
   0.041119	[D2] - Product: USB to Serial
   0.042261	[D2] - Serial Number: unknown
   0.042335	[D2] - Bus: 001
   0.042391	[D2] - Device: unknown
   0.042517	[D2] - Device release number: 0000
   0.042571	[D2] Trying to match device
   0.042625	[D3] match_function_regex: matching a device...
   0.044825	[D2] Device matches
   0.045186	[D2] Reading first configuration descriptor
   0.046393	[D3] libusb_kernel_driver_active() returned 0
   0.047206	[D2] Claimed interface 0 successfully
   0.047784	[D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   0.049516	[D3] send: QGS
   0.175883	[D3] read: �
   0.176624	[D2] qx_process_answer: input.voltage - invalid start character [f0], expected [28]
   0.178995	[D3] send: QGS
   0.306880	[D3] read: �
   0.307551	[D2] qx_process_answer: input.voltage - invalid start character [f0], expected [28]
   0.309425	[D3] send: QGS
   0.435873	[D3] read: �
   0.436628	[D2] qx_process_answer: input.voltage - invalid start character [f0], expected [28]
   0.440793	[D3] send: M
   0.568869	[D3] read: �
   0.569095	[D2] voltronic_qs_protocol: invalid protocol [�]
   0.571009	[D3] send: M
   0.698866	[D3] read: �
   0.699093	[D2] voltronic_qs_protocol: invalid protocol [�]
   0.700526	[D3] send: M
   0.827869	[D3] read: �
[... 182 lines cut ...]
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

And tracing this nobreak history, in past SMS was this driver: https://alioth-lists.debian.net/pipermail/nut-upsdev/2013-January/006379.html

This is a issue with `nutdrv_qx` driver, there is any configuration required I'm missing there or this is in fact a bug of `nutdrv_qx`?

leleobhz, 2023-07-11:
Just for reference: Another repo build a python support for this nobreak: and the chat is at https://github.com/dmslabsbr/smsUps/blob/master/smsUPS.py#L1673

jimklimov, 2023-07-11:
The response characters seem to be hex, not ascii :\

One driver that comes to mind is voltronic_hex (IIRC the name?), give it a shot...

Also, would nut-scanner say anything about the device?

leleobhz, 2023-07-11:
Hello,

nut-scanner replies like this:

```
root@8f628cce7de1:/# /nut/bin/nut
bash: /nut/bin/nut: No such file or directory
root@8f628cce7de1:/# /nut/bin/nut-scanner 
Scanning USB bus.
No start IP, skipping SNMP
Scanning XML/HTTP bus.
No start IP, skipping NUT bus (old connect method)
Scanning NUT bus (avahi method).
Scanning IPMI bus.
Failed to create client: Daemon not running
[nutdev1]
	driver = "riello_usb"
	port = "auto"
	vendorid = "04B4"
	productid = "5500"
	product = "USB to Serial"
	vendor = "Cypress Semiconductor"
	bus = "001"
```

But running with `riello_usb`:

```
root@840d4051e392:/# /nut/bin/riello_usb -a nutdev1 -DDD
Network UPS Tools - Riello USB driver 0.07 (2.8.0-signed)
Warning: This is an experimental driver.
Some features may not function correctly.

   0.000000	[D1] debug level is '3'
   0.026157	[D2] Checking device 1 of 4 (04B4/5500)
   0.040315	[D2] - VendorID: 04b4
   0.040419	[D2] - ProductID: 5500
   0.040483	[D2] - Manufacturer: Cypress Semiconductor
   0.040542	[D2] - Product: USB to Serial
   0.040610	[D2] - Serial Number: unknown
   0.040669	[D2] - Bus: 001
   0.040850	[D2] - Device: unknown
   0.040912	[D2] - Device release number: 0000
   0.041420	[D2] Trying to match device
   0.041483	[D3] match_function_regex: matching a device...
   0.041546	[D2] Device matches
   0.041614	[D2] Reading first configuration descriptor
   0.043434	[D3] libusb_kernel_driver_active() returned 0
   0.043622	[D2] Claimed interface 0 successfully
   0.043701	[D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   0.046127	[D3] HID descriptor, method 1: (9 bytes) => 09 21 00 01 00 01 22 25 00
   0.046270	[D3] HID descriptor length (method 1) 37
   0.046341	[D3] HID descriptor, method 2: (9 bytes) => 09 21 00 01 00 01 22 25 00
   0.046405	[D3] HID descriptor length (method 2) 37
   0.046456	[D2] HID descriptor length 37
   0.051613	[D2] Report descriptor retrieved (Reportlen = 37)
   0.051707	[D2] Found HID device
   0.051828	[D2] entering start_ups_comm()

   0.052476	[D3] send: features report ok
   0.457260	[D3] send ok
   0.459253	[D3] read: F0 00 00 00 00 00 00 00
   0.472170	[D3] read: F0 00 00 00 00 00 00 00
   0.485175	[D3] read: F0 00 00 00 00 00 00 00
   0.498179	[D3] read: F0 00 00 00 00 00 00 00
   0.511167	[D3] read: F0 00 00 00 00 00 00 00
   0.524165	[D3] read: F0 00 00 00 00 00 00 00
   0.537165	[D3] read: F0 00 00 00 00 00 00 00
   0.550166	[D3] read: F0 00 00 00 00 00 00 00
   0.563188	[D3] read: F0 00 00 00 00 00 00 00
[... 645 lines cut ...]
   0.045554	[D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   0.045803	[D2] Skipping protocol Voltronic 0.06
   0.046014	[D2] Skipping protocol Voltronic-QS 0.07
   0.047469	[D3] send: M
   0.175087	[D3] read: �
   0.175482	[D2] voltronic_qs_hex_protocol: invalid protocol [�]
   0.177592	[D3] send: M
   0.305100	[D3] read: �
   0.305632	[D2] voltronic_qs_hex_protocol: invalid protocol [�]
   0.308152	[D3] send: M
   0.435089	[D3] read: �
   0.436203	[D2] voltronic_qs_hex_protocol: invalid protocol [�]
   0.436376	[D2] Skipping protocol Mustek 0.07
   0.436522	[D2] Skipping protocol Megatec/old 0.07
   0.436628	[D2] Skipping protocol BestUPS 0.06
   0.436734	[D2] Skipping protocol Mecer 0.07
   0.436834	[D2] Skipping protocol Megatec 0.06
   0.437014	[D2] Skipping protocol Zinto 0.06
   0.437379	[D2] Skipping protocol Masterguard 0.02
   0.437511	[D2] Skipping protocol Hunnox 0.01
   0.437614	[D2] Skipping protocol Ablerex 0.01
   0.437714	[D2] Skipping protocol Q1 0.07
   0.437820	Device not supported!
   0.437916	Device not supported!
```

leleobhz, 2023-07-11:
@jimklimov 

Since this nobreak uses a Serial USB adapter, nut-scanner does have anyway to check only serial ports? In the help, there is only a eaton option (-E).

Also, there is something I can do to help the troubleshooting?

leleobhz, 2023-07-11:
@jimklimov hello!

Hacking a bit with Official SMS application, i found a file called `upsilon.xml` and opening serial port with `screen /dev/ttyUSB0 2400`, commands `Q\n` and `I\n` got results! (Commands from file `voltronic.xml` does not works).

This is the file about upsilon [upsilon.xml.txt](https://github.com/networkupstools/nut/files/12018586/upsilon.xml.txt)

This can help to find out why nutdrv_qx did not found my device?

leleobhz, 2023-07-11:
Also, now inspecting again the repo, I found https://github.com/networkupstools/nut/blob/master/docs/cables/sms.txt that describes preciselly my Nobreak - but still dont working.

Looking at `blazer_ser` output, I saw no `Q\n` got tested:

```
root@40c72052d126:/# /nut/bin/blazer_ser -a NobreakZenith -DDDD
Network UPS Tools - Megatec/Q1 protocol serial driver 1.58 (2.8.0-signed)
   0.000000	[D1] debug level is '4'
   0.108330	Please note that this driver is deprecated and will not receive
new development. If it works for managing your devices - fine,
but if you are running it to try setting up a new device, please
consider the newer nutdrv_qx instead, which should handle all 'Qx'
protocol variants for NUT. (Please also report if your device works
with this driver, but nutdrv_qx would not actually support it with
any subdriver!)

   0.108628	[D2] Trying megatec protocol...
   0.109035	[D3] send: 'Q1'
   0.199178	[D3] read: '=`X�6'
   0.199338	[D2] blazer_status: short reply
   0.199443	[D2] Status read 1 failed
   0.199851	[D3] send: 'Q1'
   1.201110	[D3] read: timeout
   1.201264	[D2] blazer_status: short reply
   1.201368	[D2] Status read 2 failed
   1.201757	[D3] send: 'Q1'
   2.203009	[D3] read: timeout
   2.203177	[D2] blazer_status: short reply
   2.203277	[D2] Status read 3 failed
   2.203375	[D2] Trying mustek protocol...
   2.204303	[D3] send: 'QS'
   2.294087	[D3] read: '=`X�6'
   2.294729	[D2] blazer_status: short reply
   2.295327	[D2] Status read 1 failed
   2.296166	[D3] send: 'QS'
   3.297725	[D3] read: timeout
   3.298089	[D2] blazer_status: short reply
   3.298585	[D2] Status read 2 failed
   3.299425	[D3] send: 'QS'
   4.301058	[D3] read: timeout
   4.301418	[D2] blazer_status: short reply
   4.302080	[D2] Status read 3 failed
   4.302651	[D2] Trying megatec/old protocol...
   4.303396	[D3] send: 'D'
   4.388015	[D3] read: '=`X�@'
   4.388432	[D2] blazer_status: short reply
   4.388832	[D2] Status read 1 failed
   4.389507	[D3] send: 'D'
   5.391347	[D3] read: timeout
   5.391804	[D2] blazer_status: short reply
   5.392112	[D2] Status read 2 failed
   5.393124	[D3] send: 'D'
   6.394576	[D3] read: timeout
   6.394960	[D2] blazer_status: short reply
   6.395617	[D2] Status read 3 failed
   6.396059	[D2] Trying zinto protocol...
   6.397543	[D3] send: 'Q1'
   7.399196	[D3] read: timeout
   7.399942	[D2] blazer_status: short reply
   7.400483	[D2] Status read 1 failed
   7.402572	[D3] send: 'Q1'
   8.405651	[D3] read: timeout
   8.407347	[D2] blazer_status: short reply
   8.407874	[D2] Status read 2 failed
   8.408640	[D3] send: 'Q1'
   9.410162	[D3] read: timeout
   9.410992	[D2] blazer_status: short reply
   9.411486	[D2] Status read 3 failed
   9.411796	No supported UPS detected
```

May this nobreak have a kind of "old" upsilon protocol?

jimklimov, 2023-07-14:
Got one more idea: try with `port=/dev/ttyUSB0` (and no/commented-away `vendorid`/`productid`) so it tries the serial-port protocol right away over the OS-driver data encapsulation handling.

leleobhz, 2023-07-14:
> Got one more idea: try with `port=/dev/ttyUSB0` (and no/commented-away `vendorid`/`productid`) so it tries the serial-port protocol right away over the OS-driver data encapsulation handling.

Hello @jimklimov 

This is the output:

cat /nut/etc/ups.conf

```
[NobreakZenith]
        driver = nutdrv_qx
        port = /dev/ttyUSB0
        desc = "Nobreak Zenith"
        sdorder = -1
```

/nut/bin/nutdrv_qx -a NobreakZenith -DDD

```
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0-signed)
USB communication driver (libusb 1.0) 0.43
   0.000000	[D1] debug level is '3'
   0.003901	[D1] upsdrv_initups...
   0.109930	[D3] send: 'QGS'
   0.204329	[D3] read: '=��BX��'
   0.204732	[D2] qx_process_answer: short reply (input.voltage)
   0.205580	[D3] send: 'QGS'
   1.207173	[D3] read: timeout (0)
   1.207589	[D2] qx_process_answer: short reply (input.voltage)
   1.208235	[D3] send: 'QGS'
   1.239221	[D3] read: '=
'
   1.239723	[D2] qx_process_answer: short reply (input.voltage)
   1.241274	[D3] send: 'M'
   1.302234	[D3] read: 'VX�r'
   1.302399	[D2] voltronic_qs_protocol: invalid protocol [VX�r]
   1.303043	[D3] send: 'M'
   2.304324	[D3] read: timeout (0)
   2.304509	[D2] qx_process_answer: short reply (ups.firmware.aux)
   2.304930	[D3] send: 'M'
   3.306190	[D3] read: timeout (0)
   3.306356	[D2] qx_process_answer: short reply (ups.firmware.aux)
   3.307177	[D3] send: 'M'
   4.308446	[D3] read: timeout (0)
   4.308613	[D2] qx_process_answer: short reply (ups.firmware.aux)
   4.309116	[D3] send: 'M'
   5.310370	[D3] read: timeout (0)
   5.310550	[D2] qx_process_answer: short reply (ups.firmware.aux)
   5.311290	[D3] send: 'M'
   6.312549	[D3] read: timeout (0)
   6.312718	[D2] qx_process_answer: short reply (ups.firmware.aux)
   6.313150	[D3] send: 'QS'
   7.314414	[D3] read: timeout (0)
   7.314583	[D2] qx_process_answer: short reply (input.voltage)
   7.315012	[D3] send: 'QS'
   8.316271	[D3] read: timeout (0)
   8.316439	[D2] qx_process_answer: short reply (input.voltage)
   8.316862	[D3] send: 'QS'
   9.318111	[D3] read: timeout (0)
   9.318277	[D2] qx_process_answer: short reply (input.voltage)
   9.318691	[D3] send: 'D'
   9.403959	[D3] read: '=��VX��'
   9.404122	[D2] qx_process_answer: short reply (input.voltage)
   9.404720	[D3] send: 'D'
  10.405968	[D3] read: timeout (0)
  10.406141	[D2] qx_process_answer: short reply (input.voltage)
  10.406541	[D3] send: 'D'
  11.407809	[D3] read: timeout (0)
  11.407980	[D2] qx_process_answer: short reply (input.voltage)
  11.408507	[D3] send: 'Q1'
  12.409757	[D3] read: timeout (0)
  12.409930	[D2] qx_process_answer: short reply (input.voltage)
  12.410986	[D3] send: 'Q1'
  13.412304	[D3] read: timeout (0)
  13.412483	[D2] qx_process_answer: short reply (input.voltage)
  13.413054	[D3] send: 'Q1'
  14.414300	[D3] read: timeout (0)
  14.414471	[D2] qx_process_answer: short reply (input.voltage)
  14.414902	[D3] send: 'Q1'
  14.504796	[D3] read: '=��VX��'
  14.504962	[D2] qx_process_answer: short reply (input.voltage)
  14.505375	[D3] send: 'Q1'
  15.506632	[D3] read: timeout (0)
  15.506824	[D2] qx_process_answer: short reply (input.voltage)
  15.507245	[D3] send: 'Q1'
  16.508511	[D3] read: timeout (0)
  16.508685	[D2] qx_process_answer: short reply (input.voltage)
  16.509102	[D3] send: 'Q1'
  16.535701	[D3] read: '=
'
  16.535878	[D2] qx_process_answer: short reply (input.voltage)
  16.536324	[D3] send: 'Q1'
  16.598700	[D3] read: 'X�h'
  16.598859	[D2] qx_process_answer: short reply (input.voltage)
  16.599271	[D3] send: 'Q1'
  17.600517	[D3] read: timeout (0)
  17.600688	[D2] qx_process_answer: short reply (input.voltage)
  17.601102	[D3] send: 'Q1'
  17.690665	[D3] read: '=��BX��'
  17.690830	[D2] qx_process_answer: short reply (input.voltage)
  17.691231	[D3] send: 'Q1'
  18.692476	[D3] read: timeout (0)
  18.692645	[D2] qx_process_answer: short reply (input.voltage)
  18.693047	[D3] send: 'Q1'
  19.694318	[D3] read: timeout (0)
  19.694488	[D2] qx_process_answer: short reply (input.voltage)
  19.694656	[D2] find_nut_info: info type series not found
  19.694784	[D2] claim: cannot find series
  19.694898	[D2] find_nut_info: info type series not found
  19.695003	[D2] claim: cannot find series
  19.695165	[D2] find_nut_info: info type series not found
  19.695320	[D2] claim: cannot find series
  19.695767	[D3] send: 'Q1'
  19.785597	[D3] read: '=��VX��'
  19.785787	[D2] qx_process_answer: short reply (input.voltage)
  19.786278	[D3] send: 'Q1'
  20.787555	[D3] read: timeout (0)
  20.787737	[D2] qx_process_answer: short reply (input.voltage)
  20.788272	[D3] send: 'Q1'
  21.789530	[D3] read: timeout (0)
  21.789718	[D2] qx_process_answer: short reply (input.voltage)
  21.790233	[D3] send: 'Q1'
  21.880537	[D3] read: '=��VX��'
  21.880704	[D2] qx_process_answer: short reply (input.voltage)
  21.881140	[D3] send: 'Q1'
  22.882409	[D3] read: timeout (0)
  22.882607	[D2] qx_process_answer: short reply (input.voltage)
  22.883031	[D3] send: 'Q1'
  23.884295	[D3] read: timeout (0)
  23.884468	[D2] qx_process_answer: short reply (input.voltage)
  23.885562	[D3] send: 'Q1'
  23.911475	[D3] read: '=
'
  23.911680	[D2] qx_process_answer: short reply (input.voltage)
  23.912218	[D3] send: 'Q1'
  23.974452	[D3] read: 'X�r'
  23.974630	[D2] qx_process_answer: short reply (input.voltage)
  23.975231	[D3] send: 'Q1'
  24.976504	[D3] read: timeout (0)
  24.976694	[D2] qx_process_answer: short reply (input.voltage)
  24.976800	Device not supported!
  24.976897	Device not supported!
```

jimklimov, 2023-07-15:
Looking at that python script, looks like it mostly does translation from a megatec-like "Q" protocol to binary queries for the UPS side, posting `cmd[v]` etc. to the device. So maybe a new NUT driver is in order to handle this device - perhaps a different firmware or even hardware is posted under an older brand/model name?..

jimklimov, 2023-07-15:
Although if your test with `screen` gave a reasonable output as you said, I'm a bit more at a loss. As a wild guess, maybe the driver does not send a `\n` after the command lines, and other devices don't mind but this one does?..

leleobhz, 2023-07-17:
> Looking at that python script, looks like it mostly does translation from a megatec-like "Q" protocol to binary queries for the UPS side, posting `cmd[v]` etc. to the device. So maybe a new NUT [...]

Hello @jimklimov 

This drive was created once in the thread https://nut-upsdev.alioth.debian.narkive.com/bL2wNz3r/nut-driver-for-sms-brazil-ups

About creating a new driver inside nutdrv_qx, how this can be made considering this translation? As you pointed about `screen`, how is the best way to intercept this communication to device and generate log to you? 

Also, the XML I posted here helps with something?

jimklimov, 2023-07-18:
Thanks for the link, but it seems that driver's sources are not available anymore. The reference in HCL that the "SMS Manager III" is supported by `blazer*` drivers is inherited from earlier claim of support by `megatec*` drivers (which were removed in 2010 and absorbed into `blazer*`, and now into `nutdrv_qx` umbrella). Sadly I haven't got any more info about that.

You can see the likes of expected Megatec Qx protocol at https://networkupstools.org/protocols/megatec.html and that document's numerous clones like https://powerman.ru/assets/files/soft/Megatec-single-phase-UPS-protocal.pdf or https://www.upspowercom.com/pcm-download/UPS-Protocol.pdf, or an overview at https://prog.world/ups-emulator-with-megatec-and-voltronic-protocols/ - either way, it is a text-based protocol.

As for that script and possible Qx-Binary translations, there's only so much that I can make of it without a device, nor a long stretch of time to read into it, and with all comments and commands written in Spanish ;) 

* At https://github.com/dmslabsbr/smsUps/blob/master/smsUPS.py#L136-L153 I saw a `cmd` dictionary, which seems to map Q style commands to some character blocks.
* These are used in `on_message()` for sending device commands (test, beep, cancel...) https://github.com/dmslabsbr/smsUps/blob/master/smsUPS.py#L773-L796 along with other options not listed in `cmd[]`.
* The `send_command()` may add a `cmd['Q']` optionally at https://github.com/dmslabsbr/smsUps/blob/master/smsUPS.py#L835-L837 and converts the resulting character block into a byte array written to serial port. Notably, it is the only mention of `ser.write` (and of `ser.read` of 32-byte replies) in the whole script.
* The array is also mentioned in `tempo2hexCMD()` and `getNoBreakInfo()` at https://github.com/dmslabsbr/smsUps/blob/master/smsUPS.py#L1014-L1020 and `queryQ()` at https://github.com/dmslabsbr/smsUps/blob/master/smsUPS.py#L1078

So while the high-level protocol does seem Q-based, at least in that script the actual serial protocol written and read seems to be based on "magic sequences" and not all of them in ASCII text character range.

jimklimov, 2023-07-18:
As for `screen`, it should have logging. I am not sure how helpful it would be to me (can't personally commit to writing a driver) but might help you if you dare get your feet wet with coding and testing :)

The XML seems to provide a list of commands that are expected to work, so helpful in that at least. I now think some of the errors reported in driver screenshots above may be due to either the device getting confused from unsupported commands (when `nutdrv_qx` sends various probes to identify the device and protocol), or if it does answer in 32-byte chunks while the driver expects ASCII-like text sessions with line breaks - then the buffer is full with bytes from an older reply as the driver tries to interpret reaction to a newer query?..

leleobhz, 2023-07-18:
> As for that script and possible Qx-Binary translations, there's only so much that I can make of it without a device, nor a long stretch of time to read into it, and with all comments and commands written in Spanish ;)

Hello @jimklimov 

I've asked to author of `smsbrasil` for the lost code (And with hope to get awnsers). I'll also translate in next days the XML and any code in Portuguese about this Nobreak.

Also I'll try to do a effort with documentation you written before to create a little helper to talk with device. I had some sucess trying to send only `Q`, but I had to enter Q<enter><enter>Q<enter><enter> a lot of times to get awnser that is not garbage - and the awnser was textual in fact, fact may enforce the case you written `either the device getting confused from unsupported commands (when nutdrv_qx sends various probes to identify the device and protocol),`

Thank you for you time, I'll try to dig a bit and I'll awnser here with some progress.

jimklimov, 2023-07-18:
Thanks for the update, and sorry about mis-identifying the language.

jimklimov, 2023-07-18:
Also, I think it could be helpful for experiments to see if that Python script actually talks to your device (is an example of *working* protocol), e.g. try the container "as is" or to remove web-server parts and replace MQTT posts with `sys.stdout.write()` to extract just the UPS-related handling.

alexwbaule, 2023-07-21:
Hi @leleobhz  and @jimklimov 

If i can help, i have SMS PREMIUM 1500 VA, get the same XML debuging the powerView. For this version, get a different XML.

For what i see, the PowerView query the UPS to get the correct protocol, i will get some files to post here too.

alexwbaule, 2023-07-26:
I will try to make a driver based on skel.c , following the python scripts.
Using some python code, i get the model, and some infos.

alexwbaule, 2023-07-26:
@leleobhz and @jimklimov i found this:
https://github.com/petersenna/nobreakSMS/blob/master/userland/nobreakSMS.c
Maybe it can help it ..

alexwbaule, 2023-07-27:
its work the nobreakSMS.c,, with some modifications, i can get my SMS status.
[image]

jimklimov, 2023-07-27:
Nice result, congrats! And that driver is also GPLv2 so ok to borrow parts into NUT skel.c layout.

leleobhz, 2023-07-27:
> its work the nobreakSMS.c,, with some modifications, i can get my SMS status. [image]

Hello @alexwbaule 

What did you changed in the code? I have sucess fixing the overflow issue but no output:

```
dietpi@piprinter:~/nobreakSMS/userland$ sudo strace -f  ./nobreakSMS 
execve("./nobreakSMS", ["./nobreakSMS"], 0x7fcb337cd8 /* 13 vars */) = 0
brk(NULL)                               = 0x4f11000
faccessat(AT_FDCWD, "/etc/ld.so.preload", R_OK) = -1 ENOENT (Arquivo ou diretório inexistente)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=33115, ...}) = 0
mmap(NULL, 33115, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7fa0a07000
close(3)                                = 0
openat(AT_FDCWD, "/lib/aarch64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0`\17\2\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=1455120, ...}) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa0a05000
mmap(NULL, 1527752, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fa086d000
mprotect(0x7fa09ca000, 61440, PROT_NONE) = 0
mmap(0x7fa09d9000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x15c000) = 0x7fa09d9000
mmap(0x7fa09df000, 12232, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7fa09df000
close(3)                                = 0
mprotect(0x7fa09d9000, 16384, PROT_READ) = 0
mprotect(0x411000, 4096, PROT_READ)     = 0
mprotect(0x7fa0a13000, 4096, PROT_READ) = 0
munmap(0x7fa0a07000, 33115)             = 0
openat(AT_FDCWD, "/dev/ttyUSB0", O_RDWR|O_NOCTTY|O_NONBLOCK) = 3
fcntl(3, F_SETFL, O_RDONLY|O_NONBLOCK)  = 0
ioctl(3, TCGETS, {B2400 -opost -isig -icanon -echo ...}) = 0
ioctl(3, SNDCTL_TMR_START or TCSETS, {B2400 -opost -isig -icanon -echo ...}) = 0
ioctl(3, TCGETS, {B2400 -opost -isig -icanon -echo ...}) = 0
write(3, "Q\377\377\377\377\263\r\0", 8) = 8
clock_nanosleep(CLOCK_REALTIME, 0, {tv_sec=0, tv_nsec=15000000}, NULL) = 0
read(3, "", 64)                         = 0
clock_nanosleep(CLOCK_REALTIME, 0, {tv_sec=0, tv_nsec=15000000}, NULL) = 0
read(3, "", 64)                         = 0
clock_nanosleep(CLOCK_REALTIME, 0, {tv_sec=0, tv_nsec=15000000}, NULL) = 0
read(3, "=\366\4", 64)                  = 3
clock_nanosleep(CLOCK_REALTIME, 0, {tv_sec=0, tv_nsec=15000000}, NULL) = 0
read(3, "\366\4B\1", 64)                = 4
clock_nanosleep(CLOCK_REALTIME, 0, {tv_sec=0, tv_nsec=15000000}, NULL) = 0
read(3, "\30\2X\3", 64)                 = 4
clock_nanosleep(CLOCK_REALTIME, 0, {tv_sec=0, tv_nsec=15000000}, NULL) = 0
read(3, "\350\1\220\10", 64)            = 4
clock_nanosleep(CLOCK_REALTIME, 0, {tv_sec=0, tv_nsec=15000000}, NULL) = 0
read(3, "\222\r", 64)                   = 2
clock_nanosleep(CLOCK_REALTIME, 0, {tv_sec=0, tv_nsec=15000000}, NULL) = 0
read(3, "", 64)                         = 0
```

And this way goes forever.

alexwbaule, 2023-07-27:
Hi @leleobhz , i change the connection part, get exacly like NUT.

I changed the "open_config_tty" to this:

```
int open_config_tty(char *tty)
{
    int fd;
    struct termios options;

    options.c_cflag = CS8 | CLOCAL | CREAD;
    options.c_iflag = IGNPAR;
    options.c_oflag = 0;
    options.c_lflag = 0;
    options.c_cc[VMIN] = 1;
    options.c_cc[VTIME] = 0;

    fd = open(tty, O_RDWR | O_NOCTTY | O_EXCL | O_NONBLOCK);
    if (fd == -1)
        return fd; /* ERROR! */

//  fcntl(fd, F_SETFL, 0); /* Block and wait for data */
    fcntl(fd, F_SETFL, FNDELAY); /* Return immediately */

    /* BPS */
    cfsetispeed(&options, BAUDRATE);
    cfsetospeed(&options, BAUDRATE);

    /* APPLY SETTINGS */
    tcflush(fd, TCIFLUSH);
    tcsetattr(fd, TCSANOW, &options);

    return fd;
}
```

I copy this from serial.c in NUT source code... 

At night i will build a driver , following the logic from power view...

alexwbaule, 2023-07-27:
@leleobhz 

I comment to some parts.

[image]

leleobhz, 2023-07-27:
I guess is not only a query6 and query4 issue:

```
openat(AT_FDCWD, "/dev/ttyUSB0", O_RDWR|O_EXCL|O_NOCTTY|O_NONBLOCK) = 3
fcntl(3, F_SETFL, O_RDONLY|O_NONBLOCK)  = 0
ioctl(3, TCFLSH, TCIFLUSH)              = 0
ioctl(3, TCGETS, {B2400 -opost -isig -icanon -echo ...}) = 0
ioctl(3, SNDCTL_TMR_START or TCSETS, {B2400 -opost -isig -icanon -echo ...}) = 0
ioctl(3, TCGETS, {B2400 -opost -isig -icanon -echo ...}) = 0
write(3, "I\377\377\377\377\273\r\0", 8) = 8
clock_nanosleep(CLOCK_REALTIME, 0, {tv_sec=1, tv_nsec=0}, 0x7fe65175d8) = 0
clock_nanosleep(CLOCK_REALTIME, 0, {tv_sec=0, tv_nsec=15000000}, NULL) = 0
read(3, 0x7fe6517640, 64)               = -1 EAGAIN (Recurso temporariamente indisponível)
write(3, "Q\377\377\377\377\263\r\0", 8) = 8
clock_nanosleep(CLOCK_REALTIME, 0, {tv_sec=1, tv_nsec=0}, 0x7fe65175d8) = 0
clock_nanosleep(CLOCK_REALTIME, 0, {tv_sec=0, tv_nsec=15000000}, NULL) = 0
read(3, 0x7fe6517640, 64)               = -1 EAGAIN (Recurso temporariamente indisponível)
write(3, "Q\377\377\377\377\263\r\0", 8) = 8
clock_nanosleep(CLOCK_REALTIME, 0, {tv_sec=0, tv_nsec=15000000}, NULL) = 0
read(3, 0x7fe6517640, 64)               = -1 EAGAIN (Recurso temporariamente indisponível)
dup(2)                                  = 4
fcntl(4, F_GETFL)                       = 0x2 (flags O_RDWR)
brk(NULL)                               = 0x3d3ba000
brk(0x3d3db000)                         = 0x3d3db000
fstat(4, {st_mode=S_IFCHR|0600, st_rdev=makedev(0x88, 0), ...}) = 0
write(4, "Error reading tty. : Resource temporarily unavailable\n", 54Error reading tty. : Resource temporarily unavailable
) = 54
close(4)                                = 0
exit_group(-1)                          = ?
+++ exited with 255 +++
```

And commenting sanity check:

```
openat(AT_FDCWD, "/dev/ttyUSB0", O_RDWR|O_EXCL|O_NOCTTY|O_NONBLOCK) = 3
fcntl(3, F_SETFL, O_RDONLY|O_NONBLOCK)  = 0
ioctl(3, TCFLSH, TCIFLUSH)              = 0
ioctl(3, TCGETS, {B2400 -opost -isig -icanon -echo ...}) = 0
ioctl(3, SNDCTL_TMR_START or TCSETS, {B2400 -opost -isig -icanon -echo ...}) = 0
ioctl(3, TCGETS, {B2400 -opost -isig -icanon -echo ...}) = 0
write(3, "I\377\377\377\377\273\r\0", 8) = 8
clock_nanosleep(CLOCK_REALTIME, 0, {tv_sec=1, tv_nsec=0}, 0x7fd1f1a548) = 0
clock_nanosleep(CLOCK_REALTIME, 0, {tv_sec=0, tv_nsec=15000000}, NULL) = 0
read(3, 0x7fd1f1a4b0, 64)               = -1 EAGAIN (Recurso temporariamente indisponível)
write(3, "Q\377\377\377\377\263\r\0", 8) = 8
clock_nanosleep(CLOCK_REALTIME, 0, {tv_sec=1, tv_nsec=0}, 0x7fd1f1a548) = 0
clock_nanosleep(CLOCK_REALTIME, 0, {tv_sec=0, tv_nsec=15000000}, NULL) = 0
read(3, 0x7fd1f1a4b0, 64)               = -1 EAGAIN (Recurso temporariamente indisponível)
close(3)                                = 0
fstat(1, {st_mode=S_IFCHR|0600, st_rdev=makedev(0x88, 0), ...}) = 0
brk(NULL)                               = 0x38613000
brk(0x38634000)                         = 0x38634000
write(1, "          Input(Vac):0.00\n", 26          Input(Vac):0.00
) = 26
write(1, "         Output(Vac):0.00\n", 26         Output(Vac):0.00
) = 26
write(1, "     Output Power(%):0.00\n", 26     Output Power(%):0.00
) = 26
write(1, "          Output(Hz):0.00\n", 26          Output(Hz):0.00
) = 26
write(1, "    Battery level(%):0.00\n", 26    Battery level(%):0.00
) = 26
write(1, "      Temperature(C):0.10\n", 26      Temperature(C):0.10
) = 26
write(1, "             Beep on:1\n", 23             Beep on:1
) = 23
write(1, "     Active shutdown:1\n", 23     Active shutdown:1
) = 23
write(1, "         Active test:1\n", 23         Active test:1
) = 23
write(1, "              UPS OK:1\n", 23              UPS OK:1
) = 23
write(1, "            Boost ON:1\n", 23            Boost ON:1
) = 23
write(1, "         On AC Power:1\n", 23         On AC Power:1
) = 23
write(1, "         Low battery:1\n", 23         Low battery:1
) = 23
write(1, "    On battery power:0\n", 23    On battery power:0
) = 23
exit_group(0)                           = ?
+++ exited with 0 +++
```

But comparing this code to nutdrv_qx, it sounds like a variant of megatron (Not megatron-old) that does not use Q1 but Q only.

alexwbaule, 2023-07-27:
The nobreakSMS only treat Query6

In the drive i will
check what the protocol to use, like PowerView does (follwing verificaProtocolo.xml)
after that, implement the Query to get UPS Name, UPS mesurements, and Shutdown in any protocol. (nonofasico, trifasico, upsilon and voltronic)

I think this is the "MVP" to work.

leleobhz, 2023-07-27:
> The nobreakSMS only treat Query6
>
> In the drive i will check what the protocol to use, like PowerView does (follwing verificaProtocolo.xml) after that, implement the Query to get UPS Name, UPS [...]

@alexwbaule 

Keep Q6 did not changed notting here:

```
openat(AT_FDCWD, "/dev/ttyUSB0", O_RDWR|O_EXCL|O_NOCTTY|O_NONBLOCK) = 3
fcntl(3, F_SETFL, O_RDONLY|O_NONBLOCK)  = 0
ioctl(3, TCFLSH, TCIFLUSH)              = 0
ioctl(3, TCGETS, {B2400 -opost -isig -icanon -echo ...}) = 0
ioctl(3, SNDCTL_TMR_START or TCSETS, {B2400 -opost -isig -icanon -echo ...}) = 0
ioctl(3, TCGETS, {B2400 -opost -isig -icanon -echo ...}) = 0
write(3, "Q\377\377\377\377\263\r\0", 8) = 8
clock_nanosleep(CLOCK_REALTIME, 0, {tv_sec=1, tv_nsec=0}, 0x7fde67d438) = 0
clock_nanosleep(CLOCK_REALTIME, 0, {tv_sec=0, tv_nsec=15000000}, NULL) = 0
read(3, 0x7fde67d3a0, 64)               = -1 EAGAIN (Recurso temporariamente indisponível)
close(3)                                = 0
```

Something tells me this Q coding are not understandable by my UPS - Sadly. I guess the approach you suggested to check XML file and implement may be a good path, but I do not mind how to even start this code.

alexwbaule, 2023-07-27:
I will start this, and share my code here... i will fork the nut and we can make some debug together.

jimklimov, 2023-07-28:
Sounds like a good idea - making a fork and pushing around the branch(es) (not named "master") with your driver to share the development and testing would help you both avoid the typos in discussion (code in comments or email is a clumsy option nowadays), post references to line ranges of a specific commit/revision in github comments, and to eventually propose a pull request back to main NUT project :)

leleobhz, 2023-07-28:
@alexwbaule and @jungeonkim 

Indeed. I'm also trying to understand the Q protocol properly. Awnsers from Q requests seems to be caothic, but I was able to send simple requests over serial port, like Q. 

My reading is SMS protocol is a mesh of Q implementations but NUT comments about this documentation seems a bit confusing to me - maybe because I don't understood properly the protocol.

If @alexwbaule create a fork with at least a skel, I can test and code exceptions if necessary.

alexwbaule, 2023-08-02:
Hi @leleobhz 

Forked and started.

https://github.com/alexwbaule/nut/tree/feature/add-support-sms-brazil

Im looking into riello_ser to get some ideias too.

im testing on a ArchLinux, and you are welcome to test, code, etc...

leleobhz, 2023-08-02:
Hello @alexwbaule 

I made you code available as container to `quay.io/pqatsi/docker-nut:add-support-sms-brazil` available for amd64 and arm64.

That said, this is a preliminar run of you code:

```
root@10c704b8f28c:/# cat /nut/etc/ups.conf
[NobreakZenith]
        driver = sms_ser
        port = /dev/ttyUSB0
        desc = "Nobreak Zenith"
        sdorder = -1
root@10c704b8f28c:/# /nut/bin/sms_ser -DDDDD -a NobreakZenith   
Network UPS Tools - SMS Brazil UPS driver 1.00 (2.8.0.1)
upsdrv_makevartable
   0.000305	[D5] do_upsconf_args: confupsname=NobreakZenith, var=driver, val=sms_ser
   0.001303	[D5] do_upsconf_args: call main_arg()
   0.001492	[D3] main_arg: var='driver' val='sms_ser'
   0.001579	[D5] do_upsconf_args: not a main_arg()
   0.001659	[D5] do_upsconf_args: this is a 'driver' setting, may we proceed?
   0.001747	[D5] do_upsconf_args: 'driver' setting already applied with this value
   0.001877	[D5] do_upsconf_args: confupsname=NobreakZenith, var=port, val=/dev/ttyUSB0
   0.004392	[D5] do_upsconf_args: call main_arg()
   0.004717	[D3] main_arg: var='port' val='/dev/ttyUSB0'
   0.005084	[D5] send_to_all: SETINFO driver.parameter.port "/dev/ttyUSB0"
   0.006116	[D5] do_upsconf_args: confupsname=NobreakZenith, var=desc, val=Nobreak Zenith
   0.006543	[D5] do_upsconf_args: call main_arg()
   0.006812	[D3] main_arg: var='desc' val='Nobreak Zenith'
   0.007228	[D5] do_upsconf_args: confupsname=NobreakZenith, var=sdorder, val=-1
   0.007674	[D5] do_upsconf_args: call main_arg()
   0.008817	[D3] main_arg: var='sdorder' val='-1'
   0.009281	[D1] Network UPS Tools version 2.8.0.1 (release/snapshot of 2.8.0.1) built with gcc (Debian 12.2.0-14) 12.2.0 and configured with flags: --prefix=/nut --with-all --with-ssl --with-libltdl --with-wrap
   0.009407	[D1] debug level is '5'
   0.009636	[D5] send_to_all: SETINFO driver.debug "5"
   0.009853	[D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.012617	[D1] Succeeded to become_user(nobody): now UID=65534 GID=65534
   0.012946	[D5] send_to_all: SETINFO device.type "ups"
   0.013175	[D5] send_to_all: SETINFO driver.state "init.device"
   0.013365	upsdrv_initups
   0.017773	[D5] send_to_all: SETINFO driver.state "init.quiet"
   0.018037	[D5] send_to_all: SETINFO driver.version "2.8.0.1"
   0.018240	[D5] send_to_all: SETINFO driver.version.internal "1.00"
   0.018486	[D5] send_to_all: SETINFO driver.name "sms_ser"
   0.018837	[D5] send_to_all: SETINFO driver.state "init.info"
   0.019019	upsdrv_initinfo
]:[I�����
Returned: 15
3a4d332053454e4f4944414c20332e00
]:[F�����
Returned: 15
303b3131352f32323030303030313200
]:[Q�����
Returned: 15
363d04f604f60442fa025803e8019000
   0.299067	Short read from UPS
   0.299216	[D5] send_to_all: SETINFO driver.state "init.updateinfo"
   0.299314	upsdrv_updateinfo
   0.299415	[D5] send_to_all: SETINFO driver.state "init.quiet"
   0.300021	[D2] dstate_init: sock /var/state/ups/sms_ser-NobreakZenith open on fd 4
   0.300167	[D5] send_to_all: SETINFO driver.parameter.pollinterval "2"
   0.300280	[D5] send_to_all: SETINFO driver.parameter.synchronous "auto"
   0.300385	Running as foreground process, not saving a PID file
   0.300499	[D5] send_to_all: SETINFO driver.flag.allow_killpower "0"
   0.300626	[D5] send_to_all: SETFLAGS driver.flag.allow_killpower RW NUMBER
   0.300725	[D5] send_to_all: ADDCMD driver.killpower
   0.300836	[D5] send_to_all: ADDCMD driver.reload
   0.300940	[D5] send_to_all: ADDCMD driver.reload-or-exit
   0.301042	[D5] send_to_all: ADDCMD driver.reload-or-error
   0.301148	[D5] send_to_all: SETINFO driver.state "quiet"
[... 217 lines cut ...]
   9.497379	[D2] mainloop: polling 2 filedescriptors
   9.498172	[D3] sstate_dead: driver for UPS [NobreakZenith] says data is stale
   9.499707	[D2] mainloop: polling 2 filedescriptors
  11.498941	[D3] sstate_dead: driver for UPS [NobreakZenith] says data is stale
  11.499521	[D2] mainloop: polling 2 filedescriptors
  13.500919	[D3] sstate_dead: driver for UPS [NobreakZenith] says data is stale
  13.501426	[D2] mainloop: polling 2 filedescriptors
  13.501933	[D3] sstate_dead: driver for UPS [NobreakZenith] says data is stale
  13.502328	[D2] mainloop: polling 2 filedescriptors
  15.503039	[D3] sstate_dead: driver for UPS [NobreakZenith] says data is stale
  15.503552	[D2] mainloop: polling 2 filedescriptors
  15.504061	[D3] sstate_dead: driver for UPS [NobreakZenith] says data is stale
  15.504540	[D2] mainloop: polling 2 filedescriptors
  17.505142	[D3] sstate_dead: driver for UPS [NobreakZenith] says data is stale
  17.505694	[D2] mainloop: polling 2 filedescriptors
  17.506207	[D3] sstate_dead: driver for UPS [NobreakZenith] says data is stale
  17.506566	[D2] mainloop: polling 2 filedescriptors
^C  17.984594	mainloop: Interrupted system call
  17.985134	Signal 2: exiting
  17.985892	[D1] driver_free: forgetting UPS [NobreakZenith] (FD 4)
```

I'll investigate this more, but any thoughts are wellcome now.

Also, thank you a lot @alexwbaule !

alexwbaule, 2023-08-02:
Tks @leleobhz 

Its just de "first commit", there is a lot to do, i need to translate all hex to validate the response, etc.

i will do this this week, using the riello_ser as exemple.

leleobhz, 2023-08-02:
> Its just de "first commit", there is a lot to do, i need to translate all hex to validate the response, etc.

Fine! I just make sure this code will run here.

You think add support like riello_ser is better than attach a new interpreter to nutdrv_qx?

alexwbaule, 2023-08-02:
Maybe, 
The time to understand the nutdrv_qx code for me is longer than that of riello, so to have a "driver done faster" I preferred to go for a separate protocol, just "SMS". Maybe, after implemented, migrate it to nutdrv_qx.

jimklimov, 2023-08-02:
Ideally, if it is a Qx variant after all - better put it (and a detector) under that umbrella. The slow-moving long term plan is to get rid of other implementations and have one driver per major protocol...

If it *is* too different and warrants a separate driver, that's another story...

alexwbaule, 2023-08-02:
@jimklimov  i need to understand all Qx implementation, to see if its a variant.

When I'm done with the implementation, we can see for sure where it fits best, and if that's the case I'll migrate it to Qx.
For now, I'm doing it separately.
To understand the entire nut code, there's a lot to understand.

alexwbaule, 2023-08-02:
@jimklimov ,

Please, give-me a help ? 

im looking in the developer documentation, and don´t  find something.
Is there a "default" values for "dstate_setinfo" and "dstate_addcmd" ??

Taking this driver as example:
I have:
imput voltage
output voltage
output power (in percent)
output hertz
battery level (in percent)
temperature (in celsius)

and some booleans for "the device is in test ?  its OK ? etc"

there is a default values to set to the dstate_setinfo  thats represents the itens above, or i can call every item with any name that i want and its Ok ?

alexwbaule, 2023-08-03:
@leleobhz , 

There is some news.

[image]

alexwbaule, 2023-08-03:
Tks a lot @jimklimov , its the `docs/nut-names.txt` that i want.

Its almost complete the driver, now i will match the names with the docs.

leleobhz, 2023-08-03:
> Tks a lot @jimklimov , its the `docs/nut-names.txt` that i want.
> 
> Its almost complete the driver, now i will match the names with the docs.

Feedback of my Nobreak:

```
root@3f2f4660f611:/# /nut/bin/sms_ser -a NobreakZenith -DDDDDD
Network UPS Tools - SMS Brazil UPS driver 1.00 (2.8.0.1)
upsdrv_makevartable
   0.000359	[D1] Network UPS Tools version 2.8.0.1 (release/snapshot of 2.8.0.1) built with gcc (Debian 12.2.0-14) 12.2.0 and configured with flags: --prefix=/nut --with-all --with-ssl --with-libltdl --with-wrap
   0.000544	[D1] debug level is '6'
   0.000813	[D5] send_to_all: SETINFO driver.debug "6"
   0.001048	[D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.003643	[D1] Succeeded to become_user(nobody): now UID=65534 GID=65534
   0.004222	[D5] send_to_all: SETINFO device.type "ups"
   0.004451	[D5] send_to_all: SETINFO driver.state "init.device"
   0.004666	upsdrv_initups
   0.009255	[D5] send_to_all: SETINFO driver.state "init.quiet"
   0.009411	[D5] send_to_all: SETINFO driver.version "2.8.0.1"
   0.009494	[D5] send_to_all: SETINFO driver.version.internal "1.00"
   0.009567	[D5] send_to_all: SETINFO driver.name "sms_ser"
   0.009644	[D5] send_to_all: SETINFO driver.state "init.info"
   0.009709	upsdrv_initinfo
   0.009771	get_ups_nominal
   0.116366	[D3] Get nominal Ok: received byte 18
   0.116532	get_ups_nominal
   0.222412	[D3] Get nominal Ok: received byte 18
   0.222566	get_ups_nominal
   0.328346	[D3] Get nominal Ok: received byte 18
   0.328558	[D5] send_to_all: SETINFO ups.model "M3 SENOIDAL "
   0.328711	[D5] send_to_all: SETINFO ups.firmware "3.0"
   0.328823	[D5] send_to_all: SETINFO ups.range.voltage "M3 SENOIDAL "
   0.328936	[D5] send_to_all: SETINFO ups.range.current "_ser"
"
   0.329065	[D5] send_to_all: SETINFO ups.frequency "0"
   0.329165	[D5] send_to_all: SETINFO ups.battery.voltage "12"
   0.329279	[D5] send_to_all: SETINFO ups.temperature "41.00"
   0.329402	[D5] send_to_all: SETINFO output.frequency "60.00"
   0.329515	[D5] send_to_all: SETINFO input.voltage "129.00"
   0.329625	[D5] send_to_all: SETINFO output.voltage "111.00"
   0.329734	[D5] send_to_all: SETINFO output.power "25.00"
   0.329840	[D5] send_to_all: SETINFO battery.level "100.00"
   0.330016	[D5] send_to_all: SETINFO battery.voltage "12.00"
   0.330124	[D5] send_to_all: SETINFO driver.state "init.updateinfo"
   0.330232	upsdrv_updateinfo
   0.330337	get_ups_nominal
   0.437348	[D3] Get nominal Ok: received byte 18
   0.437617	[D5] send_to_all: SETINFO driver.state "init.quiet"
   0.438858	[D2] dstate_init: sock /var/state/ups/sms_ser-NobreakZenith open on fd 4
   0.439045	[D5] send_to_all: SETINFO driver.parameter.pollinterval "5"
   0.439177	[D5] send_to_all: SETINFO driver.parameter.synchronous "auto"
   0.439293	[D5] send_to_all: SETINFO device.model "M3 SENOIDAL "
   0.439403	Running as foreground process, not saving a PID file
   0.439511	[D5] send_to_all: SETINFO driver.flag.allow_killpower "0"
   0.439620	[D5] send_to_all: SETFLAGS driver.flag.allow_killpower RW NUMBER
   0.439725	[D5] send_to_all: ADDCMD driver.killpower
   0.439827	[D5] send_to_all: ADDCMD driver.reload
   0.439944	[D5] send_to_all: ADDCMD driver.reload-or-exit
   0.440044	[D5] send_to_all: ADDCMD driver.reload-or-error
   0.440147	[D5] send_to_all: SETINFO driver.state "quiet"
   0.440245	[D1] Driver initialization completed, beginning regular infinite loop
   0.440345	[D6] upsnotify: failed to notify about state 2: no notification tech defined, will not spam more about it
   0.440462	[D5] send_to_all: SETINFO driver.state "updateinfo"
   0.440558	upsdrv_updateinfo
   0.440652	get_ups_nominal
   0.547347	[D3] Get nominal Ok: received byte 18
   0.547582	[D5] send_to_all: SETINFO output.power "26.00"
   0.547707	[D5] send_to_all: SETINFO driver.state "quiet"
   5.445483	[D5] send_to_all: SETINFO driver.state "updateinfo"
   5.445638	upsdrv_updateinfo
   5.445734	get_ups_nominal
   5.552291	[D3] Get nominal Ok: received byte 18
   5.552524	[D5] send_to_all: SETINFO output.power "25.00"
   5.552650	[D5] send_to_all: SETINFO driver.state "quiet"
  10.450488	[D5] send_to_all: SETINFO driver.state "updateinfo"
  10.450649	upsdrv_updateinfo
  10.450748	get_ups_nominal
  10.557250	[D3] Get nominal Ok: received byte 18
  10.557460	[D5] send_to_all: SETINFO input.voltage "127.00"
  10.560104	[D5] send_to_all: SETINFO output.power "26.00"
  10.560311	[D5] send_to_all: SETINFO driver.state "quiet"
  15.455478	[D5] send_to_all: SETINFO driver.state "updateinfo"
  15.455639	upsdrv_updateinfo
  15.455750	get_ups_nominal
  15.562207	[D3] Get nominal Ok: received byte 18
  15.562593	[D5] send_to_all: SETINFO output.power "25.00"
  15.563011	[D5] send_to_all: SETINFO driver.state "quiet"
  20.460474	[D5] send_to_all: SETINFO driver.state "updateinfo"
  20.460727	upsdrv_updateinfo
  20.461401	get_ups_nominal
  20.568129	[D3] Get nominal Ok: received byte 18
  20.568586	[D5] send_to_all: SETINFO driver.state "quiet"
^C  22.265541	[D1] set_exit_flag: raising exit flag due to signal 2
  22.266260	Signal 2: exiting
  22.266991	[D5] send_to_all: SETINFO driver.state "cleanup.upsdrv"
  22.267606	upsdrv_cleanup
  23.354773	[D5] send_to_all: SETINFO driver.state "cleanup.exit"
```

I can look at code on weekend, until them, there is something I can test/do right now?

jimklimov, 2023-08-03:
One test that comes to mind is to ask the driver to produce an `upsc`-style data dump after an initial connection, and after a few update cycles, e.g.:
````
:; /nut/bin/sms_ser -a NobreakZenith -DDDDDD -d 1
````

and
````
:; /nut/bin/sms_ser -a NobreakZenith -DDDDDD -d 3
````

alexwbaule, 2023-08-04:
@leleobhz  and @jimklimov 

Another update, (just pull from /feature/add-support-sms-brazil branch )

Following the `nut-names.txt` and looking another drivers, i have implemented all itens that i think is ok to the driver.

implemented all commands that the UPS has:

- test.battery.start
- test.battery.start.quick
- test.battery.stop
- beeper.toggle
- shutdown.return
- shutdown.stop
- shutdown.reboot

implemented major dstate_setinfo, from query and someones from calculations.

- device.model (from device)
- ups.firmware (from device)
- battery.voltage.nominal (from device)
- output.frequency.nominal (from device)
- ups.beeper.status (from device)
- input.voltage.extended (from device)
- input.voltage (from device)
- output.voltage (from device)
- ups.load (from device)
- output.frequency (from device)
- battery.charge (from device)
- battery.voltage.nominal (from device)
- battery.packs (calculated)
- battery.voltage (calculated)
- ups.temperature (from device)
- battery.charger.status (calculated)

Implemented DELAY for shutdown and  SECONDS for test.battery.start

Implemented "ups.status" based on `new-drivers.txt`

Now we can test it, and if is OK i will make a PullRequest to nut.... 

Im watching the "QX" driver, but looking for SMS Brazil documentation, there is 4 protocols, in the PowerView:

- monofasico (this developed driver)
- trifasico
- upsilon
- voltronic

maybe if migrate to QX, create a "sms_mono", "sms_tri" ?

jimklimov, 2023-08-04:
Since this is about phase count, perhaps something like `sms1` and `sms3`? or `sms1ph`..?

leleobhz, 2023-08-04:
AFAIK monofasico, trifasico are custom variants of Qx. May be a good idea other users of SMS nobreaks check their nobreaks that speaks upsilon and voltronic if SMS also break standards for these ones. 

Particularly in Brazil, this line of nobreaks I have are very popular in SOHO environments.

I would check if monofasico and trifasico can be a nutdrv_qx variant and lead upsilon and voltronic to be checked for HCL if someone report it - since the change SMS uses standard of this protocol is relativelly high.

About @alexwbaule code, I pulled a new docker image with these new features to be tested. Also, I'll check on weekend if I can understand a way to merge it within nutdrv_qx as variant.

leleobhz, 2023-08-04:
> Yesterday I did some tests with the driver I already made, and apparently it's OK, reporting status, commands working, but it's good for other people to test it too.

Ones who want a Docker image to this I made it available today.

Thank you so much for this work @alexwbaule :)

leleobhz, 2023-08-04:
With today version, my nobreak worked well!

This was the output of debug on Android app [APCUPSD Monitor](https://play.google.com/store/apps/details?id=com.nitramite.apcupsdmonitor):

```
BEGIN LIST VAR NobreakZenith
battery.charge: 100.00
battery.charger.status: resting
battery.packs: 1
battery.voltage: 12.00
battery.voltage.nominal: 12
device.model: M3 SENOIDAL: 
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: sms_ser
driver.parameter.pollinterval: 5
driver.parameter.port: /dev/ttyUSB0
driver.parameter.synchronous: auto
driver.state: quiet
driver.version: 2.8.0.1
driver.version.internal: 1.00
input.current.nominal: 000
input.voltage: 129.00
input.voltage.extended: 129.00
input.voltage.nominal: 115/220
output.frequency: 60.00
output.frequency.nominal: 60
output.voltage: 111.00
ups.beeper.status: enabled
ups.firmware: 3.0
ups.load: 24.00
ups.status: OL
ups.temperature: 40.00
END LIST VAR NobreakZenith
```

alexwbaule, 2023-08-11:
Hi @jimklimov 

im looking in the documentation to create a "qx" sub-driver, and i got some questions, if this driver is a qx compatible sub-driver.

The "leading" on "item_t", is a char, but in the "Q" command on SMS 1Phase, the valid answer can be 3 values (<.>,=).

I Thinking in maybe open a PR from this driver and after that, migrate the driver if need some changes to qx driver itself.

You are ok with that ?

jimklimov, 2023-08-12:
I've asked on the mailing list if someone better versed with the driver structure is online these summer days. Perhaps also CC @blaa who was fixing a subdriver recently.

If no quick solutions come up, a sparrow in hand is better than a swan in the sky, for a first catch anyway ;)

alexwbaule, 2023-08-27:
> Hi @jimklimov
>
> im looking in the documentation to create a "qx" sub-driver, and i got some questions, if this driver is a qx compatible sub-driver.
>
> The "leading" on "item_t", is a char, but in the [...]

Hi @blaa !
My only concern is the "valid value" that can be readed on "leading" can be a option between  3 possibilities. but the Qx sub-driver only accepts one.
Im building the sub-driver based on a megatec-hex driver, its basically the same thing.

cristovamperes, 2024-03-26:
I'm trying to make nut work with my SMS Manager III using the nutdrv_qx driver with no luck.

Here's my ups.conf

`pollinterval = 15
maxretry = 3

offdelay = 120
ondelay = 240

[apc]
# SMS Manager III
driver = nutdrv_qx
port = /dev/ttyUSB0
desc = "SMS Manager III"`

Am I missing something?

alexwbaule, 2024-03-26:
@cristovamperes in nuts 2.8.0 up, there is a SMS driver, that i built.

cristovamperes, 2024-03-26:
I'm running proxmox with bullseye so the package available is < 2.8.0.
Created a new lxc with debian, updated to bookwork passingthrought the usb port. 
Using apt install nut was able to install 2.8.0-7

```
# nut-scanner -V
Network UPS Tools - 2.8.0
```

But running nut-scanner -U won't detect the UPS and after checking /lib/nut/ for the sms_ser driver found out that it's not there.

alexwbaule, 2024-03-26:
Sorry @cristovamperes 

The merge was for 2.8.1 milestone... 

https://github.com/networkupstools/nut/pull/2090

cristovamperes, 2024-03-27:
Great! It's now running. Testing with PeaNUT from https://github.com/Brandawg93/PeaNUT

raphaeleduardo42, 2024-05-19:
Worked great whe using with SMS PRO 1500va, really thank you guys.

Is it possible to implement the battery runtime by the data got?

vinicius-trev, 2026-02-26:
I believe that I already disabled that feature, it also only activate it while on battery mode; will double check later just for sanity check

marlop352, 2026-09-09:
I'm using a python script as a proxy for each nut client (running locally in the client so that I don't need to deal with more than one client using the same proxy) to change "BYPASS" to "ECO" when input voltage is inside a defined range and to "OB" when outside, that's running as a workaround until the driver has a way to deal with the wrong status being reported by the firmware.

karloscarrijo, 2026-09-09:
Maybe we could create a flag on the driver like `ignore_bypass` for these instances where it is reporting a fake Bypass state (for SMS Gamer/newer lines)?

diogot, 2026-09-22:
I have opened #3701, a rework of `sms_ser` developed against an SMS Premium 1500 VA (PRM1500BiBiW), which is one of the newer line-interactive models with the fake BYPASS discussed above.

What it changes, in short:

- `ups.status` is built from all flags. The bypass flag is only reported for on-line models, decided from the type byte of the status reply, which is what @alexwbaule found the app doing ("UPS Line Interative" ignores it). So on the Premium and, I expect, the Gamer line, an outage is now `OB`, with `LB` when the UPS raises its low-battery flag (on my unit about two minutes before it cuts the output) or when the charge falls to a configurable `battery.charge.low`. No `ignore_bypass` flag is needed (@karloscarrijo), it follows from the type byte; `upsmon` and DSM react to it as usual.
- Replies are validated (checksum and terminator) and the serial port is reopened after a USB replug or after the UPS powers itself down.
- Test and shutdown delays are sent in the format the vendor software uses. On this unit the UPS counts them in 0.6 s units and the restore delay in minutes, so a 10 s test used to run for 6 s and `shutdown.return` did not work at all; now `shutdown.return`, `shutdown.stop`, `shutdown.reboot` and the killpower path behave as documented. `test.battery.start.quick` is now 10 s and the new `test.battery.start.deep` runs until low battery (the names were swapped relative to their effect).
- Two opt-out flags, `legacydelays` and `novalidate`, restore the old behaviour if a device turns out to differ.

All of this was measured on one unit, so before it is merged I would like owners of other SMS models to try the branch (`diogot/nut`, branch `sms_ser-premium-support`, or the PR's tarball) and report here or in the PR:

1. `upsc` output at rest, during a short mains cut and during a self test.
2. With `-DDDD`, the `sms_ser read` lines (raw replies), in particular the first byte after the length (`=`, `<` or `>`) and whether any reply is rejected with "Bad checksum or terminator".
3. If you can do it safely (nothing but a lamp on the UPS): how long `test.battery.start 30` really runs, and what `shutdown.return` does.
4. If something that worked before stops working, whether `novalidate` or `legacydelays` brings it back.

@vinicius-trev @carnei-ro @karloscarrijo (Gamer 1500/2000, including a 220 V model), @marlop352, @leleobhz (Manager III) and @alexwbaule: your models would cover what I could not test. On-line models (`<` and `>` replies) I could not test at all.

Tell me once it is posted and I will tick it in the checklist.
A device dump for the Premium is in networkupstools/nut-ddl#68. 

Developed with Claude Code (model Claude Fable 5.1); I ran the tests on the hardware.