networkupstools/nut #1988: 2 instances of NUT (one via Ethernet and one via USB)

jimklimov, 2023-07-13:
> Re-posted from https://github.com/networkupstools/nut-website/issues/38 by @Briandr73

Hello, 

Not so much of an issue, but more of an inquiry. I have a HP Micro Mini setup running home assistant and I connected my CyberPower UPS to it via the USB cable. I added the NUT addon and integration. All works well. Very happy with it. I do have two other UPS's in my home that have a RMCARD205 in them. I am not sure if this is even possible without the use of a small Raspberry Pi, but can I somehow have two instances of nut and one of which would connect via Ethernet. The HP Micro Mini and both RMCARD205's are on the same vlan. Any help and suggestions would be appreciated.

jimklimov, 2023-07-13:
Not quite sure about the envisioned topology in the original post, so I'll try to rephrase it and you tell me if I got it right, please :)

One system is the HP Microserver running NUT to monitor an USB-connected UPS, as an add-on in Home Assistant. 
* Please note that maybe HA packaging has some constraints that we in upstream project are not much aware about, so the answer is about what "vanilla" NUT can do.

Two other UPSes are CyberPowers with an RMC so they can be monitored, I suppose, using SNMP over Ethernet (and TCP/UDP).

In this case you do not need "extra NUT instances" (note that NUT by architecture is 3-layered with different daemons for different roles, so a definition of an "instance" may be questionable). With "vanilla" NUT you would just set up `ups.conf` to run a couple more `snmp-ups` drivers with `port=IP` of each RMC (and probably `upsmon.conf` to monitor them, even if powering 0 inputs of this NUT monitoring system). No additional Raspberries are required (that said, not prohibited either - they could have the drivers+`uipsd`, and your upsmon's or other clients on other boxes would monitor them as any other remote NUT server).

So from this point, it is a question of whether your NUT HA plugin supports setting up more than one driver (and if it includes numerous NUT drivers and not only USB), and if it can set up a device for monitoring that would not trigger this server's shutdown if something happens. I gather there may be several NUT container and HA plugin recipes out there, but haven't had direct contact with any. So you are encouraged to track down the author (or source) of yours and ask/grok the answers to the few questions above.

As far as NUT is concerned, this is a normal set-up with 3 drviers (1 `upshid-ups` and 2 `snmp-ups`), 1 data-server `upsd` and 1 `upsmon` if at least one of those devices feeds the current box (or you want otherwise to be notified about power events on just-monitored UPSes).

Briandr73, 2023-07-14:
Hi -

Thanks for the response and I will try to add relevant comments and answer your questions. I kind of figured I would not need separate instances as I originally indicated. The Cyber Power UPS connecting via USB is managed by the HA add-on as shown below.

[image]

I believe the settings specified in HA would normally be specified in the ups.conf and upsd.users files as you mentioned. There is probably couple other files I am not mentioning. While this process is not 100% clear to me I did follow a video and set NUT up successfully on a Orange Pi. I do agree it will come down to whether HA supports another driver. Can you tell based on my screenshot where the relevant information (snmp-ups) for the RMCARD would go since HA is kind of combing all of these NUT files together? thanks.

jimklimov, 2023-07-14:
Probably it would go into `devices` part as additional blocks with something like:
````
- name: RemoteUPS1
  driver: snmp-ups
  port: 10.2.3.4
  config:
    - desc: My remote UPS in the basement
    - ...
````

See https://networkupstools.org/docs/man/snmp-ups.html for `config` settings (for current Git master), or https://networkupstools.org/historic/v2.8.0/docs/man/snmp-ups.html and https://networkupstools.org/historic/v2.7.4/docs/man/snmp-ups.html for a few older releases.

Typically you would have a non-default `community` (if `snmp_version` is `v1` or `v2c`), or further auth settings if `v3`.

This driver can walk its list of `mibs` to pick the first supported one (by vendor extension OID presence) or falls back to `ietf` as the common denominator. Some devices support several MIBs with different data sets or value precision, so you may want to bolt a specific one where it matters. At the moment there is no support for mixing of several MIBs.

jimklimov, 2023-07-14:
Also note that if at least `community` settings are wrong (don't remember OTOH if it also impacts `v3`) then SNMP servers tend to not send a response. Given that the protocol is UDP-based, there is no TCP-style connection/session to break off, so a bad password is indistinguishable from a non-responsive/not-running server.

Briandr73, 2023-07-14:
Thanks -- I will probably test out Sunday when I am off and post back either that day or Monday. thanks again

Briandr73, 2023-07-16:
I tried adding the following section that you recommended, but the HA NUT addon does not like the code even after I removed reference to the existing one connected via USB. Not sure if you have any other ideas or if someone else can help. Also questioning how the NUT addon could login since its not installed. Its just a rmcard plugged into ups. thanks for offering to help.

jimklimov, 2023-07-16:
Well, this assumes that the RMCards do serve SNMP, and that you know the credentials (and protocol) involved.

For SNMP v1 and v2c the "community name" is the equivalent of login and password together, like `public` or `private` (by default, respectively with R/O or R/W access; `private` is commonly disabled in factory settings). For SNMP v3 there are actual logins and passwords, together with different auth/encryption algorithms which you have to specify correctly.

For what it's worth, you can try first to run `snmpwalk` (part of `net-snmp`, `ucd-snmp` or related packages) with different auth options to get some reply from the device, and then carry that over into NUT configuration.

These auth values on the UPS side can probably be configured with some web- or ssh-interface running on the card, same as where you had its IP addressing set up.

jimklimov, 2023-07-16:
Also, I have no idea if that HA plugin is constrained to monitoring just one device. You mentioned removing the original USB UPS section during testing; were there two or one SNMP UPS remaining mentioned?

Briandr73, 2023-07-16:
I am still thinking about your first reply to me today so apologies while I answer the second. No, the first USB attached UPS was referenced in the NUT add-on section and I had to also call it from the NUT integration so I could pull data for the individual entities. The second USB attached UPS I was testing how I could get that connected without the use of small desktop. I was able to get it worked with an Orange Pi 800. The Orange Pi 800 worked well except when it started it would make a popping sound. Completely usable but I got it because I could not Pi 400 at the time and was not willing to pay the scalpers who were charging double or triple the normal value. Sorry if I digressed. 

As it relates to your first reply of today yes the rmcard205's do support snmp and I logon to the card using a user name and pw I setup. In concluding this reply I did try snmpwalk but for whatever the exe for it was not in the download. I am guessing while you know NUT but no so much HA I do have package.yaml files for the 2 rmcard205's, but one is giving errors so I was hoping to bypass the code I found online and use HA plugin or integration directly.

[image]

jimklimov, 2023-07-16:
Well, I think a "second USB UPS" was not mentioned before. If it is somewhere remote (too far to pull USB cabling to the MicroServer) then indeed a computer like a Pi (or the actual computer "fed" by it, if any) is useful to have it monitored on the network. Such Pi would run the NUT driver and `upsd` locally, and anyone including itself or the MicroServer may be a NUT client. At least as far as NUT is concerned; no idea about HA plugins being clients to remote UPSes. As an idea, you might get away with a `dummy-ups` driver running on the MicroServer as a relay of a remote NUT device (so it is a NUT networked client on one side, and a local driver on the other).

As for SNMP, the screenshot suggests you have at least `snmp_version=v1` available with `community="public"` for reads or `community="private"` for reads and writes (settings, commands), from any IP address. You might want to constrain these :)

On a side note, I see it also has a ModBus TCP server, though not sure NUT has a driver for that variant yet.

Briandr73, 2023-07-16:
I am not too deep of an IT guy so I will see how far I can go. If your curious and I am also posting this in the event a HA guru stops by for a visit I am using two package.yaml files to get data from two UPS's with remcard's installed. The only difference is the second on references items with and  _2 since I already used the one found here. Don't get me wrong both appear to be fine except one is complaining about an expected numeric value. 

https://peterkieser.com/2020/11/16/home-assistant-sensors-and-cyberpower-ups-via-rmcard205/

[image]