networkupstools/nut #1989: Connection failure: Connection refused on a GreenCell USV

KayGundhardt, 2023-07-17:
Good morning everyone!

First of all: I am quite a Linux newbie and also just had my first experience with setting up a NUT server. 

So I would be happy if you could understand if I did something wrong or overlooked something and thank you very much for your help - thank you!

For two days I tried unsuccessfully to set up a UPS with a NUT server on different systems (hardware [a,b,c] and / software [1,2]):

- **OS Name and Version:** 
[a1] MacMini 2018 (A1993) with MacOS Ventura 13.4
[a2] same MacMini 2018 (A1993) with Windows
[b1] RaspberryPi 3 Model B+ with Raspberry PI OS (32-BIT) V2023-05-03
[b1] same RaspberryPi 3 Model B+with Ubuntu Server 20.04.5 (64-BIT) ** 
[c1] RaspberryPi 4 Model B+ with Raspberry PI OS (32-BIT) V2023-05-03 
[c2] same RaspberryPi 4 Model B+ with Ubuntu Server 20.04.5 (64-BIT) **

**  following [these instructions](https://mic22.medium.com/basic-nut-configuration-for-green-cell-ups-and-home-assistant-integration-e08affcc54f)

- **NUT Version**: 2.7.4-13 in all cases

- **Device Name**: [GreenCell UPS USV 2000VA 1200W](https://greencell.global/en/ups-and-agm-accumulators/1089-green-cell-uninterruptible-power-supply-ups-2000va-1200w-with-lcd-display-eu-version-4x-schuko-sockets.html)

- **Problem Description**: I always get the following error message: `Connection failure: Connection refused`

- Further Informations:
(a) `lsusb` gives me `Bus 001 Device 004: ID 0001:0000 Fry's Electronics MEC0003`
(b) `nut-scanner` results in:
```
[nutdev1]
	driver = "nutdrv_atcl_usb"
	port = "auto"
	vendorid = "0001"
	productid = "0000"
	bus = "001"
```

- Steps I tried: 
(a) Using differente hard- and software combinations (see above) using all available USB-ports on each.
(b) Tried 4 different drivers: `blazer_usb`, `genericups`, `nutdrv_atcl_usb`, `nutdrv_qx` 
(c) Repaired permissions: `sudo chown -R nut:nut /var/run/nut` and `sudo chown -R nut:nut /etc/nut`
(d) Used the following different `/etc/nut/ups.conf` minimum versions (also following the above linked description):

```
maxretry = 5
pollinterval = 5

[ups-redqueen]
  driver = (see above)
  port = auto
  desc = "UPS Server"
```

and 

```
maxretry = 5
pollinterval = 5

[ups-redqueen]
  driver = (see above)
  port = auto
  desc = "UPS Server"
```

Now, before posting the different drivers debug info, this [manual document](https://networkupstools.org/docs/man/nutupsdrv.html) says: 

_"The level of debugging needed depends both on the driver and the problem you’re trying to diagnose. Therefore, first explain the problem you have with a driver to a developer/maintainer, before sending them debugging output. More often than not, if you just pick a level, the output may be either too limited or too verbose to be of any use."_ 

So I tried to explain my problem now and really hope some could help - thank you!
:)

Kay

jimklimov, 2023-07-18:
Hello, thanks for tge report. Do you have any visibility into the driver (its co sole messages if executed directly, or service log) whether it starts successfully or fails to identify the device and quits?

The "Connection refused" looks like a networking error - e.g. if you did not start the data server (`upsd`) and the client (`upsc`, `upsmon` and others) has noone to talk to, or if firewalls block something, or if you e.g. LISTEN on localhost but connect to a routable IP. Beware also IPv4 vs. IPv6 mix-ups (`localhost` may be evil in this regard on some OS releases).

KayGundhardt, 2023-07-18:
Hey @jimklimov ,

THANK YOU very much for your answer! My next step would be to check the drivers behavior, but as I am new to Linux I do not fully understand the debug command to check the driver. Could you maybe help me to find the right syntax to check the drivers status?

Thank you!

Kay

jimklimov, 2023-07-18:
It depends on the distribution. The ones you mentioned seem to be Debian based, so the likely commands would be:
* `systemctl status nut-driver.service nut-server.service nut-monitor.service` to see current status of each relevant unit (for 2.7.4) and a few latest log lines
* `journalctl -lu nut-driver.service` to see the full log of that unit
* `journalctl -flu nut-driver.service` to see new entries added to the log of that unit in real time (as you experiment and restart stuff) - I'd suggest opening separate terminals for each such log and for your interactive work; otherwise it is possible to background the log-printing process with `&` at the end of command line (and use `fg` to bring it back to foreground so you can `Ctrl+C` it later).

KayGundhardt, 2023-08-24:
Yippee! :) Made an entire clean system install and now it works - not sure what the reason was…

KayGundhardt, 2023-08-24:
Yippee! :) Made an entire clean system install and now it works - not sure what the reason was… 

Btw: it works with the blazer_usb

closed 2023-08-24 by KayGundhardt