networkupstools/nut #1990: Problem detecting Legrand and APC UPS

ITPotatoo, 2023-07-17:
Hello

I installed a NUT server and other components in a Ubuntu server VM, then I wanted to add the 2 UPS :

APC Easy-UPS 3kVA SRV3KRI

Legrand Daker DK + 3kVA (exist in the compability list)

Using USB to USB and RS232 to USB, but the NUT server cannot detect them even with changing driver name every time.

when i use commande lsusb :

for APC it shows :

Device 004 is RS232 to USB from the APC UPS

Device 005 is USB to USB from the same APC UPS
[image]

And for Legrand its shows something like : Bus...….reserved reserved

I followed the same configurations as the Techno Tim youtube channel.

The NUT configurations are follows :
[image]
[image]
[image]
[image]
[image]

Can you help me please

jimklimov, 2023-07-18:
In case of the APC, the serial port device name should be a path - e.g. `/dev/ttyUSB0` (with the USB converter) on most Linux systems. For USB devices `port=auto` is currently correct. I don't see a "Device 005" in the screenshots above, but that setup would be using a `port=auto` too.

As for the driver, not sure what the EasyUPS could be handled by - did not find them in NUT HCL nor DDL. Might be one of `apcsmart`, `apcsmart-old`, `microsol-apc`, or ideally `usbhid-ups` - but per #139 many newer APCs (nearly) dropped the standard USB HID in favor of a Modbus based protocol, and nobody so far stepped up to update the drivers (or port `apcupsd` code into NUT).

Finally, the "Bus... reserved" does not ring a bell, sadly. Could it be some poor USB forwarding through VMWare? Or the VendorID/ProductID of the device not listed in `udev.rules` or similar and so not forwarded to NUT?

In fact, looking  at https://github.com/networkupstools/nut/commit/258276ac1a1892690fe5e4472309ae5d8804efeb which mentions "Daker" and PR #1075 which that commit is part of, considerable support enhancement for Legrand devices landed in NUT v2.8.0. Which one are you using (sadly, old 2.7.4 is still popular in many distros)?..

ITPotatoo, 2023-07-23:
Hello @jimklimov thanks for response, I really appreciate your help
for Legrand, this is the config :
[image]

And this is the lsusb, as you can see , it is bus 002, i installed the nut server on a physical server instead of a VM
[image]

And this is the error message when i do upsdrvctrl start command
[image]

I hope this helps identify my problem with legrand
i listed installed package and i found nut running with version 2.7.4

ITPotatoo, 2023-07-23:
@jimklimov i just found that the last version of nut in ubuntu is 2.7.4
i will try in a Debian server Couse the version is 2.8.0 there.

ITPotatoo, 2023-07-24:
Hello @jimklimov , it works for the Legrand UPS in Debian ,because the packages are  version 2.8.0 while they are 2.7.4 in Ubuntu
Thank you for your helps i appreciate it.
Hope one day NUT developers add the APC Easy UPS series, because this series are width used.

jimklimov, 2023-07-24:
Well, it's not that there's a formal body of NUT developers - just the likes of you and me that bring together the hardware, curiosity and code... PRs are welcome ;)

tbsky, 2023-08-09:
we have some APC easy ups from 1kv-6kv. these are cost down ups. driver apcsmart works fine under nut 2.8.0

ITPotatoo, 2023-11-29:
Hello @tbsky 
Please can you share with me the configurations you did ? Im struggling with this problem

tbsky, 2023-11-29:
ITPotatoo ***@***.***>

> Please can you share with me the configurations you did ? Im struggling
> with this problem
>
you mean APC easy-ups series?
with serial cable it is easy:

> cat ups.conf
[myups]
- driver = "apcsmart"
- port = "/dev/ttyS0"
- sdtype = 5

if you want to play with usb, you need an os which support driver
"xr_serial". or you need to compile the driver yourself. you load the
module "xr_usb_serial_common" then everything is like serial. usb config
like below:

>cat ups.conf
[myups]
driver = "apcsmart"
port = "/dev/ttyXRUSB0"
sdtype = 5

Regards,
tbskyd
Message ID: ***@***.***>

>

tbsky, 2023-12-03:
Hi:
    your requirement is different than mime. I have only few requirement for ups: 
1. can shutdown/return when on battery/on utility.  "sdtype = 5" is the value I need to do this. this is found by real shutdown/return testing. (you can try other value to find what you like)
2. can report important ups values like ups.load
3. battery can be calibrated so I can extend the battery life and understand the minimal time the battery can hold.

APC easy-series is cost-down series. and higher kv model have more functions (eg: 6kv has more function than 1kv). so I don't know if your model has the function you want.

to have correct  battery.runtime/battery.charge you need to at least do manually calibration (with the correct load and drain the battery). ups calibration function is useless for this because it will recover at about 25% charge.  the ups may took 5 minutes to 25%, but spent 30 minutes with the 25%. and with time goes by, these values will change. so I only use early shutdown.

ITPotatoo, 2023-12-20:
Thank you @tbsky its working