networkupstools/nut #1991: CyberPower 550va Not Supported?

Briandr73, 2023-07-17:
Hello,

I may have asked this question 6 months ago or loner, but from checking the page below it does not appear my CP 550va is supported. 

https://networkupstools.org/ddl/Cyber_Power_Systems/

I cannot remember what the upsc command was returning, but it was not even something remotely close to the 550va. In fact I think the command did not even return any results because a driver could not be loaded. Any suggestions for this type of USP that does not show up on supported devices list?

Thank you.

jimklimov, 2023-07-18:
Can't find a model named "CP 550va" on CyberPower site or general internet. Some web-shops indicate models with 550 VA power rating and different model numbers and exterior designs.

The general suggestion is to try loading different drivers, most likely `usbhid-ups` or `snmp-ups` (if yours is networked with an RMCard).

There were some improvements for CPS in NUT v2.8.0 release and later in the master branch, so you can try to custom-build NUT as suggested at e.g. https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests to run the current driver right from the build workspace to see if it picks up your device.

Briandr73, 2023-07-18:
Hello and thank you for replying her. Apologies the correct name is SX550Ga from the bottom of the unit. Until I have the Pi 400 setup with NUT I can't confirm what happened in the past. I want to say usbhid-ups loaded, I think but it was flagged as different CyberPower UPS. This particular SX550Ga looks like ass big power strip with no LED display, to me basically a dumb ups. So networking options for it. Getting some data on it from NUT would be a huge help.

Briandr73, 2023-07-22:
This is a SX550Ga from CyberPower UPS

brian@raspberrypi:~ $ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 04d9:0007 Holtek Semiconductor, Inc. Raspberry Pi Internal Keyboard
Bus 001 Device 008: ID 0764:0501 Cyber Power System, Inc. CP1500 AVR UPS
Bus 001 Device 007: ID 275d:0ba6  USB OPTICAL MOUSE 
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

brian@raspberrypi:~ $ sudo upsdrvctl start
Network UPS Tools - UPS driver controller 2.7.4
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
No matching HID UPS found
Driver failed to start (exit status=1)

from ups.conf
pollinterval = 1
maxretry = 3

[lexups]
        driver=usbhid-ups
        port=auto
        desc = "CyberPower 550va"
        vendorid=0764
        productid=0501
        product=ST Series
        vendor=CPS
        bus=001

Any ideas?

jimklimov, 2023-07-23:
For upsmon - it misses "master"/"slave" keyword

jimklimov, 2023-07-23:
For driver - did you have a chance to build and check with current git sources?

Briandr73, 2023-07-23:
Hi -

Apologies as its around 3:45am as I am replying to your last message. Has upsmon changed recently? I'd have to check my Orange Pi (no monitor connected). I did find this online and was wondering if that would fix this.

[upsmon]
  password = 123456
  upsmon master

I am not sure what you mean by build, but I had only tried usbhid-ups. Honestly if this UPS is not supported its not a huge let down. I'd like to see it work and I probably should have checked to see if it was compatible with nut but my interest in nut only grew after acquiring ups's and messing with HA.

Briandr73, 2023-07-23:
Hello,

Adding something similar to the line below resolved one issue

MONITOR qnapups@localhost 1 upsmon 123456 master

For some reason this line was not liked in the ups.conf 

product=ST Series

Once removed I was able to run the upsc command and I got data returned

jimklimov, 2023-07-23:
Ah, single tokens with spaces should be quoted  :\

There are some issues filed about input/output values so maybe even a newer build won't immediately solve that part. As for voltage, I'd say it is always there, like height of a cliff regardless of there being an active waterfall. The current or load readings would be 0 in your example.

jimklimov, 2023-07-23:
As for "build" - you know, FOSS is there for anyone to tinker and compile (build) programs. Might get you closer to fixing those bogus readings for everyone ;)

Briandr73, 2023-09-06:
Hello,

Wanted to revisit for a quick question. How many CyberPower UPS's are going to detect via lsusb as a CP1500 AVR UPS. I know have two including the one discussed in this issue raised (SX550Ga) as well as a SL750U. At this point I probably won't need any more UPS's, but how many more are being detected as a CP1500 AVR. Is there anywhere to see which respective models will detect to a generic one such as CP1500 AVR.

jimklimov, 2023-09-06:
Well, that sounds like a firmware bug to me, so tge question os to CPS if they behave so. Numbers should generally sit with W or VA ratings.

Briandr73, 2023-09-06:
You may very well be correct, but just checked CyberPower's support page and I see no firmware updates for either of the two UPS's. My other UPS's that have Rmcard205 I do from time to time see firmware updates, but not these. I will be checking the input and output wattage based on manufacturer specs to compare to what NUT is telling me. I have a feeling they are not spot on.

Briandr73, 2023-09-07:
Just to confirm what I suspect you will say, there is no other driver I could try besides usbhid-ups. Sorry have not had a chance to check out the input and output voltages on my two units. hope to do so later.