networkupstools/nut #1998: NUT Server unresponsive after a day or two

tipplad, 2023-07-28:
Hi all and thanks in advance for the help.
I've got NUT-Server setup and working on a raspberry pi 4, all works for a day or two then both my clients cannot get any data from NUT-server.
I have to run "systemctl restart nut-driver.service" on the NUT-server and it works again for a few days.

Is there a way to automate the restart using crontab? Sorry, but of a noob so i'd need the exact line to enter on crontab if possible.
The UPS is an APC BX1400UI running.
If you need any other information to help with the issue then let me know.

jimklimov, 2023-07-28:
Hello, can you please check if you are running old NUT 2.7.4 or a newer 2.8.0 (or, unlikely to be packaged, some even newer snapshot)? The 2.8.0 release tried some changes to how drivers restart a connection (and/or kill off earlier running copies when one manually launches another); however some other currently open recent issues suggest it did not help in 100% of cases.

On explanation side: Some problems may be physical, either with cabling/noise or USB chips along the data path. Even if some of USB HW resets, the OS might not fully hand it over back to the driver (or with NUT 2.7.4 - the driver might not even expect that situation to re-discover the device).

Sorry I'm on a phone these days so can't help much with code examples or links. Check the `scripts/usb_resetter` however for a third-party script that ensures an OS USB port "recycling", which could be usefully coupled with a NUT driver restart.

Also there's a sample script for Solaris/illumos USB resetting and driver restart that includes use of `upsc` to gauge a data outage so as to proceed with the restart - this logic can be useful on all platforms.

tipplad, 2023-07-28:
Thanks for the response, the version is 2.7.4, only reinstalled the whole thing two days ago so thought that was the latest version.
How to update to 2.8?

jimklimov, 2023-07-28:
Depends on your distro - what they package (e.g. debian-ish "experimental" not "stable" repos, as it was for some months after 2.8.0 release). Or you can build your own in true open-source manner - see wiki footer about "in-place upgrade" builds.

tipplad, 2023-07-28:
I'm currently running Debian GNU/Linux 11 (bullseye) on the pi4

tipplad, 2023-07-31:
Any clear / easy instructions on how to upgrade to 2.8.0 with my setup?

jimklimov, 2023-07-31:
As far as NUT is concerned, you can build it and replace the packaged setup: https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests

Otherwise, need to figure out packaging for the particular OS/distro you are using and the repositories they use.

PilaScat, 2024-05-24:
same problem, unraid with nut 2.8.2, since months now