networkupstools/nut #1999: Driver changed after updating to 2.8.0 and ups shutdown stop working

yonesmit, 2023-07-29:
Hi!
My UPS is  Salicru SPS 1200 SOHO+ and I was using it in UNRAID with NUT version 2.7.4.1 and was running correctly. Now the UNRAID NUT plugin updated to use NUT version 2.8.0 and it detects my UPS as another (different usbhid-ups subdriver). The big problem is that UPS shutdown command stopped working.
The UPS data is:

VendorID: 06da
ProductID: ffff
Manufacturer: PPC
Product: Offline UPS
Serial Number: 000000000  

In 2.7.4.1 subdriver used was: MGE HID 1.43 and in 2.8.0 the subdriver used is: Phoenixtec/Liebert HID 0.41 and it doesn't accept the shutdown command!

I don't find any way to force NUT to use MGE subdriver.
Is there any way to fix or is it possible to make this manufacturer: PPC to use MGE subdriver? 

I'm attaching the shutdown command debug logs for both  versions

[shutdown_2.7.4.1.log](https://github.com/networkupstools/nut/files/12206641/shutdown_2.7.4.1.log)
[shutdown_2.8.0.log](https://github.com/networkupstools/nut/files/12206642/shutdown_2.8.0.log)

Thanks in advance
Best Regards,

jimklimov, 2023-07-29:
Hello, this sounds reminiscent of a discussion posted a couple of months ago; possibly the change was even proposed and integrated to NUT master branch - but can't say for sure, currently on phone not PC.

Beside that, there was a salicru subdriver merged some time ago, though maybe after the 2.8.0 release.

yonesmit, 2023-07-29:
Hi, although my UPS brand is salicru it doesn't use salicru vendorid (2e66), it uses 06da. 
So we can assume it's not a salicru but a rebranded ups.
The code 06da:ffff is assigned to 2 subdrivers: Phoenixtec/Liebert and MGE (only if vendor is "AEG" but mine is "PPC"). The problem is Phoenixtec/Liebert subdriver is very basic and doesn't support shutdown. Old version using MGE subdrivers was ok.
UPS shutdown is a very important feature, I need it to protect battery and to turn on my servers on again when power returns.
May be you can add same exception you do for 06da:ffff AEG vendor work if vendor is PPC. Or allow any way to select the subdriver to use.
Thanks

jimklimov, 2023-07-29:
Selection of a subdriver in usbhid-ups (similar to how nutdrv_qx and blazer allow it) was logged some time ago, but so far nobody contributed the code...

jimklimov, 2023-07-29:
That all said, "PPC" is something I see often. Wondering if that is a generic chip identifier - so for some devices the  Liebert mapping does work better... Or if there is some other field to (additionally) match for?..

jimklimov, 2023-07-29:
I think #1968 is the PR/discussion I mentioned, which ports some abilities into Liebert mapping, including a `shutdown.stop` backend for an instant command.

desertwitch, 2023-08-24:
Reported working with build based on current master:
https://forums.unraid.net/topic/60217-plugin-nut-v2-network-ups-tools/?do=findComment&comment=1298239