networkupstools/nut #2001: Disconnection Issue with tripplite_usb

Jsalas424, 2023-07-30:
My tripplite_usb connected UPS keeps disconnecting and reconnecting rapidly throwing a bunch of errors into my logs and interfering with my monitoring tool [nut_exporter](https://github.com/DRuggeri/nut_exporter). 

Relevant Syslog Output: 
```
Jul 30 11:44:46 Server tripplite_usb[2745772]: Successfully reconnected
Jul 30 11:44:46 Server tripplite_usb[2745772]: Using binary SMART protocol (3005)
Jul 30 11:44:47 Server tripplite_usb[2745772]: Unit ID: 65535
Jul 30 11:46:16 Server tripplite_usb[2745772]: libusb_set_report() returned 0 instead of 8
Jul 30 11:46:16 Server tripplite_usb[2745772]: Error reading 'V' value: Device detached? (error 0: error sending control message: Broken pipe)
Jul 30 11:46:16 Server tripplite_usb[2745772]: Reconnect attempt #1
Jul 30 11:46:16 Server tripplite_usb[2745772]: Successfully reconnected
Jul 30 11:46:16 Server tripplite_usb[2745772]: Using binary SMART protocol (3005)
Jul 30 11:46:17 Server tripplite_usb[2745772]: Unit ID: 65535
Jul 30 11:46:46 Server tripplite_usb[2745772]: libusb_set_report() returned 0 instead of 8
Jul 30 11:46:46 Server tripplite_usb[2745772]: Error reading 'P' value: Device detached? (error 0: error sending control message: Broken pipe)
Jul 30 11:46:46 Server tripplite_usb[2745772]: Reconnect attempt #1
Jul 30 11:46:47 Server tripplite_usb[2745772]: Successfully reconnected
Jul 30 11:46:47 Server tripplite_usb[2745772]: Using binary SMART protocol (3005)
Jul 30 11:46:48 Server tripplite_usb[2745772]: Unit ID: 65535
Jul 30 11:46:58 Server tripplite_usb[2745772]: libusb_set_report() returned 0 instead of 8
Jul 30 11:46:58 Server tripplite_usb[2745772]: Error reading L value: Device detached? (error 0: error sending control message: Broken pipe)
Jul 30 11:46:58 Server tripplite_usb[2745772]: Reconnect attempt #1
Jul 30 11:46:58 Server upsd[2745774]: Data for UPS [TrippLiteUPS] is stale - check driver
Jul 30 11:46:59 Server tripplite_usb[2745772]: Successfully reconnected
Jul 30 11:46:59 Server tripplite_usb[2745772]: Using binary SMART protocol (3005)
Jul 30 11:47:00 Server tripplite_usb[2745772]: Unit ID: 65535
Jul 30 11:47:01 Server upsd[2745774]: UPS [TrippLiteUPS] data is no longer stale
Jul 30 11:47:05 Server tripplite_usb[2745772]: libusb_set_report() returned 0 instead of 8
Jul 30 11:47:05 Server tripplite_usb[2745772]: Error reading M (min/max input voltage): Device detached? (error 0: error sending control message: Broken pipe)
Jul 30 11:47:05 Server tripplite_usb[2745772]: Reconnect attempt #1
Jul 30 11:47:05 Server upsd[2745774]: Data for UPS [TrippLiteUPS] is stale - check driver
Jul 30 11:47:05 Server tripplite_usb[2745772]: Successfully reconnected
Jul 30 11:47:05 Server tripplite_usb[2745772]: Using binary SMART protocol (3005)
Jul 30 11:47:06 Server tripplite_usb[2745772]: Unit ID: 65535
Jul 30 11:47:07 Server upsd[2745774]: UPS [TrippLiteUPS] data is no longer stale
```

Here's some debug data where you can see that 1 second in we get connectivity issues followed by reconnection at 9 seconds. 
```
root@TracheNodeB:~# /lib/nut/tripplite_usb -a TrippLiteUPS  -DDDDDD
Network UPS Tools - Tripp Lite OMNIVS / SMARTPRO driver 0.29 (2.7.4)
Warning: This is an experimental driver.
Some features may not function correctly.

   0.000000     debug level is '6'
   0.164905     Checking device (8087/8000) (003/002)
   0.233076     - VendorID: 8087
   0.233093     - ProductID: 8000
   0.233099     - Manufacturer: unknown
   0.233113     - Product: unknown
   0.233117     - Serial Number: unknown
   0.233131     - Bus: 003
   0.233135     - Device release number: 0004
   0.233139     Trying to match device
   0.233145     Device does not match - skipping
   0.233175     Checking device (1D6B/0002) (003/001)
   0.233230     - VendorID: 1d6b
   0.233243     - ProductID: 0002
   0.233246     - Manufacturer: unknown
   0.233249     - Product: unknown
   0.233264     - Serial Number: unknown
   0.233267     - Bus: 003
   0.233270     - Device release number: 0515
   0.233283     Trying to match device
   0.233287     Device does not match - skipping
   0.233304     Checking device (1D6B/0003) (004/001)
   0.264061     - VendorID: 1d6b
   0.264074     - ProductID: 0003
   0.264086     - Manufacturer: unknown
   0.264088     - Product: unknown
[... 166 lines cut ...]
Added `DEADTIME 25` and  `MAXAGE 25` to `/etc/nut/upsmon.conf`

`/etc/nut/nut.conf`:
```
MODE=netserver
```

`/etc/nut/upsmon.conf`:
```
MINSUPPLIES 1
SHUTDOWNCMD "/sbin/shutdown -h +0"
POLLFREQ 5
POLLFREQALERT 5
HOSTSYNC 15

DEADTIME 25
MAXAGE 25

POWERDOWNFLAG /etc/killpower
RBWARNTIME 43200
NOCOMMWARNTIME 300
FINALDELAY 5

MONITOR TrippLiteUPS@localhost 1 username password master
```

jimklimov, 2023-07-30:
Hello, is this some problem that started recently?

How long do you have/monitor this device?

Were there any other recent hardware changes (can we implicate e.g. USB cabling and/or noise on it - you do mention a new cable) or software ones (something else trying to kidnap the libusb device)?

Are you sure one instance of the driver is running; in case of restart - that the older instance did die (and no service/script/crontab resurrects it in a separate process tree)?

CC: @dzomaya @aquette

Jsalas424, 2023-07-31:
I will certainly look into all these factors, in the meantime I'm dropping off some new interesting logs:

```
Jul 30 20:36:48 Server tripplite_usb[2778403]: Successfully reconnected
Jul 30 20:36:48 Server tripplite_usb[2778403]: Using binary SMART protocol (3005)
Jul 30 20:36:48 Server tripplite_usb[2778403]: libusb_set_report() returned 0 instead of 8
Jul 30 20:36:48 Server tripplite_usb[2778403]: Could not reset watchdog. Please send model information to nut-upsdev mailing list
Jul 30 20:36:49 Server tripplite_usb[2778403]: libusb_get_interrupt() returned 0 instead of 8 while sending 3a 53 ac 0d 00 00 00 00 '.S......'
Jul 30 20:36:50 Server tripplite_usb[2778403]: libusb_get_interrupt() returned 0 instead of 8 while sending 3a 53 ac 0d 00 00 00 00 '.S......'
Jul 30 20:36:51 Server tripplite_usb[2778403]: libusb_get_interrupt() returned 0 instead of 8 while sending 3a 53 ac 0d 00 00 00 00 '.S......'
```

Data has been flowing okay for the past couple of hours until ~10min ago when it suddenly stopped altogether and now all I'm getting is `Jul 30 20:49:11 Server upsmon[2778364]: Poll UPS [TrippLiteUPS@localhost] failed - Driver not connected`

[image]

Jsalas424, 2023-08-02:
> Were there any other recent hardware changes (can we implicate e.g. USB cabling and/or noise on it - you do mention a new cable) or software ones (something else trying to kidnap the libusb device)?

No other hardware changes. This is running directly on my proxmox hypervisor, this is one of the only software's installed on top of the base OS. I checked every VM to make sure I wasn't inadvertently passing through any USB ports, but I don't pass through anything at all. 

> Are you sure one instance of the driver is running; in case of restart - that the older instance did die (and no service/script/crontab resurrects it in a separate process tree from e.g. systemd unit lifecycle)?

There are no cron jobs for the user

```
root@Server:~# crontab -l
no crontab for root
```

The systemwide cron jobs don't mention anything NUT related:

```
root@Server:~# cat /etc/crontab /etc/cron.d/*
# /etc/crontab: system-wide crontab
# Unlike any other crontab you don't have to run the `crontab'
# command to install the new version when you edit this file
# and files in /etc/cron.d. These files also have username fields,
# that none of the other crontabs do.

SHELL=/bin/sh
PATH=/usr/local/sbin:/usr/local/bin:/sbin:/bin:/usr/sbin:/usr/bin

# Example of job definition:
# .---------------- minute (0 - 59)
# |  .------------- hour (0 - 23)
# |  |  .---------- day of month (1 - 31)
# |  |  |  .------- month (1 - 12) OR jan,feb,mar,apr ...
# |  |  |  |  .---- day of week (0 - 6) (Sunday=0 or 7) OR sun,mon,tue,wed,thu,fri,sat
# |  |  |  |  |
# *  *  *  *  * user-name command to be executed
17 *    * * *   root    cd / && run-parts --report /etc/cron.hourly
25 6    * * *   root    test -x /usr/sbin/anacron || ( cd / && run-parts --report /etc/cron.daily )
47 6    * * 7   root    test -x /usr/sbin/anacron || ( cd / && run-parts --report /etc/cron.weekly )
52 6    1 * *   root    test -x /usr/sbin/anacron || ( cd / && run-parts --report /etc/cron.monthly )
#
30 3 * * 0 root test -e /run/systemd/system || SERVICE_MODE=1 /usr/lib/x86_64-linux-gnu/e2fsprogs/e2scrub_all_cron
10 3 * * * root test -e /run/systemd/system || SERVICE_MODE=1 /sbin/e2scrub_all -A -r
# cluster wide vzdump cron schedule
# Automatically generated file - do not edit

PATH="/usr/sbin:/usr/bin:/sbin:/bin"

PATH=/usr/local/sbin:/usr/local/bin:/sbin:/bin:/usr/sbin:/usr/bin

# TRIM the first Sunday of every month.
24 0 1-7 * * root if [ $(date +\%w) -eq 0 ] && [ -x /usr/lib/zfs-linux/trim ]; then /usr/lib/zfs-linux/trim; fi

# Scrub the second Sunday of every month.
24 0 8-14 * * root if [ $(date +\%w) -eq 0 ] && [ -x /usr/lib/zfs-linux/scrub ]; then /usr/lib/zfs-linux/scrub; fi
```

It's not the systemd timers:
```
root@Server:~# systemctl list-timers
NEXT                        LEFT          LAST                        PASSED       UNIT                         ACTIVATES
Wed 2023-08-02 18:54:50 EDT 3min 29s left Wed 2023-08-02 18:49:50 EDT 1min 30s ago smart_exporter.timer         smart_exporter.service
Wed 2023-08-02 22:32:57 EDT 3h 41min left Tue 2023-08-01 22:32:57 EDT 20h ago      systemd-tmpfiles-clean.timer systemd-tmpfiles-clean.service
Thu 2023-08-03 00:00:00 EDT 5h 8min left  Wed 2023-08-02 00:00:57 EDT 18h ago      logrotate.timer              logrotate.service
Thu 2023-08-03 00:00:00 EDT 5h 8min left  Wed 2023-08-02 00:00:57 EDT 18h ago      man-db.timer                 man-db.service
Thu 2023-08-03 00:25:39 EDT 5h 34min left Wed 2023-08-02 11:27:46 EDT 7h ago       apt-daily.timer              apt-daily.service
Thu 2023-08-03 05:54:00 EDT 11h left      Wed 2023-08-02 05:38:46 EDT 13h ago      pve-daily-update.timer       pve-dai>
Thu 2023-08-03 06:27:43 EDT 11h left      Wed 2023-08-02 06:45:46 EDT 12h ago      apt-daily-upgrade.timer      apt-dai>
Sun 2023-08-06 03:10:21 EDT 3 days left   Sun 2023-07-30 03:11:09 EDT 3 days ago   e2scrub_all.timer            e2scrub>
Mon 2023-08-07 01:26:07 EDT 4 days left   Mon 2023-07-31 00:33:57 EDT 2 days ago   fstrim.timer                 fstrim.>

9 timers listed.
[... 47 lines cut ...]
    Process: 2385 ExecStart=/sbin/upsd (code=exited, status=0/SUCCESS)
   Main PID: 2386 (upsd)
      Tasks: 1 (limit: 38361)
     Memory: 1.2M
        CPU: 12.515s
     CGroup: /system.slice/nut-server.service
             └─2386 /lib/nut/upsd

root@Server:~# cat /lib/systemd/system/nut-server.service
[Unit]
Description=Network UPS Tools - power devices information server
After=local-fs.target network.target nut-driver.service
# We don't Require drivers to be successfully started! This would be
# a change of behavior compared to init SysV, and could prevent from
# accessing successfully started, at least to audit a system.
Wants=nut-driver.service
Before=nut-monitor.service

[Service]
ExecStart=/sbin/upsd 
Type=forking

[Install]
WantedBy=multi-user.target
```

Jsalas424, 2023-11-01:
Resolved with upgrade to 2.8.0. 
[image]

There are still occasional disconnects but it's able to come back online now unlike in v2.7.4. 

```
Nov 01 16:33:02 Server upsd[1092085]: Data for UPS [TrippLiteUPS] is stale - check driver
Nov 01 16:33:03 Server nut-monitor[1092077]: Poll UPS [TrippLiteUPS@localhost] failed - Data stale
Nov 01 16:33:03 Server kernel: usb 3-6: new low-speed USB device number 70 using xhci_hcd
Nov 01 16:33:03 Server kernel: usb 3-6: New USB device found, idVendor=09ae, idProduct=0001, bcdDevice= 0.0a
Nov 01 16:33:03 Server kernel: usb 3-6: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Nov 01 16:33:03 Server kernel: usb 3-6: Product: TRIPP LITE SMART500RT1U 
Nov 01 16:33:03 Server kernel: usb 3-6: Manufacturer: Tripp Lite 
Nov 01 16:33:03 Server kernel: hid-generic 0003:09AE:0001.0140: hiddev0,hidraw0: USB HID v1.10 Device [Tripp Lite  TRIPP LITE SMART500RT1U ] on usb-0000:00:14.0-6/input0
Nov 01 16:33:06 Server tripplite_usb[1921583]: libusb_set_report() returned 0 instead of 8
Nov 01 16:33:06 Server tripplite_usb[1921583]: Error reading S value: Device detached? (error 0: Success)
Nov 01 16:33:06 Server tripplite_usb[1921583]: Reconnect attempt #4
Nov 01 16:33:06 Server tripplite_usb[1921583]: Successfully reconnected
Nov 01 16:33:07 Server tripplite_usb[1921583]: Using binary SMART protocol (3005)
Nov 01 16:33:07 Server tripplite_usb[1921583]: Unit ID: 65535
Nov 01 16:33:08 Server nut-monitor[1092077]: Poll UPS [TrippLiteUPS@localhost] failed - Data stale
Nov 01 16:33:12 Server nut-server[1092085]: UPS [TrippLiteUPS] data is no longer stale
Nov 01 16:33:12 Server upsd[1092085]: UPS [TrippLiteUPS] data is no longer stale
Nov 01 16:33:13 Server nut-monitor[1092077]: Communications with UPS TrippLiteUPS@localhost established
Nov 01 16:33:13 Server nut-monitor[1921710]: Network UPS Tools upsmon 2.8.0
```

The only other interesting point is that I'm getting the following error:
`libusb1: Could not open any HID devices: insufficient permissions on everything`

But that's a fight for another day.

closed 2023-11-01 by Jsalas424