networkupstools/nut #2008: what is the warning?

bjlockie, 2023-08-03:
I did a deep battery test and the result has a warning.
What is the warning?

```
battery.charge: 48
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 1092
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 12.3
battery.voltage.nominal: 12
device.mfr: CPS
device.model: UPS CP850AVRLCD 
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: CyberPower HID 0.4
driver.version.internal: 0.41
input.transfer.high: 140
input.transfer.low: 90
input.voltage: 123.0
input.voltage.nominal: 120
output.voltage: 124.0
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 17
ups.mfr: CPS
ups.model: UPS CP850AVRLCD 
ups.productid: 0501
ups.realpower.nominal: 510
ups.status: OL CHRG
ups.test.result: Done and warning
ups.timer.shutdown: -60
ups.timer.start: 0
ups.vendorid: 0764
```

bjlockie, 2023-08-03:
If I unplug it from the wall and I get the expected runtime then is it ok?

jimklimov, 2023-08-03:
So, this text seems to be among decodings of standard USB UPS HID status values for device test, referenced from many subdrivers as `UPS.BatterySystem.Test` or `UPS.BatterySystem.Battery.Test` etc.: https://github.com/networkupstools/nut/blob/80767e3961ac1a1694ed2a713948ab6989c8a95b/drivers/usbhid-ups.c#L378-L387

The rest of definition is up to hardware/firmware/vendor...