networkupstools/nut #2011: SMT750i cable?

tonygh77, 2023-08-04:
Hi, I found this compatibility for my UPS
"Smart-UPS SMT/SMX/SURTD
Microlink models with RJ45 socket - they *require* AP9620 SmartSlot expansion card and smart cable"
What does smart cable mean? a simple USB cable or a RS232 - USB cable?
Many thanks for your time

jimklimov, 2023-08-04:
Hello, good question. Currently googling for the "AP9620" comes up with https://www.apc.com/us/en/product/AP9620/legacy-communications-smartslot-card/ which per photo does not have an RJ45 socket, so either it is a different HW revision or I misunderstood that comment and see next paragraph :)
[image]

The description *could* however be read as the UPS itself having an RJ45 port (e.g. for serial over UTP cabling, possibly with uncommon wire layout and/or active signalling - APC did such tricks with plain serial cables back in the day). Then the card adding a DB-9 serial and an USB-A port makes sense - and the former may still need the special APC serial cable.

Overall this seems to match "microlink" commentary in the driver man page, see e.g. https://networkupstools.org/docs/man/apcsmart.html and FWIW lingering issue #139 about support for the protocol used in newer APC devices (allegedly their USB HID offering is also lackluster).

tonygh77, 2023-08-04:
Thank you for your reply
I found the used board (AP9620) at a great price
My ups should support this 'apcsmart' driver
The driver page mentions this cable: 940-0024C
This is a simple RS232 to RS232 cable
I ask you another question:
My nuc doesn't have a RS232 port, can I use a USB to RS232 adapter in addition to the RS232 + RS232 cable?

jimklimov, 2023-08-04:
I think so, the computer-side expectation is of a standard serial/uart port - just the APC side (port+cable) are sprinkled with pixie dust. So the USB-serial dongle would be `port=/dev/ttyUSB0` or similar on a Linux system, not sure about others but I think it would follow common COM port names (`/dev/cuua`, `COM5` etc.)?...

tonygh77, 2023-08-04:
Perfect, I will buy everything and then I will update the discussion
Thanks again for your welcome support