networkupstools/nut #2014: Changing the shutdown battery level for UPS Legrand

ITPotatoo, 2023-08-06:
Hello
i want to know how i can change the shutdown battery charge level, so my nut client linux host shutdown when for exemple the battery level fall to 40%
i tried with upswr command in a client nut server : 

upsrw -s battery.charge.low=40 Legrand@192.168.1.10

i got this error message
Unexpected response from upsd: ERR ACCES-DNIED

here the information of my UPS :

the battery.charge.low value does not exist

upsc Legrand@localhost
Init SSL without certificate database
battery.charge: 100
battery.voltage: 2.22
battery.voltage.high: 78.00
battery.voltage.low: 62.40
battery.voltage.nominal: 72.0
device.mfr: Legrand
device.model: Dk+ 3kVA
device.type: ups
driver.name: nutdrv_qx
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 1
driver.parameter.port: auto
driver.parameter.productid: 0035
driver.parameter.synchronous: auto
driver.parameter.vendorid: 1CB0
driver.version: 2.8.0
driver.version.data: Megatec 0.06
driver.version.internal: 0.32
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.current.nominal: 13.0
input.frequency: 49.9
input.frequency.nominal: 50
input.voltage: 222.8
input.voltage.fault: 105.0
input.voltage.nominal: 230
output.voltage: 229.6
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware: AS1L185H
ups.load: 14
ups.productid: 0035
ups.status: OL
ups.temperature: 45.0
ups.type: online
ups.vendorid: 1cb0

Please can you help, i don't  find the solution in the internet

jimklimov, 2023-08-06:
For `upsrw` you need to pass the user name and password, as defined in your `upsd.conf` (see docs/examples if nothing is defined yet; check OS permissions on the file as insecure ones ought to be ignored) as an account with a `SET` permission.

jimklimov, 2023-08-06:
Generally, though, you might have to set such value in `ups.conf` as an `override.battery.charge.low`

And note there is currently an open discussion whether this works as it should (at least as far as propagating the value into the device, for UPSes that do not remember settings after power cycling).

ITPotatoo, 2023-08-06:
Here my configuration of upsd 

upsd.conf                                                     
LISTEN 0.0.0.0 3493

upsd.users                                                    
[admin]
password = secret
instcmds = all
actions = SET

when i use the command : upsrw -s battery.charge.low=40 Legrand@192.168.1.10

Username (root) : admin
Password : secret

is there anything wrong ?

ITPotatoo, 2023-08-06:
Unexpected response from upsd: ERR VAR-NOT-SUPPORTED

jimklimov, 2023-08-06:
Seems right regarding credentials (and since it no longer complains about access denied - OS permissions on the file are also ok)

jimklimov, 2023-08-06:
So the remaining bit is that this var for this subdriver is not defined, possibly the device does not handle it, or the driver authors were not aware of protocol bits to request/query the value (if they exist).

jimklimov, 2023-08-06:
I think there were some discussions over time, recently too, that an "LB" state could be raised by the drivers generally - based on available `battery.charge` or `battery.runtime` readings and respective `(override.|default.)*.low` settings. But I do not quickly see anything in current codebase to that effect, so probably a PR is welcome.

Alternatives per discussion could be to take a look at `upssched` and handling scripts you write that it would call, if it can do something in this regard (especially to ensure different charges/runtimes remaining to shut down different hosts based on their priority to you).

ITPotatoo, 2023-08-06:
I didnt get it, what i can try or configure to solve it ?
im sorry, im new to this NUT solution :)

jimklimov, 2023-08-06:
As a quick approach, I'd suggest investigating (googling) if `upssched` makes something like this possible, to work around UPSes/drivers that do not have a built-in way to report a low-power situation by customized settings.

I am a bit surprised to not see a built-in solution based on those measurements after the decades that NUT is out there, so maybe I'm missing or forgetting something (and/or spoiled by experience with more capable devices/drivers which did not need workarounds), but it *seems* that currently it is not there, or too well hidden :)

I'll ask if the mailing list users have fresher experience with that.