networkupstools/nut #2015: apcsmart driver "nument or entlen out of range" when connecting

finos2, 2023-08-11:
After clean installing NUT 2.8.0 on Debian 12 for use with APC Smart-UPS RT 2000 over serial, while starting the `apcsmart` driver got the error 
```
apc_getcaps: nument (-1) or entlen (3) out of range
nument or entlen out of range
Please report this error
ERROR: capability overflow!
``` 
followed by the driver quitting.

A similar error is produced by the `apcsmart-old` driver.

Replacing the `apcsmart` binary with the one from 2.7.4 seems to resolve the issue for now.

jimklimov, 2023-08-12:
That rings a bell a bit, at least the driver was recently discussed. Not on PC now - can you please search issues for apcsmart?

Not sure if this driver was addressed by code changes after 2.8.0 release, but it's always worth trying a custom build of current NUT to know if the issue still needs solving :} Wiki has an article on "in-place" build configs to have a custom build configured similarly to your packaged one; a driver test run can be done from a build workspace without impacting the OS.

Also, can you try starting the driver binary with debug options (wiki suggests some ways) to help pinpoint where/why it gets that "bad" data.

jimklimov, 2023-08-12:
So, dug a bit into code history - the message originates from https://github.com/networkupstools/nut/commit/beb4b549b47c472c6e6680083da171ae14dc6438

So before this fix, the code went:
https://github.com/networkupstools/nut/blame/cbfe9ab178183a85a216fd8f789895d786d3055c/drivers/apcsmart.c#L1019-L1023

...and might end up with negative `nument` and/or `entlen`. Which, looking a bit more at the rest of code, would make the rest of the cycle no-op. Probably bailing out was too harsh here, reporting and ignoring (going to next loop) may be the proper approach after all.

jimklimov, 2023-08-12:
CC @aquette as the initial committer (though I guess it is further shrouded in SVN history).

jimklimov, 2023-08-12:
Posted a PR - if you have a chance to build and test that branch, would be great!

PR #2016 (2023-08-16): drivers/apcsmart(-old).c: do not abort on bad caps, skip them like before
Closes: #2015
commit bc14bf8f (2023-08-16): drivers/apcsmart(-old).c: do not abort on bad caps, skip them like before [#2015]

Closes: #2015
closed 2023-08-16 by jimklimov