networkupstools/nut #2017: Parse error on sock: Invalid ctx buffer: usbhid-ups regularly looses connection (CyperPower UPS)

segdy, 2023-08-15:
I have a CyperPower OR500LCDRM1U connected via USB to nut on Debian (proxmox). After some time (hours?), the connection to UPS is lost and the driver (nut-driver.service) becomes stale. This looke like:

```
# upsc ups

Broadcast message from nut@pve1 (somewhere) (Mon Aug 14 17:35:48 2023):

UPS ups@localhost is unavailable

Error: Connection failure: Connection timed out
#
```

```
# systemctl status nut-driver
● nut-driver.service - Network UPS Tools - power device driver controller
     Loaded: loaded (/lib/systemd/system/nut-driver.service; static)
     Active: active (running) since Tue 2023-08-01 23:55:33 PDT; 1 weeks 5 days ago
    Process: 1506046 ExecStart=/sbin/upsdrvctl start (code=exited, status=0/SUCCESS)
   Main PID: 1506049 (usbhid-ups)
      Tasks: 1 (limit: 9331)
     Memory: 476.0K
        CPU: 18.100s
     CGroup: /system.slice/nut-driver.service
             └─1506049 /lib/nut/usbhid-ups -a ups

Aug 04 02:24:14 pve1 usbhid-ups[1506049]: Parse error on sock: Invalid ctx buffer
Aug 04 02:24:14 pve1 usbhid-ups[1506049]: Parse error on sock: Invalid ctx buffer
Aug 04 02:24:14 pve1 usbhid-ups[1506049]: Parse error on sock: Invalid ctx buffer
Aug 04 02:24:14 pve1 usbhid-ups[1506049]: Parse error on sock: Invalid ctx buffer
Aug 04 02:24:14 pve1 usbhid-ups[1506049]: Parse error on sock: Invalid ctx buffer
Aug 04 02:24:14 pve1 usbhid-ups[1506049]: Parse error on sock: Invalid ctx buffer
Aug 04 02:24:14 pve1 usbhid-ups[1506049]: Parse error on sock: Invalid ctx buffer
Aug 04 02:24:14 pve1 usbhid-ups[1506049]: Parse error on sock: Invalid ctx buffer
Aug 04 02:24:14 pve1 usbhid-ups[1506049]: Parse error on sock: Invalid ctx buffer
Aug 04 02:24:14 pve1 usbhid-ups[1506049]: Parse error on sock: Invalid ctx buffer
#
```

Extremely frustrating because UPS is a critical component. It might be that USB reinitializes, temporarily device becomes unavailable for a few seconds or similar. I don't know. Nothing in particular in the logs but I can't really monitor it either because it only happens after several hours and then I see realize it the next day.

My logs become filled with messages like:

```
Jul 30 00:19:36 pve1 upsmon[2066730]: UPS [ups@localhost]: connect failed: Connection failure: Connection timed out
Jul 30 00:19:36 pve1 upsmon[2066730]: UPS ups@localhost is unavailable
Jul 30 00:21:51 pve1 upsmon[2066730]: UPS [ups@localhost]: connect failed: Connection failure: Connection timed out
Jul 30 00:24:06 pve1 upsmon[2066730]: UPS [ups@localhost]: connect failed: Connection failure: Connection timed out
Jul 30 00:26:21 pve1 upsmon[2066730]: UPS [ups@localhost]: connect failed: Connection failure: Connection timed out
Jul 30 00:26:21 pve1 upsmon[2066730]: UPS ups@localhost is unavailable
```

When I `strace -p` the `usbhid-ups -a ups` process, I get:

```
select(7, [5 6], NULL, NULL, {tv_sec=1, tv_usec=898414}) = 0 (Timeout)
ioctl(4, USBDEVFS_SUBMITURB, 0x7ffe4f699750) = 0
ioctl(4, USBDEVFS_REAPURBNDELAY, 0x7ffe4f699718) = -1 EAGAIN (Resource temporarily unavailable)
select(5, NULL, [4], NULL, {tv_sec=0, tv_usec=1000}) = 0 (Timeout)
ioctl(4, USBDEVFS_REAPURBNDELAY, 0x7ffe4f699718) = -1 EAGAIN (Resource temporarily unavailable)
select(5, NULL, [4], NULL, {tv_sec=0, tv_usec=1000}) = 0 (Timeout)
ioctl(4, USBDEVFS_REAPURBNDELAY, 0x7ffe4f699718) = -1 EAGAIN (Resource temporarily unavailable)
select(5, NULL, [4], NULL, {tv_sec=0, tv_usec=1000}) = 0 (Timeout)
ioctl(4, USBDEVFS_REAPURBNDELAY, 0x7ffe4f699718) = -1 EAGAIN (Resource temporarily unavailable)
select(5, NULL, [4], NULL, {tv_sec=0, tv_usec=1000}) = 0 (Timeout)
ioctl(4, USBDEVFS_REAPURBNDELAY, 0x7ffe4f699718) = -1 EAGAIN (Resource temporarily unavailable)
select(5, NULL, [4], NULL, {tv_sec=0, tv_usec=1000}) = 0 (Timeout)
ioctl(4, USBDEVFS_REAPURBNDELAY, 0x7ffe4f699718) = -1 EAGAIN (Resource temporarily unavailable)
select(5, NULL, [4], NULL, {tv_sec=0, tv_usec=1000}) = 0 (Timeout)
ioctl(4, USBDEVFS_REAPURBNDELAY, 0x7ffe4f699718) = -1 EAGAIN (Resource temporarily unavailable)
select(5, NULL, [4], NULL, {tv_sec=0, tv_usec=1000}) = 0 (Timeout)
ioctl(4, USBDEVFS_REAPURBNDELAY, 0x7ffe4f699718) = -1 EAGAIN (Resource temporarily unavailable)
select(5, NULL, [4], NULL, {tv_sec=0, tv_usec=1000}) = 0 (Timeout)
ioctl(4, USBDEVFS_REAPURBNDELAY, 0x7ffe4f699718) = -1 EAGAIN (Resource temporarily unavailable)
select(5, NULL, [4], NULL, {tv_sec=0, tv_usec=1000}) = 0 (Timeout)
ioctl(4, USBDEVFS_REAPURBNDELAY, 0x7ffe4f699718) = -1 EAGAIN (Resource temporarily unavailable)
select(5, NULL, [4], NULL, {tv_sec=0, tv_usec=1000}) = 0 (Timeout)
ioctl(4, USBDEVFS_REAPURBNDELAY, 0x7ffe4f699718) = -1 EAGAIN (Resource temporarily unavailable)
select(5, NULL, [4], NULL, {tv_sec=0, tv_usec=1000}) = 0 (Timeout)
ioctl(4, USBDEVFS_REAPURBNDELAY, 0x7ffe4f699718) = -1 EAGAIN (Resource temporarily unavailable)
select(5, NULL, [4], NULL, {tv_sec=0, tv_usec=1000}) = 1 (out [4], left {tv_sec=0, tv_usec=251})
ioctl(4, USBDEVFS_REAPURBNDELAY, 0x7ffe4f699718) = 0
```

If I restart the nut driver (`systemctl restart nut-driver.service`) it always works again (for a few hours).

I have created `/etc/udev/rules.d/90-nut-ups.rules` with content:

```
# Rule for the UPS - use lsusb for idVendor and idProduct
ACTION=="add", SUBSYSTEM=="usb", ATTR{idVendor}=="0764", ATTR{idProduct}=="0601", MODE="0660", GROUP="nut"
```

In my `/etc/nut/ups.conf` I have:

```
maxretry = 3

[ups]
driver = usbhid-ups
port = auto
desc = "CyberPower PR1500LCDRT2U"
vendorid = 0764
productid = 0601
```

My UPS becomes available as the following device file:

```
# ls -la /dev/bus/usb/001/004 
crw-rw-r-- 1 root nut 189, 3 Aug  4 02:30 /dev/bus/usb/001/004
```

I am not providing the remaining nut config files because the issue is clearly related to the usbhid-ups driver and connection via `upsc ups` fails.

Finally, version infos:

```
# cat /etc/debian_version 
11.6
# uname -a
Linux pve1 5.15.85-1-pve #1 SMP PVE 5.15.85-1 (2023-02-01T00:00Z) x86_64 GNU/Linux
# upsc -V
Network UPS Tools upscmd 2.7.4
# 
```

segdy, 2023-08-15:
Additional note: Curiously, when I do a strace via `strace -p $(pidof usbhid-ups)` I suddenly get and magically get

```
Broadcast message from nut@pve1 (somewhere) (Mon Aug 14 17:45:09 2023):

Communications with UPS ups@localhost established
```

Of course, after some time, driver will be stale again. So, whatever is stuck here, it magically not only gets unstuck by reloading the driver but also via stracing...

jimklimov, 2023-08-15:
Hi, my first reaction here is that of several red flags:

1) NUT v2.7.4 is quite old, sadly many distros still ship it as a package. Can you give it a shot to replace it with the current git master build? See wiki page on "in-place replacement" configs. Crucially, there were fixes related to auto-reconnection and I wonder if some would directly help this case.

2) Also on wiki, and in issue history, CPS (and some other vendors) is known for UPSes with regular disconnections, and if USB support in the OS finds a "new" connection - it tends to make it anew, permissions and everything, so it has to be re-delegated to NUT. Should be solved by (1).

3) Also there are tricks like `usb_resetter` to actively cause USB port resets when connection is lost...

jimklimov, 2023-08-15:
UPDATE: Wiki links for the above subjects:

* https://github.com/networkupstools/nut/wiki/Troubleshooting-eventual-disconnections-(Data-stale)
* https://github.com/networkupstools/nut/wiki/CyberPower-Systems-(CPS)-know-how
* https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests

segdy, 2023-08-23:
Thank you!!
With pollinterval and MAXAGE adjusted it even worked for my (old) version.