networkupstools/nut #2018: Eaton 5E 700 G2 compatibily

vosatka, 2023-08-15:
Im reporting promising signs of compatibility with new Eaton UPS.

`root@proc:~# service nut-server status
● nut-server.service - Network UPS Tools - power devices information server
     Loaded: loaded (/lib/systemd/system/nut-server.service; enabled; preset: enabled)
     Active: active (running) since Tue 2023-08-15 22:44:58 CEST; 14s ago
    Process: 13651 ExecReload=/sbin/upsd -c reload -P $MAINPID (code=exited, status=0/SUCCESS)
   Main PID: 13563 (upsd)
      Tasks: 1 (limit: 18886)
     Memory: 624.0K
        CPU: 17ms
     CGroup: /system.slice/nut-server.service
             └─13563 /lib/nut/upsd -F

Aug 15 22:44:58 proc upsd[13563]: Running as foreground process, not saving a PID file
Aug 15 22:44:59 proc systemd[1]: Reloading nut-server.service - Network UPS Tools - power devices information server...
Aug 15 22:44:59 proc nut-server[13563]: mainloop: Interrupted system call
Aug 15 22:44:59 proc nut-server[13563]: SIGHUP: reloading configuration
Aug 15 22:44:59 proc upsd[13563]: mainloop: Interrupted system call
Aug 15 22:44:59 proc nut-server[13651]: Network UPS Tools upsd 2.8.0
Aug 15 22:44:59 proc systemd[1]: Reloaded nut-server.service - Network UPS Tools - power devices information server.
Aug 15 22:44:59 proc nut-server[13563]: Connected to UPS [eaton5e7002g]: usbhid-ups-eaton5e7002g
Aug 15 22:44:59 proc upsd[13563]: SIGHUP: reloading configuration
Aug 15 22:44:59 proc upsd[13563]: Connected to UPS [eaton5e7002g]: usbhid-ups-eaton5e7002g`

status returns couple disturbing rows, but at the end its looks connected and OK

`upsc eaton650
Init SSL without certificate database
Error: Unknown UPS
root@proc:~# upsc eaton5e7002g
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 20
battery.runtime: 3345
battery.type: PbAc
device.mfr: EATON
device.model: Eaton 5E 700 G2
device.serial: BLANK
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0463
driver.version: 2.8.0
driver.version.data: MGE HID 1.46
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.voltage: 239.0
outlet.1.status: on
outlet.desc: Main Outlet
outlet.id: 0
outlet.switchable: no
output.frequency: 50.0
output.frequency.nominal: 50
output.voltage: 239.0
output.voltage.nominal: 230
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.firmware: 01.01.0007
ups.load: 9
ups.mfr: EATON
ups.model: Eaton 5E 700 G2
ups.power: 82
ups.power.nominal: 700
ups.productid: ffff
ups.realpower: 32
ups.realpower.nominal: 360
ups.serial: BLANK
ups.start.battery: yes
ups.status: OL
ups.timer.shutdown: -1
ups.timer.start: -1
ups.vendorid: 0463`

`upscmd -l eaton5e7002g
Instant commands supported on UPS [eaton5e7002g]:

beeper.disable - Disable the UPS beeper
beeper.enable - Enable the UPS beeper
beeper.mute - Temporarily mute the UPS beeper
beeper.off - Obsolete (use beeper.disable or beeper.mute)
beeper.on - Obsolete (use beeper.enable)
load.off - Turn off the load immediately
load.off.delay - Turn off the load with a delay (seconds)
load.on - Turn on the load immediately
load.on.delay - Turn on the load with a delay (seconds)
shutdown.return - Turn off the load and return when power is back
shutdown.stayoff - Turn off the load and remain off
shutdown.stop - Stop a shutdown in progress`

`upsrw eaton5e7002g
[battery.charge.low]
Remaining battery level when UPS switches to LB (percent)
Type: STRING
Maximum length: 5
Value: 20

[outlet.desc]
Outlet description
Type: STRING
Maximum length: 20
Value: Main Outlet

[output.voltage.nominal]
Nominal output voltage (V)
Type: ENUM STRING
Maximum length: 3
Option: "200"
Option: "208"
Option: "220"
Option: "230" SELECTED
Option: "240"

[ups.delay.shutdown]
Interval to wait after shutdown with delay command (seconds)
Type: STRING
Maximum length: 10
Value: 20

[ups.delay.start]
Interval to wait before (re)starting the load (seconds)
Type: STRING
Maximum length: 10
Value: 30

[ups.start.battery]
Allow to start UPS from battery
Type: STRING
Maximum length: 5
Value: yes`

Hope this helps. Cheers

AnhQuanTrl, 2024-10-19:
Hi @vosatka are these UPS (Eaton 5E Gen 2) fully compatible with `nut`? They are not listed in the compatible list right now.

jimklimov, 2024-10-19:
"Fully" is a big word; absent a spec of everything they should be capable of (on program/signal) side and a state machine from a vendor, perhaps detailed per device model, hardware and firmware revisions, we can talk about sufficiently well supported for monitoring and some management.

Note that with work in PR #2637 and some others being done "right now" (for the past few days), more features can get detected, reported and tamed.

As for "compatible list", its maintenance takes time and is a sort of menial work that gets delayed compared to adding features and fixing bugs. The list would eventually grow (also happens now) as a result of reports like this one. There are by now a couple hundred tagged issues to sift through to update the list... (As ever, help is welcome).