networkupstools/nut #2019: DKC Rackmount Pro,3000

DS455, 2023-08-17:
Tell me which driver to use for DKC Rackmount Pro,3000?
https://www.dkceurope.com/

[image]

jimklimov, 2023-08-17:
Hello, looking at `ATCL` in the identification string, it seems this is one of the few devices with that (sadly very non-unique) Vendor+Product IDs that should be actually supported by `nutdrv_atcl_usb` driver. Most others with such ID have a variant of Megatec Q<number> protocols, so are handled by `nutdrv_qx`...

DS455, 2023-08-17:
Driver `nutdrv_qx` worked, thanks!
[image]