networkupstools/nut #2021: [HCL] Powershield Ninja Slimline Compatibility

shanelord01, 2023-08-22:
The [Powershield Ninja Slimline](https://powershield.com.au/solutions/ninja-slimline-ups) LifePO4 based model works correctly using the _blazer_ser_ driver over [CableCreation USB RS232 Cable](https://www.amazon.com.au/Adapter-CableCreation-Converter-Chipset-register/dp/B0758B6MK6) 
---
**ups.conf**
```
[ninja]
driver = blazer_ser
desc = "Ninja UPS"
port = /dev/ttyUSB0
vendorid = 067b
productid = 23a3
```

Note: port should be the USB port of your adapter - user must have access to dialout group to access /dev/tty. vendorid & productid are those reported by the USB RS232 adapter cable and may not be required.

----
**upsc**
```
Init SSL without certificate database
battery.charge: 100
battery.voltage: 13.70
battery.voltage.high: 13.00
battery.voltage.low: 10.40
battery.voltage.nominal: 12.0
device.type: ups
driver.name: blazer_ser
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/ttyUSB0
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.internal: 1.57
input.current.nominal: 2.0
input.frequency: 49.8
input.frequency.nominal: 50
input.voltage: 241.2
input.voltage.fault: 241.2
input.voltage.nominal: 240
output.voltage: 240.8
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.load: 0
ups.status: OL
ups.type: offline / line interactive
```

jimklimov, 2023-08-22:
Thanks. Is the `ups.load==0` correct? (Nothing substantial was connected to the UPS?)

shanelord01, 2023-08-22:
Ahh. Didn’t notice that as everything else working. No it’s not. Any ideas what I can do to test/fix?

jimklimov, 2023-08-22:
For one, if you can try with newer NUT (2.8.0 or even perhaps a custom build of master branch, can test from the build workspace without changes to the OS - see wiki for in-place builds), or with `nutdrv_qx` as the newer generation of driver family for variants of Megatec Q<x> protocpls.

shanelord01, 2023-08-23:
> For one, if you can try with newer NUT

Endeavouring to do this now. On Rpi4 using RPi 64bit OS (Debian bullseye) and it is default 2.7.4 - wish there was an easy way to install 2.8.x

shanelord01, 2023-08-23:
OK - update. Followed [this step by step](https://github.com/networkupstools/nut/wiki/Building-NUT-on-Debian,-Raspbian-and-Ubuntu#x86_64-64-bit) and now running 2.8.0-2306-g3d9630e0f.

```
[ninja]
driver = nutdrv_qx
desc = "Ninja UPS"
port = /dev/ttyUSB0
```

and

```
Init SSL without certificate database
battery.charge: 100
battery.voltage: 13.70
battery.voltage.high: 13.00
battery.voltage.low: 10.40
battery.voltage.nominal: 12.0
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: nutdrv_qx
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/ttyUSB0
driver.parameter.synchronous: auto
driver.state: quiet
driver.version: 2.8.0-2306-g3d9630e0f
driver.version.data: Voltronic-QS 0.08
driver.version.internal: 0.33
driver.version.usb: libusb-0.1 (or compat)
input.voltage: 239.0
input.voltage.fault: 241.2
output.current.nominal: 2.0
output.frequency: 49.8
output.frequency.nominal: 50
output.voltage: 238.5
output.voltage.nominal: 240
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware.aux: PM-V
ups.load: 0
ups.status: OL
ups.type: offline / line interactive
```

jimklimov, 2023-08-23:
Thanks for the update. So perhaps the device does not report it (or does so in some way our drivers do not recognize currently/still).

The `upsdrvctl` itself is a (helper) program so it does need to be run every boot-up. However usually either legacy init-scripts or modern services (systemd there) handle this.

If you've configured your build with systemd support, it should have installed the files for `nut-server`, `nut-driver-enumerator` and templates for `nut-driver@` among other things - but you have to `systemd enable <unit>` them to start on reboot (and separately `systemd start <unit>` to start one now).

Some samples to copy-paste are at the end of https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests page.

shanelord01, 2023-08-24:
Thanks. When I get a spare moment I'll put some much heavier load on it to see if anything shows. If nothing then I guess no luck. If something, perhaps a scaling issue? I'll revert once I know more.