networkupstools/nut #2024: Multiplier Issues with liebert-esp2

Funtime60, 2023-08-25:
My GXT2-2000RT120 is reporting a battery voltage of 560.0VDC. I know for a fact that it's actually 56VDC.

Either the GXT2-2000RT120 needs different multipliers than the GXT2-6000RT208 or it's not being assigned the correct multiplier at all.

jimklimov, 2023-08-25:
Which NUT version, please? Possibly a data dump to help identify the device and firmware in our code?..

Funtime60, 2023-08-25:
I'm pretty sure it's NUT 2.7.4, I'm not sure how to do a data dump or what firmware you're asking for. As for where in the code? I'm pretty sure it's in here https://github.com/networkupstools/nut/blob/master/drivers/liebert-esp2.c
Maybe around line #59, #260, or #261

Funtime60, 2023-08-28:
Thank you for the suggestion, I always forget that the raspbian repos are pretty out dates. I give it a try.