networkupstools/nut #2025: snmp-ups compiler warnings on Slackware v15.1

desertwitch, 2023-08-27:
When building on Slackware v15.0 with gcc 11.2.0 everything works without any problems.

When building on Slackware v15.1 with gcc 13.2.0 I'm getting the following warnings:

```
snmp-ups.c: In function â€˜su_alarm_setâ€™:
snmp-ups.c:1803:17: warning: dangling pointer â€˜info_valueâ€™ to â€˜alarm_info_value_moreâ€™ may be used [-Wdangling-pointer=]
 1803 |                 alarm_set(info_value);
      |                 ^~~~~~~~~~~~~~~~~~~~~
snmp-ups.c:1792:30: note: â€˜alarm_info_value_moreâ€™ declared here
 1792 |                         char alarm_info_value_more[SU_LARGEBUF + 32]; /* can sprintf() SU_LARGEBUF plus markup into here */
      |                
```

```
At top level:
cc1: note: unrecognized command-line option â€˜-Wno-unknown-warning-optionâ€™ may have been intended to silence earlier diagnostics
cc1: note: unrecognized command-line option â€˜-Wno-reserved-identifierâ€™ may have been intended to silence earlier diagnostics
```

Despite earlier configure tests confirming that those flags are in fact recognized:

```
checking whether C compiler accepts -Wno-reserved-identifier... yes
checking whether C++ compiler accepts -Wno-reserved-identifier... yes
```

```
checking whether C compiler accepts -Wno-unknown-warning-option... yes
checking whether C++ compiler accepts -Wno-unknown-warning-option... yes
```

Upon investigating the _snmp-ups.c_ file I could not in fact find a dangling pointer there.
The build finishes without errors, but I thought I'd share the warnings here in case they become relevant.

Should these warnings pose a problem to the function of NUT or can I ignore them as oddities for now?

jimklimov, 2023-08-27:
I'll take a look, but I think oddities for now. I've seen those warnings in Windows builds but not on any other platform though...

desertwitch, 2023-08-29:
Thanks for your investigating, Jim.

I've actually found this warning to occur in another (perhaps similar) setting, might be of interest too:
https://mail.gnu.org/archive/html/grub-devel/2022-03/msg00259.html
In that case it seems to also have been a kind of false-positive and they ended up suppressing it.

I've extensively tested the build result since and have not found anything out of the ordinary.
Should anything arise, I'll report back here, will treat it as a distribution oddity in the meantime.

jimklimov, 2023-08-29:
Thanks for the tip!

jimklimov, 2023-10-02:
Possibly drilled better into it now, should be fixed by https://github.com/networkupstools/nut/commit/7c9405cfc13d13b5412262284d7b75b33cb7b5e6

BTW, duplicate of #1705

closed 2023-10-02 by jimklimov

desertwitch, 2023-10-04:
confirmed working here now, thanks!