networkupstools/nut #2028: cyberpower-mib.c: Instant commands don't work

Bomorav, 2023-08-30:
UPS: Cyberpower OLS1500ERT2U + RMCARD205
OS: Rocky Linux 9.2
nut-2.8.0 patched with newest cyberpower-mib (1.54) and snmp-ups (1.30) drivers

Instant commands don't work,

I found that the problem is in the instant commands declarations in **cyberpower-mib.c**.
OID values are stored in the third parameter but should be in place of the fifth parameter (default value) which is NULL.
If I place OID values to default value parameter, the commands work.

Looking to the cyberpower-mib.c history I don't understand this commit
https://github.com/networkupstools/nut/commit/8bcf255b624e6ba6c89a2a3eeab37658d9cb8cb4

From **snmp-ups.c**

```
/* Common function for setting OIDs, from a NUT variable name,
 * used by su_setvar() and su_instcmd()
 * Params:
 * @mode: SU_MODE_INSTCMD for instant commands, SU_MODE_SETVAR for settings
 * @varname: name of variable or command to set the OID from
 * @val: value for settings, NULL for commands

 * Returns
 *   STAT_SET_HANDLED if OK,
 *   STAT_SET_INVALID or STAT_SET_UNKNOWN if the command / setting is not supported
 *   STAT_SET_FAILED otherwise
 */
static int su_setOID(int mode, const char *varname, const char *val)
{
. . .
```
So for SU_MODE_INSTCMD val is NULL and OID value is taken from default value.

Bomorav, 2023-08-30:
Simply the commit was wrong and based on an incorrect statement "default value for SU_TYPE_CMD should be NULL".

The command declarations before commit was correct.

Also looking at the drivers for other snmp ups models I can see that they insert the OID value in the "default value" string parameter, e.g. apc-mib.c
`{ "shutdown.return", 0, 1, ".1.3.6.1.4.1.318.1.1.1.6.1.1.0", "2", SU_TYPE_CMD | SU_FLAG_OK, NULL },`

overhacked, 2023-08-30:
Found what happened! I was working with the NUT code prior to https://github.com/networkupstools/nut/commit/5d97d0706994888aa37342940bf7720fb9eb7500 when I pushed #686. So much happened leading up to v2.8.0 that my PR didn't get updated to match the new way of specifying default values. Opening a PR to fix this.

jimklimov, 2023-08-30:
Thanks for the quick analysis and fix! :)

PR #2029 (2023-08-31): Revert "cyberpower-mib: fix instant command declarations"
This reverts commit 8bcf255b624e6ba6c89a2a3eeab37658d9cb8cb4 from PR #686.

Cyberpower `snmp_info_t` got updated to conform with pre 5d97d07 instant
commands layout, but got merged after it. Reverting 8bcf255 so that command
values are correctly in `su_info_t->dfl`.

Fixes #2028.