networkupstools/nut #2030: Support for Tripp Lite SMART1500PSGLCD (3028)

michaelfucci, 2023-08-30:
Hi,

I was wondering whether anyone has been able to a Tripp Lite SMART1500PSGLCD working with NUT. Adding `productid=3028` to the _Extra Arguments to driver_  section on the pfsense NUT configuration page results in a bare minimum of information returned (mfr., model, serial, driver, etc) and reports _ups.status_ as _OB_ (On Battery) when it's plugged in. Tripp Lite support was unable to help. Initially, they said it should work but then stated that it's unsupported (with no known plans to add support).

ThatWileyGuy, 2023-12-23:
I will be taking a look at this.

ThatWileyGuy, 2023-12-23:
I made similar changes to #963 and it doesn't seem to have helped:
```
$ sudo upsc tifa-ups
[sudo] password for wiley:
device.mfr: TRIPP LITE UPS
device.model: TRIPP LITE UPS
device.serial: 3234BV4SM896700217
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 3028
driver.parameter.synchronous: auto
driver.parameter.vendorid: 09ae
driver.state: quiet
driver.version: 2.8.1
driver.version.data: TrippLite HID 0.85
driver.version.internal: 0.52
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
ups.mfr: TRIPP LITE UPS
ups.model: TRIPP LITE UPS
ups.productid: 3028
ups.serial: 3234BV4SM896700217
ups.status: OB
ups.vendorid: 09ae
```

... but that status should be OL.

ThatWileyGuy, 2023-12-23:
Are there any tools I can try to see how this UPS's communication differs from the previous ones?

ThatWileyGuy, 2024-03-07:
I set debug=6 per https://github.com/networkupstools/nut/wiki/Changing-NUT-daemon-debug-verbosity and got this:
```
Mar 06 16:34:41 jenova nut-driver@ups[6210]:  177.665330        [D2] nut_libusb_get_interrupt: Connection timed out
Mar 06 16:34:41 jenova nut-driver@ups[6210]:  177.665455        [D1] Got 0 HID objects...
Mar 06 16:34:41 jenova nut-driver@ups[6210]:  177.665500        [D1] Quick update...
Mar 06 16:34:41 jenova nut-driver@ups[6210]:  177.665554        [D5] send_to_all: SETINFO driver.state "quiet"
Mar 06 16:34:41 jenova nut-driver@ups[6210]:  177.665605        [D6] send_to_all: write 29 bytes to socket 10 succeeded (ret=29): SETINFO driver.state "quiet"
Mar 06 16:34:42 jenova nut-driver@ups[6210]:  178.915768        [D5] send_to_all: SETINFO driver.state "updateinfo"
Mar 06 16:34:42 jenova nut-driver@ups[6210]:  178.915913        [D6] send_to_all: write 34 bytes to socket 10 succeeded (ret=34): SETINFO driver.state "updateinfo"
Mar 06 16:34:42 jenova nut-driver@ups[6210]:  178.915958        [D1] upsdrv_updateinfo...
Mar 06 16:34:43 jenova nut-driver@ups[6210]:  179.666286        [D2] nut_libusb_get_interrupt: Connection timed out
Mar 06 16:34:43 jenova nut-driver@ups[6210]:  179.666408        [D1] Got 0 HID objects...
Mar 06 16:34:43 jenova nut-driver@ups[6210]:  179.666454        [D1] Quick update...
Mar 06 16:34:43 jenova nut-driver@ups[6210]:  179.666505        [D5] send_to_all: SETINFO driver.state "quiet"
Mar 06 16:34:43 jenova nut-driver@ups[6210]:  179.666556        [D6] send_to_all: write 29 bytes to socket 10 succeeded (ret=29): SETINFO driver.state "quiet"
```

... so it seems like we're not getting anything from the UPS at all?

HeedfulCrayon, 2024-03-07:
> Are there any tools I can try to see how this UPS's communication differs from the previous ones?

I am in the same boat, looking for support for the SMART1000PSGLCD.  I did find that there is TrippLite's PowerAlert Local for fedora, but it is closed source... don't know if the traffic over the wire could potentially be sniffed to reverse engineer the communication

ThatWileyGuy, 2024-03-07:
Oh, it could be interesting to get an `strace` of that? Would let us see the syscalls and buffers of USB traffic going back and forth...

HeedfulCrayon, 2024-03-08:
@ThatWileyGuy another option could be to use the recommendation here to capture the USB traffic
https://wiki.wireshark.org/CaptureSetup/USB#a-word-of-warning-about-usbpcap

ThatWileyGuy, 2024-12-15:
I'm scanning through this and comparing it to other drivers to see if this is a derivative of some existing product, but if anyone else wants to look, here's a traffic dump from plugging in one of these UPSes to a Windows box with PowerAlert Office installed. The UPS was charging and had the optional external battery attached.
[UPS.zip](https://github.com/user-attachments/files/18138835/UPS.zip)

ThatWileyGuy, 2024-12-15:
[image]
Looking at this and comparing it to what sense I can make of the HID metadriver, it seems like this controller might be something totally new? The reports look dramatically smaller than what the HID code seems to expect right now, but it's also using multiple report IDs and doesn't look like serial hacked onto HID...