networkupstools/nut #2031: Add an ups.conf toggle for voltage multipliers

jimklimov, 2023-08-31:
> > battery.voltage: 1.3 battery.voltage.nominal: 1.2
> > is this a bug? I thought that voltage should be a bit higher.

_The original post by @gyulaasztalos in https://github.com/networkupstools/nut/issues/616#issuecomment-1640371303 experimented with neutering of `divide_by_10_conversion()` and `legrand_times10M_info()` in the mapping tables._

This issue is about adding a configuration toggle (and perhaps ways of guessing a default if not specified - e.g. based on some mapping of device/model/firmware... identifiers) to report original voltages or factor them. Seems to be a common problem for many vendors, and sometimes for same brand/model devices with different HW/FW releases...

Note that currently not many drivers (or mapping tables where used) suggest having such a multiplication/division at all. For those that do, if we guess a default value - log this, with a suggestion to use the toggle if numbers seem off the target.

Loosely related to #1279 / #1652 area as well.

user01010111, 2026-09-20:
I have prepared a local draft for #2031 that adds four opt-in Legrand voltage factors: measured input, output and battery voltage, plus nominal battery voltage separately. An explicit factor replaces the correction after HID decoding and before formatting; 1 disables that correction. Existing defaults stay unchanged, and the draft adds no automatic guesses or HID descriptor changes.

The regression tests cover both mapping alternatives for each field, independent factors, fractional readings, invalid settings and out-of-range results. A local demonstration through the driver configuration and publication code also confirms unchanged default output. This uses supplied decoded readings; I have not validated it against an affected UPS.

Before I prepare a PR, is this Legrand-only option contract a useful first step, or would you prefer this to use the common conversion work in #3615/#3614 on FTY? I would describe it as a partial response to #2031, with the wider interface and automatic detection left for agreement.

I used OpenAI Codex to assist with the draft and validation.

jimklimov, 2026-09-23:
Have not decided yet, too lost in the recent storm of PRs to think straight ;)

user01010111, 2026-09-23:
> Have not decided yet, too lost in the recent storm of PRs to think straight ;)

Completely understandable, sorry 😅

Ping me back on this issue when you/we have some breathing room and we can loop back around.