networkupstools/nut #2034: APC SM-750i with AP 9620 not run

tonygh77, 2023-09-04:
Hello, i have a APC SM750I and I bought the AP 9620 legacy card

These are the cables that came with me
- USB-RS232 cable Profilic PL2303
- Original APC cable RS232 to RS232

This is mine `ups.conf`

`pollinterval` = `15`
`maxretry` = `3`

`offdelay` = `120`
`ondelay` = `240`

`[apc]`
`# APC SMT-750I`
`driver` = `apcsmart`
`port` = `/dev/ttyUSB0`
`ttymode` = `raw`
`desc = APC SMT-750I`

If you run `upsdrvctl start` this is my output 

`Network UPS Tools - UPS driver controller 2.8.0`
`Network UPS Tools - APC Smart protocol driver 3.2 (2.8.0)`
`APC command Table version 3.1`
`[0x05] unrecognized`
`[/] unrecognized`
`[0x0d] unrecognized`

What am I doing wrong?
Thank you for your time

tonygh77, 2023-09-04:
I connected the USB directly to the port of the AP 9620 card, then direct connection ups-pc and I launched net-scanner -U
This is ouptput:

```
root@test:~# nut-scanner -U
Scanning USB bus.
[nutdev1]
        driver = "usbhid-ups"
        port = "auto"
        vendorid = "051D"
        productid = "0002"
        product = "Legacy Communication Card FW:LCC 03.1 / ID=5004"
        serial = "ASxxxxxxx"
        vendor = "American Power Conversion"
        bus = "001"
root@test:~# 

```
Later I modified the file `ups.conf`

```
pollinterval = 15
maxretry = 3

offdelay = 120
ondelay = 240

[APC]
# APC SM750I
driver = usbhid-ups
port = auto
desc = "APC750I"
vendorid = 051D
productid = 0002
product = Legacy Communication Card FW:LCC 03.1 / ID=5004
serial = ASxxxxxxxxxx
vendor = American Power Conversion
bus = 001
```
and later `upsdrvctl start`

This is output

```
Network UPS Tools - UPS driver controller 2.8.0
Error: no UPS definitions found in ups.conf
```
I still don't understand what I'm doing wrong
Help please

jimklimov, 2023-09-04:
That's a bit odd. Maybe it rejected the config due to missing double-quotes in the multi-token values (e.g. vendor, product)?

jimklimov, 2023-09-04:
You can try to simplify it and leave just the numeric identifiers, comment away the string comparisons (they are more useful when you have several devices to discern reliably). 

Also check permissions (e.g. do you run `upsdrvctl` with the user account that may read `ups.conf` and the directory it is in?)

Also, just in case - be sure to not define the same device section twice :) (e.g. earlier `apc` and new `APC` might maybe clash, not sure about case-sensitivity here)

jimklimov, 2023-09-04:
As for the originally posted issue, I wonder if this device/card do talk "APC Smart" protocol, and then which one if yes. There's also an `apcsmart-old` driver to try...

In the NUT `master` branch there was recently a small fix for regression vs. NUT v2.7.4, with misinterpreted `capabilities` not discarded like before, but causing the driver to fail. However those had logged messages to that effect => #2015

tonygh77, 2023-09-04:
I cleaned ups.conf file :

```
pollinterval = 15
maxretry = 3

offdelay = 120
ondelay = 240

[apc]
# APC SMT750I
driver = usbhid-ups
port = auto
desc = "SMT750I"
vendorid = 051D
productid = 0002
serial = ASxxxxxxxx
```

This is output of `upsdrvctl start`

```
root@test:~# upsdrvctl start
Network UPS Tools - UPS driver controller 2.8.0
Network UPS Tools - Generic HID driver 0.47 (2.8.0)
USB communication driver (libusb 1.0) 0.43
Duplicate driver instance detected (PID file /run/nut/usbhid-ups-apc.pid exists)! Terminating other driver!
HIDParse: LogMax is less than LogMin. Vendor HID report descriptor may be incorrect; interpreting LogMax -1 as 65535 in ReportID: 0x94
Using subdriver: APC HID 0.98
root@test:~# 
```

jimklimov, 2023-09-04:
Seems it has recognized the device and did not complain. A competitor (another copy of the driver) ran and was killed off. Probably it was started via `systemd` (the templated unit might be auto-populated and started by `nut-driver-enumerator` with NUT v2.8.0+ upon config file changes), so unless you `systemctl stop nut-driver@apc` first, the `systemd` unit would revive itself and in the process kill the driver instance you had started manually. Just something to keep in mind while experimenting...

tonygh77, 2023-10-18:
Hi, I would like to confirm that the SM750i works with the AP9620 board using a simple USB cable (no RS232 cable)