networkupstools/nut #2035: Cannot compile, man fails

sathio, 2023-09-05:
I get this error when trying to install the package after it compiled correctly:

```make[2]: Entering directory '/opt/nut/nut-2.8.0/docs/man'
  DOC-MAN  Generating nut.conf.5
a2x: ERROR: "xmllint" --nonet --noout --valid "/opt/nut/nut-2.8.0/docs/man/tmp/man5.nut.conf.5.1312216/nut.conf.xml" returned non-zero exit status 4

mv: cannot stat './tmp/man5.nut.conf.5.1312216/nut.conf.5': No such file or directory```

jimklimov, 2023-09-05:
Thanks for the report. Is this with a master-branch build or with the actual 2.8.0 release/snapshot, as it was a year ago? If not, which git commit are you building?

Generally, such errors tend to be some asciidoc markup problem, like mismatched heading levels in combined documents which include pages to make a book... but `make distcheck` did not complain on many systems (which should include doc generation and installation).

jimklimov, 2023-09-05:
Calling (copy-pasting) that quoted `xmllint` command usually yields the detailed diagnostics from that tool, hidden by asciidoc/a2x as the wrapper of sorts.

In the worst case, you can reconfigure `--without-docs` if that's acceptable to just have stuff running... But if it is a problem with current codebase, I'd rather discover and fix it :)

sathio, 2023-09-05:
thanks for the super fast answer, it happened with this tar file:  `nut-2.8.0.tar.gz` I think it was called `signed`  and I am on 

```Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.4 LTS
Release:	20.04
Codename:	focal``` 

I'll just compile without docs, it's totally acceptable. if you need me to make any test to hunt and squash the bug, let me know.

closed 2023-09-11 by jimklimov