networkupstools/nut #2038: Legrand "WHAD HE" should be supported but it doesn't work.

sathio, 2023-09-05:
Hi, according to the Hardware compatibility list the `WHAD HE` series should be supported with 4 stars, but in reality it doesn't work. I've looked a bit and it appears that in 2018 a Legrand representative was about to release a specific driver for this series, but as far as I see it was not merged  https://github.com/zykh/nut/pull/4

jimklimov, 2023-09-11:
Just in case - which NUT version are you running? Many systems still package 2.7.4, and fixes from 2018 would be part of 2.8.0 (or current master) codebase.

sathio, 2023-09-11:
I compiled from nut-2.8.0

sathio, 2023-09-11:
root@tank:/opt/nut-2.8.0# /usr/local/ups/sbin/upsdrvctl -u nut -D start
      Network UPS Tools - UPS driver controller 2.8.0
         0.000000 [D1] Starting UPS: Basement
      Network UPS Tools - Metasystem UPS driver 0.08 (2.8.0)
      Communications with UPS lost: Receive error (STX): 0!!!

      Communications with UPS lost: Receive error (STX): 0!!!

      Communications with UPS lost: Receive error (STX): 0!!!

      Communications with UPS lost: Receive error (STX): 0!!!

      Communications with UPS lost: Receive error (STX): 0!!!

      Communications with UPS lost: Error executing command 0

      Could not communicate with the ups: No such file or directory
         5.009398 Driver failed to start (exit status=1)

jimklimov, 2023-09-11:
Could it be that the UPS firmware (or HW it runs on) is slower than the driver expects? Can you try increasing the timeout (the last couple of "1" sec and "0" usec parameters to `ser_get_char()`) as a first shot?

While at it, since you are building NUT, can you try using the current master branch as the base (especially useful if a PR would come out of this adventure)? At the very least, there were some improvements to make run-time debugging easier.

If using an IDE like NetBeans is an option (can use GDB over SSH to run code on an appliance and an IDE on your workstation), you could even try stepping through the driver code directly. Otherwise, peppering interesting places in the code with `upsdebugx()` is an option - then also a larger debug verbosity level (and starting the driver directly, not via `upsdrvctl` wrapper) might help.

One more idea is to check FS permissions for `/dev/ttyS0` - can the `nut` driver runtime user account access it (R/W)? Probably yes, should have got permissions errors otherwise, but better check it explicitly :) Also, is it connected to a true serial port, or some USB dongle (might be `/dev/ttyUSB0` instead, then - and may need some fuss about `udev` or similar setup for auto-permissions)?

sathio, 2023-09-11:
It is most definitely not attached to a Usb dongle, I will make some tests and write here if I find something.

thanks

sathio, 2023-09-11:
root@tank:/opt/nut# /usr/local/ups/sbin/upsdrvctl -u nut  -D start
      Network UPS Tools - UPS driver controller 2.8.0-2336-ge8985bcfa
         0.000001 [D1] upsdrvctl commanding all drivers (1 found): (null)
         0.000038 [D1] Starting UPS: Basement
      Network UPS Tools - Metasystem UPS driver 0.09 (2.8.0-2336-ge8985bcfa)
      Communications with UPS lost: Receive error (STX): 0!!!

      Communications with UPS lost: Receive error (STX): 0!!!

      Communications with UPS lost: Receive error (STX): 0!!!

      Communications with UPS lost: Receive error (STX): 0!!!

      Communications with UPS lost: Receive error (STX): 0!!!

      Communications with UPS lost: Error executing command 0

      Could not communicate with the ups: No such file or directory
         5.009339 Driver failed to start (exit status=1)
         5.009397 [D1] Completed the job of upsdrvctl tool, cleaning up and exiting now

--
Git log

      commit e8985bcfa77df22c02b6ca6e0936a0bdfeafd754 (HEAD -> master, origin/master, origin/HEAD)
      Merge: 7ae19e5c5 98ee05919
      Author: Jim Klimov <jimklimov@gmail.com>
      Date:   Sun Sep 10 20:26:51 2023 +0200

          Merge pull request #2045 from jimklimov/docs-usb-explore

-- 
and this is the driver itself.

      getpid()                                = 2546612
      fstat(4, {st_mode=S_IFREG|0644, st_size=0, ...}) = 0
      write(4, "2546612\n", 8)                = 8
      close(4)                                = 0
      umask(022)                              = 022
      openat(AT_FDCWD, "/dev/ttyS0", O_RDWR|O_EXCL|O_NOCTTY|O_NONBLOCK) = 4
      flock(4, LOCK_EX|LOCK_NB)               = 0
      ioctl(4, TCGETS, {B2400 -opost -isig -icanon -echo ...}) = 0
      ioctl(4, TCFLSH, TCIFLUSH)              = 0
      ioctl(4, TCGETS, {B2400 -opost -isig -icanon -echo ...}) = 0
      ioctl(4, SNDCTL_TMR_START or TCSETS, {B2400 -opost -isig -icanon -echo ...}) = 0
      ioctl(4, TCGETS, {B2400 -opost -isig -icanon -echo ...}) = 0
      write(4, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 100) = 100
      clock_nanosleep(CLOCK_REALTIME, 0, {tv_sec=0, tv_nsec=0}, NULL) = 0
      write(4, "\2\2\0\2", 4)                 = 4
      clock_nanosleep(CLOCK_REALTIME, 0, {tv_sec=0, tv_nsec=0}, NULL) = 0
      select(5, [4], NULL, NULL, {tv_sec=1, tv_usec=0}) = 0 (Timeout)
      write(2, "Communications with UPS lost: Re"..., 57Communications with UPS lost: Receive error (STX): 0!!!

-- 

    ls -lah /dev/ttyS0
      crw-rw---- 1 root nut 4, 64 Sep 11 13:59 /dev/ttyS0

jimklimov, 2023-09-11:
For a random shot, check if they talk Qx protocols? e.g. with `nutdrv_qx` or even echo/nc/socat/minicom/... chatter to the serial port directly?

sathio, 2023-09-11:
/usr/local/ups/bin/nutdrv_qx -a Basement -u nut
    Network UPS Tools - Generic Q* Serial driver 0.33 (2.8.0-2336-ge8985bcfa)

    Device not supported!
    Device not supported!

I'll test the chatter!