networkupstools/nut #2039: Writing a custom driver with support for multipe outlets

antonlyap, 2023-09-05:
I'm trying to write a USB HID driver to support a custom UPS with multiple independent outlets. The relevant section of the `hid2nut` table looks like this:

```c
{ "outlet.1.switchable", 0, 0, "UPS.OutletSystem.Outlet.[1].Switchable", NULL, "%s", HU_FLAG_STATIC, yes_no_info },
{ "outlet.1.load.on", 0, 0, "UPS.OutletSystem.Outlet.[1].DelayBeforeStartup", NULL, DEFAULT_ONDELAY, HU_TYPE_CMD, NULL },
{ "outlet.1.load.off", 0, 0, "UPS.OutletSystem.Outlet.[1].DelayBeforeShutdown", NULL, DEFAULT_OFFDELAY, HU_TYPE_CMD, NULL },
{ "outlet.1.load.cycle", 0, 0, "UPS.OutletSystem.Outlet.[1].DelayBeforeReboot", NULL, "1", HU_TYPE_CMD, NULL },
{ "outlet.2.switchable", 0, 0, "UPS.OutletSystem.Outlet.[2].Switchable", NULL, "%s", HU_FLAG_STATIC, yes_no_info },
{ "outlet.2.load.on", 0, 0, "UPS.OutletSystem.Outlet.[2].DelayBeforeStartup", NULL, DEFAULT_ONDELAY, HU_TYPE_CMD, NULL },
{ "outlet.2.load.off", 0, 0, "UPS.OutletSystem.Outlet.[2].DelayBeforeShutdown", NULL, DEFAULT_OFFDELAY, HU_TYPE_CMD, NULL },
{ "outlet.2.load.cycle", 0, 0, "UPS.OutletSystem.Outlet.[2].DelayBeforeReboot", NULL, "1", HU_TYPE_CMD, NULL },
{ "outlet.3.switchable", 0, 0, "UPS.OutletSystem.Outlet.[3].Switchable", NULL, "%s", HU_FLAG_STATIC, yes_no_info },
{ "outlet.3.load.on", 0, 0, "UPS.OutletSystem.Outlet.[3].DelayBeforeStartup", NULL, DEFAULT_ONDELAY, HU_TYPE_CMD, NULL },
{ "outlet.3.load.off", 0, 0, "UPS.OutletSystem.Outlet.[3].DelayBeforeShutdown", NULL, DEFAULT_OFFDELAY, HU_TYPE_CMD, NULL },
{ "outlet.3.load.cycle", 0, 0, "UPS.OutletSystem.Outlet.[3].DelayBeforeReboot", NULL, "1", HU_TYPE_CMD, NULL },
{ "outlet.4.switchable", 0, 0, "UPS.OutletSystem.Outlet.[4].Switchable", NULL, "%s", HU_FLAG_STATIC, yes_no_info },
{ "outlet.4.load.on", 0, 0, "UPS.OutletSystem.Outlet.[4].DelayBeforeStartup", NULL, DEFAULT_ONDELAY, HU_TYPE_CMD, NULL },
{ "outlet.4.load.off", 0, 0, "UPS.OutletSystem.Outlet.[4].DelayBeforeShutdown", NULL, DEFAULT_OFFDELAY, HU_TYPE_CMD, NULL },
{ "outlet.4.load.cycle", 0, 0, "UPS.OutletSystem.Outlet.[4].DelayBeforeReboot", NULL, "1", HU_TYPE_CMD, NULL },
```

The report descriptor:
```
$ rexx rd.rex -d --hex 05 84 09 04 a1 01 85 01 09 18 a1 00 09 20  a1 00 75 08 95 01 09 6c b1 03 75 08 95 03 17 ff ff ff ff 27 3c 00 00 00 09 56 09 57 09 55 b1 02 c0 09 20 a1 00 75 08 95 01 09 6c b1 03 75 08 95 03 09 56 09 57 09 55 b1 02 c0 c0 c0

//--------------------------------------------------------------------------------
// Decoded Application Collection
//--------------------------------------------------------------------------------

/*
05 84        (GLOBAL) USAGE_PAGE         0x0084 Power Device Page 
09 04        (LOCAL)  USAGE              0x00840004 UPS (Application Collection)  
A1 01        (MAIN)   COLLECTION         0x01 Application (Usage=0x00840004: Page=Power Device Page, Usage=UPS, Type=Application Collection)
85 01          (GLOBAL) REPORT_ID          0x01 (1)  
09 18          (LOCAL)  USAGE              0x00840018 Outlet System (Physical Collection)  
A1 00          (MAIN)   COLLECTION         0x00 Physical (Usage=0x00840018: Page=Power Device Page, Usage=Outlet System, Type=Physical Collection)
09 20            (LOCAL)  USAGE              0x00840020 Outlet (Physical Collection)  
A1 00            (MAIN)   COLLECTION         0x00 Physical (Usage=0x00840020: Page=Power Device Page, Usage=Outlet, Type=Physical Collection)
75 08              (GLOBAL) REPORT_SIZE        0x08 (8) Number of bits per field  
95 01              (GLOBAL) REPORT_COUNT       0x01 (1) Number of fields  
09 6C              (LOCAL)  USAGE              0x0084006C Switchable (Dynamic Flag)  
B1 03              (MAIN)   FEATURE            0x00000003 (1 field x 8 bits) 1=Constant 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
75 08              (GLOBAL) REPORT_SIZE        0x08 (8) Number of bits per field <-- Redundant: REPORT_SIZE is already 8 
95 03              (GLOBAL) REPORT_COUNT       0x03 (3) Number of fields  
17 FFFFFFFF        (GLOBAL) LOGICAL_MINIMUM    0xFFFFFFFF (-1)  <-- Info: Consider replacing 17 FFFFFFFF with 15 FF
27 3C000000        (GLOBAL) LOGICAL_MAXIMUM    0x0000003C (60)  <-- Info: Consider replacing 27 3C000000 with 25 3C
09 56              (LOCAL)  USAGE              0x00840056 Delay Before Startup (Dynamic Value)  
09 57              (LOCAL)  USAGE              0x00840057 Delay Before Shutdown (Dynamic Value)  
09 55              (LOCAL)  USAGE              0x00840055 Delay Before Reboot (Dynamic Value)  
B1 02              (MAIN)   FEATURE            0x00000002 (3 fields x 8 bits) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
C0               (MAIN)   END_COLLECTION     Physical 
09 20            (LOCAL)  USAGE              0x00840020 Outlet (Physical Collection)  
A1 00            (MAIN)   COLLECTION         0x00 Physical (Usage=0x00840020: Page=Power Device Page, Usage=Outlet, Type=Physical Collection)
75 08              (GLOBAL) REPORT_SIZE        0x08 (8) Number of bits per field <-- Redundant: REPORT_SIZE is already 8 
95 01              (GLOBAL) REPORT_COUNT       0x01 (1) Number of fields  
09 6C              (LOCAL)  USAGE              0x0084006C Switchable (Dynamic Flag)  
B1 03              (MAIN)   FEATURE            0x00000003 (1 field x 8 bits) 1=Constant 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
75 08              (GLOBAL) REPORT_SIZE        0x08 (8) Number of bits per field <-- Redundant: REPORT_SIZE is already 8 
95 03              (GLOBAL) REPORT_COUNT       0x03 (3) Number of fields  
09 56              (LOCAL)  USAGE              0x00840056 Delay Before Startup (Dynamic Value)  
09 57              (LOCAL)  USAGE              0x00840057 Delay Before Shutdown (Dynamic Value)  
09 55              (LOCAL)  USAGE              0x00840055 Delay Before Reboot (Dynamic Value)  
B1 02              (MAIN)   FEATURE            0x00000002 (3 fields x 8 bits) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
C0               (MAIN)   END_COLLECTION     Physical 
C0             (MAIN)   END_COLLECTION     Physical 
C0           (MAIN)   END_COLLECTION     Application 
*/

// All structure fields should be byte-aligned...
#pragma pack(push,1)

//--------------------------------------------------------------------------------
// Power Device Page featureReport 01 (Device <-> Host)
//--------------------------------------------------------------------------------

typedef struct
{
  uint8_t  reportId;                                 // Report ID = 0x01 (1)
                                                     // Collection: CA:UPS CP:OutletSystem CP:Outlet
  int8_t   POW_UPSOutletSystemOutletSwitchable;      // Usage 0x0084006C: Switchable, Value =  to 
  int8_t   POW_UPSOutletSystemOutletDelayBeforeStartup; // Usage 0x00840056: Delay Before Startup, Value = -1 to 60
  int8_t   POW_UPSOutletSystemOutletDelayBeforeShutdown; // Usage 0x00840057: Delay Before Shutdown, Value = -1 to 60
  int8_t   POW_UPSOutletSystemOutletDelayBeforeReboot; // Usage 0x00840055: Delay Before Reboot, Value = -1 to 60
  int8_t   POW_UPSOutletSystemOutletSwitchable_1;    // Usage 0x0084006C: Switchable, Value = -1 to 60
  int8_t   POW_UPSOutletSystemOutletDelayBeforeStartup_1; // Usage 0x00840056: Delay Before Startup, Value = -1 to 60
  int8_t   POW_UPSOutletSystemOutletDelayBeforeShutdown_1; // Usage 0x00840057: Delay Before Shutdown, Value = -1 to 60
  int8_t   POW_UPSOutletSystemOutletDelayBeforeReboot_1; // Usage 0x00840055: Delay Before Reboot, Value = -1 to 60
} featureReport01_t;

#pragma pack(pop)
```

My problem is that the outlets aren't getting detected at all:
- `upsc` doesn't display `outlet.n.switchable`
- `upscmd` doesn't display any commands

I'm not seeing any errors or anything odd in the logs:
```
   0.068896	[D5] hid_lookup_usage: UPS -> 00840004
   0.068940	[D5] hid_lookup_usage: OutletSystem -> 00840018
   0.068976	[D5] hid_lookup_usage: Outlet -> 00840020
   0.069005	[D5] hid_lookup_usage: [4] -> not found in lookup table
   0.069047	[D5] string_to_path: hid_lookup_usage failed, checking if token [4] is a raw value
   0.069076	[D5] hid_lookup_usage: DelayBeforeShutdown -> 00840057
   0.069111	[D4] string_to_path: depth = 5
```
Looking at the source code, these messages are to be expected.

Further debugging shows that the driver is looking for the HID path `840004 840018 840020 FF0004 840057` but only finds `840004 840018 840020 840057`. In other words, in the logic which parses HID report descriptors, there doesn't seem to be any code responsible for indexed collections.

Am I doing something wrong here, or is my use case unsupported for now? Please let me know if any further info is needed. Any help would be appreciated.

jimklimov, 2023-09-05:
Hello, sorry I can't help that deep into USB logic myself, so would try to ping some people I know dealt with something around that :)

CC @aquette @nbriggs

antonlyap, 2023-09-06:
Thanks a lot!

I'm currently looking into the `mge-hid` driver, which appears to be the only other HID driver using the `outlet.n.*` notation. Other people have it working with real MGE devices (example: https://github.com/networkupstools/nut/issues/569#issuecomment-509116558), so I tried the following:
- changed the vendor ID and product ID on the device
- modified the driver to look for `UPS.OutletSystem.Outlet.[2].Switchable` instead of `UPS.OutletSystem.Outlet.[2].PresentStatus.Switchable`
Still, nothing shows up. This makes me think that there's something wrong with my report descriptor. So is there any chance that you have a report descriptor from an MGE UPS for me to test?

jimklimov, 2023-09-06:
I think I had access to a smallish Eaton ECO 650 which claimed several outlets, gotta check. Not sure it could actually manage them independently too, though.

For report descriptor, would NUT debug dumps suffice?

antonlyap, 2023-09-06:
Absolutely, a dump with debug level 6 would be great. Thank you.

jimklimov, 2023-09-06:
````
;: ./drivers/usbhid-ups -DDDDDD -d 1 -a eco650
Network UPS Tools - Generic HID driver 0.50 (2.8.0-2287-gb0c97f71c)
USB communication driver (libusb 1.0) 0.45
   0.000000     [D5] do_upsconf_args: confupsname=(null), var=debug_min, val=0
   0.000011     [D5] do_upsconf_args: call do_global_args()
   0.000016     [D3] do_global_args: var='debug_min' val='0'
   0.000029     [D5] do_upsconf_args: confupsname=dummy, var=driver, val=dummy-ups
   0.000034     [D5] do_upsconf_args: confupsname=dummy, var=port, val=dummy.dev
   0.000039     [D5] do_upsconf_args: confupsname=eco650, var=driver, val=usbhid-ups
   0.000042     [D5] do_upsconf_args: call main_arg()
   0.000046     [D3] main_arg: var='driver' val='usbhid-ups'
   0.000050     [D5] do_upsconf_args: not a main_arg()
   0.000054     [D5] do_upsconf_args: this is a 'driver' setting, may we proceed?
   0.000059     [D6] testval_reloadable: var=driver, oldval=usbhid-ups, newval=usbhid-ups, reloadable=0, reload_flag=0
   0.000063     [D6] testval_reloadable: verdict for (re)loading var=driver value: -1
   0.000067     [D5] do_upsconf_args: 'driver' setting already applied with this value
   0.000072     [D5] do_upsconf_args: confupsname=eco650, var=port, val=auto
   0.000075     [D5] do_upsconf_args: call main_arg()
   0.000079     [D3] main_arg: var='port' val='auto'
   0.000083     [D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=auto, reloadable=0, reload_flag=0
   0.000086     [D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000091     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000097     [D5] do_upsconf_args: confupsname=eco650, var=vendorid, val=0463
   0.000100     [D5] do_upsconf_args: call main_arg()
   0.000103     [D3] main_arg: var='vendorid' val='0463'
   0.000106     [D5] do_upsconf_args: not a main_arg()
   0.000109     [D5] do_upsconf_args: process as value
   0.000113     [D6] testvar_reloadable: searching for var=vendorid, vartype=2, reload_flag=0
   0.000117     [D6] testvar_reloadable: found var=vendorid, val='(null)' => '0463', vartype=2 => 2, found=0, reloadable=0, reload_flag=0
   0.000120     [D6] testvar_reloadable: verdict for (re)loading var=vendorid value: 1
   0.000125     [D5] send_to_all: SETINFO driver.parameter.vendorid "0463"
   0.000129     [D5] do_upsconf_args: confupsname=eco650, var=productid, val=FFFF
   0.000132     [D5] do_upsconf_args: call main_arg()
   0.000135     [D3] main_arg: var='productid' val='FFFF'
   0.000138     [D5] do_upsconf_args: not a main_arg()
   0.000141     [D5] do_upsconf_args: process as value
   0.000144     [D6] testvar_reloadable: searching for var=productid, vartype=2, reload_flag=0
   0.000148     [D6] testvar_reloadable: found var=productid, val='(null)' => 'FFFF', vartype=2 => 2, found=0, reloadable=0, reload_flag=0
   0.000152     [D6] testvar_reloadable: verdict for (re)loading var=productid value: 1
   0.000156     [D5] send_to_all: SETINFO driver.parameter.productid "FFFF"
   0.000160     [D5] do_upsconf_args: confupsname=eco650, var=product, val=Ellipse ECO
   0.000163     [D5] do_upsconf_args: call main_arg()
   0.000166     [D3] main_arg: var='product' val='Ellipse ECO'
   0.000169     [D5] do_upsconf_args: not a main_arg()
   0.000172     [D5] do_upsconf_args: process as value
   0.000174     [D6] testvar_reloadable: searching for var=product, vartype=2, reload_flag=0
   0.000178     [D6] testvar_reloadable: found var=product, val='(null)' => 'Ellipse ECO', vartype=2 => 2, found=0, reloadable=0, reload_flag=0
   0.000182     [D6] testvar_reloadable: verdict for (re)loading var=product value: 1
   0.000187     [D5] send_to_all: SETINFO driver.parameter.product "Ellipse ECO"
   0.000192     [D5] do_upsconf_args: confupsname=eco650, var=desc, val=Eaton Ellipse ECO 650
   0.000195     [D5] do_upsconf_args: call main_arg()
   0.000198     [D3] main_arg: var='desc' val='Eaton Ellipse ECO 650'
   0.000202     [D5] do_upsconf_args: confupsname=eco650, var=serial, val=000000000
   0.000205     [D5] do_upsconf_args: call main_arg()
   0.000208     [D3] main_arg: var='serial' val='000000000'
   0.000212     [D5] do_upsconf_args: not a main_arg()
   0.000215     [D5] do_upsconf_args: process as value
   0.000217     [D6] testvar_reloadable: searching for var=serial, vartype=2, reload_flag=0
   0.000221     [D6] testvar_reloadable: found var=serial, val='(null)' => '000000000', vartype=2 => 2, found=0, reloadable=0, reload_flag=0
   0.000224     [D6] testvar_reloadable: verdict for (re)loading var=serial value: 1
   0.000230     [D5] send_to_all: SETINFO driver.parameter.serial "000000000"
   0.000234     [D5] do_upsconf_args: confupsname=eco650, var=vendor, val=EATON
   0.000237     [D5] do_upsconf_args: call main_arg()
   0.000240     [D3] main_arg: var='vendor' val='EATON'
   0.000243     [D5] do_upsconf_args: not a main_arg()
   0.000246     [D5] do_upsconf_args: process as value
   0.000249     [D6] testvar_reloadable: searching for var=vendor, vartype=2, reload_flag=0
   0.000253     [D6] testvar_reloadable: found var=vendor, val='(null)' => 'EATON', vartype=2 => 2, found=0, reloadable=0, reload_flag=0
   0.000256     [D6] testvar_reloadable: verdict for (re)loading var=vendor value: 1
[... 2845 lines cut ...]
outlet.id: 1
outlet.power: 25
outlet.switchable: no
output.frequency.nominal: 50
output.voltage: 230.0
output.voltage.nominal: 230
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.firmware: 02
ups.load: 19
ups.mfr: EATON
ups.model: Ellipse ECO 650
ups.power.nominal: 650
ups.productid: ffff
ups.realpower: 99
ups.serial: 000000000
ups.status: OL
ups.timer.shutdown: -1
ups.timer.start: -1
ups.vendorid: 0463
   7.635311     [D5] send_to_all: SETINFO driver.state "cleanup.upsdrv"
   7.635315     [D1] upsdrv_cleanup...
   7.635424     [D5] send_to_all: SETINFO driver.state "cleanup.exit"
````

antonlyap, 2023-09-07:
The way MGE devices make this work is by using vendor-defined collections with different tags (`0x81`, `0x82`, `0x83`). NUT correctly recognizes them and generates the `FF000x` path entries for them. I confirmed this by editing the MGE report to use `0x81` collections for all outlets - and not all of them showed up in `upsc` output. Likewise, changing my device's report descriptor to use vendor-defined collection fixes it.

I'm not sure whether NUT not working with physical collections is a bug (or I might be missing something in the HID spec) - some feedback from the other people in this issue would be much appreciated. For now I'll be using the solution with vendor-defined collections, but if this needs fixing in NUT, I would be more than happy to try and submit a PR a bit later.

Thanks again @jimklimov :)