networkupstools/nut #2043: Add Port Number Mapping for libusb1.0

wavebvg, 2023-09-07:
Adds the ability to filter devices by USB port number using the portno variable
Necessary if it is impossible to distinguish several identical devices by other method (devices are identical, have no serial number, etc.).
[nut.portno.add.patch.zip](https://github.com/networkupstools/nut/files/12548335/nut.portno.add.patch.zip)

jimklimov, 2023-09-07:
So this essentially helps for systems that do not serve a device (on bus) number, but serve a port number?

I think the relatively recent fix to actually handle the device number tried in fact a port number method in earlier iterations.

Can you please re-post this as an actual github pull request, with attribution, CI tests and everything? Dpcs would need an update for example. If this is not implemented with libusb-0.x (or is it?), some ifdef's are due or at least log messages for skipped unsupported-by-this-build config option, etc.

I'm concerned about those "NO NO NO" abortive returns in particular. I guess they remained from some experiment to balance mge/liebert HIDs and do not directly relate to this proposed feature?

wavebvg, 2023-09-07:
> So this essentially helps for systems that do not serve a device (on bus) number, but serve a port number?

Yes, if the hardware supports it

> Can you please re-post this as an actual github pull request, with attribution, CI tests and everything?

It takes time. I added a new variable portno because according to the documentation, port is only used for COM ports. Does it make sense to use this parameter for COM and USB ports in different senses? Such a solution may simplify configuration and description in the future.

> Dpcs would need an update for example. If this is not implemented with libusb-0.x (or is it?), some ifdef's are due or at least log messages for skipped unsupported-by-this-build config option, etc.

In libusb 0.1 this is also possible. Not in the usual way, but it is possible.

> I'm concerned about those "NO NO NO" abortive returns in particular. I guess they remained from some experiment to balance mge/liebert HIDs and do not directly relate to this proposed feature?

I'm the one who couldn't handle the diff. I didn't want to use liebert-hid, which doesn't show the temperature on some devices.

I'll be back when I finish working on the integration requirements and prepare a correct patch.

jimklimov, 2023-09-07:
Well, `port` could historically be for (path to device of) serial ports. But eventually it became an ubiquitous required parameter, such as hostname (+`:portnum`) for snmp and other networked protocols, path to resource files for dummy-ups in emulator mode, or the currently neutered `auto` for USB. There were comments that libusb1 can actually allow to use devfs paths (`/dev/usb/0` or `/system/char/something` - some such) so adding that ability would be welcome by some people (number-based IDs are subject to enumeration changes).

wavebvg, 2023-09-15:
I have prepared a patch with pre-installed define.
There is no cross-platform solution for finding the hardware port number for libusb-0.1, and I really don't want to complicate the code for the sake of old solutions.
The parameter ```portno``` had to be renamed to ```busport``` so that one could guess the behavior features without reading the libusb documentation.
Used define ```WITH_USB_BUS PORT``` to enable support for the new filter.
Unfortunately, this is my first time writing code for c and I am not familiar with automake at all, so I need help in solving a specific task: it is necessary to arrange the inclusion of the flag by condition:
```
#if defined(LIBUSB_API_VERSION) && (LIBUSB_API_VERSION >= 0x01000102)
DEFINE WITH_USB_BUSPORT
#endif
```
Thank you very much
[0001-Update-blazer_usb.c-libusb0.c-and-7-more-files.patch.zip](https://github.com/networkupstools/nut/files/12623732/0001-Update-blazer_usb.c-libusb0.c-and-7-more-files.patch.zip)

jimklimov, 2023-09-16:
NOTE: Found what seems to be the previous experience in this area: #1819

jimklimov, 2023-09-16:
Thanks! I've snow-balled this a bit and made a PR #2054 starting from your patch-file :)

Also extended `nut-scanner` with this ability:
````
$ ./tools/nut-scanner/nut-scanner
Scanning USB bus.
No start IP, skipping SNMP
Scanning XML/HTTP bus.
No start IP, skipping NUT bus (old connect method)
Scanning NUT bus (avahi method).
Scanning IPMI bus.
Failed to create Avahi client: Daemon not running
[nutdev1]
        driver = "usbhid-ups"
        port = "auto"
        vendorid = "0463"
        productid = "FFFF"
        product = "Ellipse ECO"
        serial = "000000000"
        vendor = "EATON"
        bus = "003"
        device = "007"
        busport = "002"
        ###NOTMATCHED-YET###bcdDevice = "0100"

# WARNING: all-same character "serial" with 9 copies of '0' (0x30) reported in some devices: nutdev1
````

wavebvg, 2023-09-16:
Thank you very much, you have done a lot of work that I would not be able to cope with to some extent.
All the additions that I could not take into account are very clearly visible.

PR #2054 (2023-09-16): Allow USB device matching by `busport` value
Closes: #2043 (thanks @wavebvg for initial impulse!)
Related to: #1763 and like its outcome - does not impact the too-different old USB drivers like richcomm, bcmxcp, nutdrv_atcl

Note this is limited to systems that actually support the feature on hardware, OS and libusb version sides.
closed 2023-09-16 by wavebvg